OpenCPN/OpenCPN #2331: Implement DATSTA and DATEND S-57 attributes

bcn2, 2021-08-09:
Some important chart objects can have date constraints, definitions when those are valid (Start and End).
We have seen this with TSS, especially when those are changing.
Other objects are Buoys, Markers or Lights. Or Anchorages/Berths. Navigation Lines and similar. Canals. Cables. Or even Coast Guard Stations. Caution areas or similar.  Shore structures like Pylons or Pontoons. Drilling Platforms. Etc..

Hakansv, 2021-08-09:
Very good! 
One example. See pict. This buoy is moved and should not be shown now. (OK, a not updated chart but a good example)
The new buoy has a DATSTA 20200701
[image]

bcn2, 2021-08-14:
Implemented...

closed 2021-08-14 by bcn2