OpenCPN/OpenCPN #2332: Android 11/UDP

spike7638, 2021-08-09:
I've got two copies of OpenCPN running --- one under android 11, the other under Android 8 (or so). As far as I can tell, all data and settings on the two are identical. Both are listening on 192.168.1.255, port 10110

I have a data source on 192.168.1.102 that's broadcasting UDP packets for port 10110 containing NMEA sentences (including AIS). The Android 8 device shows ships, speed, etc; the Android 11 shows nothing. The NMEA debug window shows no data. And "socat" in Termux also shows nothing, although I gather that this may be because of permissions on Android and how the most recent versions don't let you look at arbitrary ports for some reason. 

What I don't know is whether I'm unusual in this regard, or others using recent Android releases have the same problem. 

If my packet-source should have a different strategy than "broadcast," that's fine too -- I'd just like to make this work, and don't understand what changed. (I should never have bought a new phone...)

rgleason, 2021-08-09:
Dave has not addressed Android yet and there have been many many changes. Don't expect it to work.

spike7638, 2021-08-09:
I guess I'll hold on to that Android 8 tablet for now.

bdbcat, 2021-08-10:
Rick...
Your comment is plain wrong, and quite misleading.  Please refrain from commenting on the state of Android code.  Comments like "Don't expect it to work."  are not helpful.
Thanks
Dave

bdbcat, 2021-08-10:
Spike...
Fault is easily replicated.  Works fine on Android 10, but same config does not receive UDP broadcast on A11.  Nothing to do with A11 port permissions, thankfully.  t seems that in Android 11, something called a "Wifi multicast lock" is required to receive broadcast UDP by WiFi connection.  Sadly, this change for A11 behavior is documented nowhere.  After some time tinkering, I stumbled onto the solution.
So, today you have two choices:
1.  Wait for an update to OpenCPN for Android, correcting this issue.  This could happen possibly in a few days.
2. ...or...  Use directed UDP.  That is, on your server side, arrange to send UDP messages to the exact and specific IP assigned to your Android device.  I know that this can be awkward, since the Android device IP is probably assigned by DHCP automatically, and may change from time-to-time.  But it does work.

Sorry for the problem.
Dave

rgleason, 2021-08-10:
Sorry, i should not have commented. I was thinking of  a time before CI, when andriod was a separate and very special wffort by Dave.

spike7638, 2021-09-11:
Sorry for the long delay in following up --- I went sailing for a while. I'll happily check whether things work in the most recent version of OpenCPN for Android (or wait a while until they do); I do still have that tablet as a workaround. :)  Thanks for the quick turnaround.

nohal, 2024-02-17:
Receiving UDP broadcasts works fine on current Android for me

closed 2024-02-17 by nohal