OpenCPN/OpenCPN #2334: Dashboard element "True Wind Direction and Speed" should use heading if available

pgrawehr, 2021-08-13:
If the boat heading is available, this should be used instead of COG for calculating and displaying the true wind direction. Basically, the heading has to be added to the relative direction of the true wind. This will make this plot valid even if the ship is moored. 

[image]

The picture shows a typical situation: The ship is moored, heading is about 290°. Apparent Wind direction is 21° from port. The true, geographic wind direction however is wrong. It should be 269° (West).

bdbcat, 2021-08-13:
Please provide a link to a VDR recording of your datastreams for tests.
Thanks
Dave

pgrawehr, 2021-08-13:
Here are two VDR recordings. I have found that due to a bug in my message router, OpenCPN only got HDM messages, not HDT or HDG. But fixing that didn't help. I'm attaching both logs anyway. Note that the orientation of the vessel on the map is correct and stable (not like what would happen if using COG). The "True compass" gadget also works correctly. 

[vdr-2021-08-13-2.txt](https://github.com/OpenCPN/OpenCPN/files/6984748/vdr-2021-08-13-2.txt)
[vdr-2021-08-13.txt](https://github.com/OpenCPN/OpenCPN/files/6984749/vdr-2021-08-13.txt)

Hakansv, 2021-08-13:
pgrawehr..
Sorry to interrupt but in your case is the wind direction, with respect to north, is not calculated by OCPN but trust incoming sensor data. If there are instruments providing data O is honoring that. The calculation is only used in the absence of data.
For example this MWD (NMEA 0183 standard Wind Direction and Speed, with respect to north, from your VDR file.)
$YDMWD,336.8,T,333.8,M,21.6,N,11.1,M*58
I agree if heading is 292° and apparent wind is 22° from port the wind direction with respect to north would be 270°.
So it may be you instrument providing the MWD message is not correct calibrated? (Such a sensor would BTW not be feasible on a moving boat. It would be stable mounted pointing to north?)
What if you exclude the MWD sentence from your connection where the YDMWD is originating?  Use the input filter "Ignore sentences". If no MWD is present O Dashboard will calculate the wind direction if a heading is present.

pgrawehr, 2021-08-14:
@Hakansv 

Good catch! I didn't even know this message was available (it's not really described anywhere). It does seem that I need to check why the sender doesn't use the Heading and where the data is coming from. My main navigation display shows the correct value. 

Nevertheless, filtering the MWD message for OpenCPN doesn't help. The wind direction gadget now just shows nothing at all.

pgrawehr, 2021-08-14:
I'm now sending an updated and fixed MWD sequence. This fixes the display. However, this should at least be better documented and checked, why the automatic calculation in case of a missing MWD is not working.

Hakansv, 2021-08-14:
Automatic calculation..
Hmm... What OCPN version are you using?
My bad I suppose. If I'm correct the calculation of wind direction, with respect to north, in the case of missing sensor data is introduced in current master only and will be available in next O release. 

Good you found a way to fix the sensor data.
Håkan

rgleason, 2021-08-14:
`However, this should at least be better documented and checked, why the automatic calculation in case of a missing MWD is not working.`

The User Manual needs updating. This is normally done with the help of Users. 
Please suggest where changes should occur in the manual and what should be changed.
I can also make you an editor.  PM me from cruisersforum  rgleason

pgrawehr, 2021-08-14:
@rgleason The most obvious missing thing is clearly the definition of the MWD message [here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:nmea_sentences). That the automatic calculation has been added on master I couldn't guess. Not sure I have much time for this, but I might try updating some docs. Which forum do you mean? (Sorry, stupid question).

rgleason, 2021-08-14:
@pgrawehr   https://www.cruisersforum.com/forums/f134/
If you PM me at rgleason.

bdbcat, 2021-08-14:
I think MWD is a strange beast.
If it is reported by equipment on a moving platform, the data must surely be modified by the speed and course(or heading) of the platform, if it is to be accurate.  But there is no evidence that this is happening, generally.
A quick test would be:  What is the reported wind speed in MWD when running downwind?
I suspect that in many cases, MWD is simply "apparent wind".  OCPN has no way of knowing wheher MWD data is compensated externally (by the equipment, supposedly)for platform velocity vector.  So we just report what we see.

@pgawehr:
"I'm now sending an updated and fixed MWD sequence."
How did you do that?

pgrawehr, 2021-08-14:
@bdbcat I have written part of the navigation software (particularly the NMEA0183 message router) myself. It's running on a Raspberry Pi and takes data from the ship's sensors (via an NMEA2000 to NMEA0183 bridge), a GPS receiver, an IMU and different other sensors and mangles them together. Data is then forwarded to (among others) a network socket for OCPN. 

Previously, as hakansv found above, the MWD sequence was directly coming from the NMEA0183 bridge, which should do the correct compensation, but seems to get it wrong. Now I'm just recreating that message (since I have all the information for it). Of course, since nobody seems to have the official NMEA specification (it's horribly expensive) we're now guessing what the directions within the MWD sequence should be (ship-relative or absolute). OCPN is apparently assuming the later, so when the angles are absolute, the gadget gets it right.

Hakansv, 2021-08-14:
re: "I think MWD is a strange beast."
I do agree! The MWD sensor is nothing to use on a boat. It would rather be meant for land based station?

Hakansv, 2021-08-14:
@pgrawehr ..
This is how we use MWD in OCPN-Dashboard:
https://github.com/OpenCPN/OpenCPN/blob/master/plugins/dashboard_pi/src/nmea0183/mwd.cpp
MWD -- with respect to north
MWV  -- in relation to the vessel's bow/centerline.

pgrawehr, 2021-08-16:
This all makes sense now, I was just not aware of the MWD message. (I do think it makes sense, though. MWV has been used for a long time, even when only STW was known, not even COG). And going from MWV to MWD is easy, if HDG is known. And having the wind direction with respect to north is nice, particularly when sitting in the cabin, working on your laptop and waiting for the weather to improve (as I do now...).

bdbcat, 2021-08-16:
"And going from MWV to MWD is easy, if HDG is known"...
Just to be thorough, if the platform is moving, also need to do the trigonometry and integrate SOG as well.  Also, when moving, COG should be considered as well..

pgrawehr, 2021-08-16:
@bdbcat True, but that's typically done already in the step when going from apparent wind direction and speed to true (boat-relative) wind direction and speed. Historically, STW was used for this, but when COG and SOG are known, that of course yields a better result for true wind, because there will be no added error from current or drift. 

So in pseudo code it's something like:
```
If (HaveApparentWindDirection && HaveApparentWindSpeed)
{
    If (HaveCOG && HaveSOG)
    {
        CalculateTrueRelativeWindDirectionAndSpeedUsingCOGandSOG();
    }
    else if (HaveSTW)
    {
        CalculateTrueRelativeWindDirectionAndSpeedUsingSTW();
    }
    else 
    {
        return;
    }
    if (HaveHeading)
    {
         CalculateTrueWindDirectionFromHeading();
    }
    else if (UseCOGasHeading)
    {
         CalculateTrueWindDirectionFromCOG();
    }
}
```

pgrawehr, 2021-08-16:
@rgleason I'll try again later, but the forum blocked me as spammer because I was trying to send you my E-Mail (not even as link!).

andyrbarrow, 2021-08-16:
I'm a lurker here, but I did want to weigh-in on the note regarding the use of SOG or STW as a means of calculating True Wind. 

There is considerable debate on which measure is appropriate for this calculation. I don't intend to get into that debate, but I would suggest that rather than implementing code that immediately uses SOG/COG and then falls back to STW, that the user be given the option of using either measure as the base for calculation. One could even provide an option for a fall-back if the other measure isn't available.

Thanks all for your hard work.

Andy

pgrawehr, 2021-08-16:
@andyrbarrow Interesting input. Do you have a link to such a debate? I was quite certain that SOG/COG would always give better results, but apparently not so.

andyrbarrow, 2021-08-16:
Again, I'm not intending to get into a debate on this, it's just that there are sailors who believe (rightly or wrongly) that True Wind should be calculated based on the moving table of water and not the speed over ground. I'll add I'm not one of them! My suggestion was that they be accommodated.

Ben Ellison refers to the debate here https://panbo.com/the-equinox-celestial-mechanics-pesky-true-wind/.

bdbcat, 2021-08-16:
@pgrawehr 
Your pseudo-code is more-or-less exactly what the unreleased OCPN 5.5 Dashboard does, so I agree with your logic.  
That is to say, when OCPN 5.6 is released (coming sometime this fall), then there will be no need for your external system to synthesize a corrected MWD anymore.  In fact, MWD is best ignored altogether in future.
OCPN does not use STW for this calculation.  Another debate....

pgrawehr, 2021-08-17:
@bdbcat I have tested the current master, and the true wind display really seems to work, even if I filter out the MWD message. However, there's now something fishy with the apparent wind display. 

The picture shows V5.5 (above) and V5.24 (below) with the exact same UDP data stream. 
[image]

For some reason, the true wind direction in the apparent wind gauge now also seems to take the compass course into account. I'm sitting in port and the wind is coming directly from the bow. It is intermittent, as sometimes the blue arrow momentarily disappears and the the number in the bottom left corner changes to (almost) zero, which would be correct.

bdbcat, 2021-08-17:
Thanks for the test.
I'll take a look at the anomaly.
Have not thought about this too much before now, but I am now wondering what exactly the "App and True Wind Angle" instrument should report for true wind direction.  That is, why should we superpose a "true wind direction" on a graphic of a ship, anyway.  The graphic only makes sense if the HDG is north. Confusing to me at first glance.
Your thoughts and logic?

pgrawehr, 2021-08-17:
@bdbcat That's actually a quite common display type (professional sail instruments show that, too). You often have apparent wind direction and true wind direction in the same instrument. The true wind angle (maybe better call it "true wind relative to ship angle", as opposed to "true wind direction") is the wind angle relative to the ship from which the true wind comes (and so also the waves). If you're moored, the two are equal, otherwise the true wind is always aft of the apparent wind (because it does not include the ship speed). 

Here's a picture of my sail instrument while underway (AWS: Apparent Wind Speed; AWA: Apparent Wind Angle; TWS: True Wind Speed; TWA: True Wind Angle)

[image]

bdbcat, 2021-08-17:
So I see the point: this is  "true wind relative to ship angle".  So we call this as "True Wind Angle", as distinct from "True Wind Direction".
I have never sailed with such an instrument.  I suppose in the good old days I must have been doing some trig in my head.  Anyway, I generally could figure out the "true wind angle" by looking at the waves.  But I can see the advantages of a precise indication when performance tweaking is important.

Hakansv, 2021-08-17:
re: "So I see the point: this is "true wind relative to ship angle". So we call this as "True Wind Angle", as distinct from "True Wind Direction"."
Yeas, that's the way our Dashboard have differentiated between wind angle with respect to North, "Direction" 0-360°, and wind angle relative to the bow, "Angle" +- 0-180°.
The instrument "App and True Wind Angle" I've used all time since it showed up years ago.

rgleason, 2021-08-18:
Tactics Plugin Documentation is an excellent source for this kind of information.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:sailing:tactics

bdbcat, 2021-08-18:
Corrected in github now.
Please confirm.
Thanks

pgrawehr, 2021-08-18:
@bdbcat Yes, looking good now. There's still a slight difference between the arrows sometimes, as the true wind arrow kind of follows the apparent wind arrow with one step delay. This is acceptable from my point of view. A more helpful improvement would probably be some smoothing (averaging over a certain time period).

bdbcat, 2021-08-28:
Close now?

pgrawehr, 2021-08-28:
Yes, this can be closed as "already works in the next release" and "has a workaround in the current". Implementing some kind of low-pass filter for the wind speed and direction might be worth a try, but is not scope of this issue.

closed 2021-08-28 by pgrawehr