OpenCPN/OpenCPN #2349: Update to Tides - Harmonics - from David Burch

rgleason, 2021-08-21:
David wrote:

```
So here is the issue as i see it.

The new harmonics are much better for US waters especially for the currents.  The  latest package of these harmonics from xtide ( a tcd file) will in fact load and run in OpenCPN. The tides are fine, but the current data in the harmonics now includes data for multiple depths. When there are multiple depths, which is true for most stations, some in AK are exceptions, then opencpn defaults to the deepest one and then does not show an arrow for it.  there is icon there and the details report are correct, but the arrows are not on the picture… and the report is for the deepest values, sometimes over 300 ft deep!

I am not sure how much work it is to address that. seems it should default to the shallowest one… at least in the vast majority of cases, and then show the arrows.

I am not sure yet what do to with our students. i could show them how to update to this one, which would make improved tides but the currents are not displayed well and when there are deep data is is often wrong…. even with the proper data sitting just a line or two above it in the file.

Another virtue of tuning up to show the new harmonics is these new ones put the current stations in the right location, where the old one in the version being shipped use the old noaa descriptions which were rounded to the whole mile so many are way off actual location.

sorry i do not have a better report at the moment.

—david

the file i sent you is the right file to use… it does however not include all the european and AU and other international tide stations.  Seems ideal world would be to fix the display to show the new US ones and then let users decide what they want to install.  without changes, i guess your default set might meet the needs of most of the users…. however, those data have to be checked as well.

seems harmonics issues never go away!

```
Please remove the   ".log"

[harmonics-dwf-20210110-free.tcd.log](https://github.com/OpenCPN/OpenCPN/files/7026287/harmonics-dwf-20210110-free.tcd.log)

bdbcat, 2021-08-29:
Rick/David...
I have tested the new .tcd file, and it works fine. 
The main issue is that the new file duplicates some stations present in the default (.idx) file, primarily in the US.  Conversely, the new .tcd file does not contain some important international stations.
What needs to be done is to manually edit the existing default .IDX file and remove (or comment away with '#') all stations that are also present in the new .tcd file. Thus, when complete, the default .IDX file will cover non-US stations, and the new .tcd file will cover all US stations.  The two files will then be renamed appropriately, distributed with OCPN, and installed by default.

To make this job easier, I will modify OCPN locally to detect and print the names of duplicate tide stations found when both datasets are loaded. Using this printed list of duplicated station names, manual editing of the .IDX file will be manageable.

The question of current at depth display is separate, will be considered independently.

bdbcat, 2021-08-29:
Here is a list of duplicated Tide stations:
file:///home/dsr/Projects/opencpn/build/TideStationDuplicates
Any takers on the editing job?

bdbcat, 2021-08-29:
[TideStationDuplicates.txt](https://github.com/OpenCPN/OpenCPN/files/7071805/TideStationDuplicates.txt)

This is the list of possible duplicated tide stations.

Any takers on the .IDX edit job?

rgleason, 2021-08-29:
I think BeeFree might be willing. I will ask. He was very good with polar files. 
Right now I am buried in pllugin editing.

bdbcat, 2021-09-04:
Rick...
This is all done now, you will see it in a coming commit.
Thanks
Dave

rgleason, 2021-09-04:
Thank you. Sorry, lost track of thiz.

closed 2021-09-04 by rgleason