OpenCPN/OpenCPN #2350: Build

olivinesguerra, 2021-08-23:
Is there any tutorial on how to run the app? or test? We do have plugin to make that connects to this project?

Thanks

transmitterdan, 2021-08-23:
There is a large user manual at opencpn.org.

There is no tutorial on testing.  For the app to be somewhat useful it requires external data such as NMEA messages containing geolocation data (GPS for example).  There are tools for artificially generating such data and they can be useful for testing some plugins.

rgleason, 2021-09-28:
Website
https://opencpn.org/

Wiki
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual

Development Manual
https://opencpn-manuals.github.io/main/ocpn-dev-manual/5.3.1/intro-AboutThisManual.html

Plugin Manual
https://opencpn-manuals.github.io/plugins/opencpn-plugins/0.1/index.html

Cruiser's Forum is a good place to ask this kind of question
https://www.cruisersforum.com/forums/f134/

I think this can be closed now. Closing

closed 2021-09-28 by rgleason