OpenCPN/OpenCPN #2353: Convert incoming xxWPL and xxRTE to waypoints and routes

pgrawehr, 2021-08-28:
I would like to see the possibility to convert incoming xxWPL and xxRTE sequences to OpenCPN routes and waypoints. I'm using another navigation system for main navigation, but it would be nice if OpenCPN catched up with the waypoints and route the (or an) external navigation system is sending. This would also allow to seamlessly switch over to using the route from OpenCPN in case the other nav system fails. I don't want to have to enter the route twice.

We currently have the option under AIS settings to "Treat WPL sentences as APRS position reports", which works and creates APRS (AIS) targets at all waypoints, but that's not really what I need here, as these are more confusing than useful. 

Before I (try to) implement this feature, I would like some more opinions.

nohal, 2021-08-28:
If you are not using xxWPL for APRS position reporting, where they keep coming very frequently and the target usually moves, then of course the existing feature may not be what you are looking for.
I see absolutely no problem implementing what you suggest, if the behavior stays configurable. The only reason it does not exist is simply that not many of us use OpenCPN as something else than the primary source of waypoint and route information so implementing the reception from NMEA sources never got priority.

antipole2, 2021-10-02:
You could do this with the JavaScript plugin.
I use it for the reverse - sending OpenCPN routes and waypoints out as xxWPL and xxRTE sequences to devices running iNavX.

rgleason, 2021-10-02:
antipole can you help us with respect to OpenCPN interoperabilty and perhaps make some suggestions and share some experience here in the  [Interoperability between Opencpn(s) and other devices](https://www.cruisersforum.com/forums/f134/interoperability-between-opencpn-s-and-devices-256169.html) Cruiser Forum thread?  Thanks

antipole2, 2021-10-02:
@rgleason I am poised to release JavaScript v0.4.  As soon as that is out, I will contribute to this discussion.

 I have this summer completed a six-week cruise during which iPhones and an iPad running iNavX were all kept in good sync with OpenCPN.
The latest scripts I was using ran under JS v0.4 and want to avoid issues with backward compatibility to v0.3.