OpenCPN/OpenCPN #2364: AIS Target list in night mode

rszemeti, 2021-09-07:
The colour scheme of the AIS target list in night mode is OK, but it appears to give a very bright "flash" before displaying in night colours ... it might be that the dialogue box is being dislayed in default colours then having its colours modified after it has been shown on the screen,  setting the colours prior to showing the box would (should?) solve this.

rszemeti, 2021-09-11:
OpenCPN 5.2.4, Raspbian.

bc38, 2021-09-13:
Windows?
https://www.cruisersforum.com/forums/f134/opencpn-night-mode-is-not-really-a-night-156580-2.html#post1973256

rszemeti, 2021-09-13:
If you look at the previous comment, it says "OpenCPN 5.2.4, Raspbian."

bdbcat, 2021-09-19:
Unfortunately, this "flashing" seems to be inherent in all wxWidgets gtk ports.  We have looked for a solution, nothing found yet.
The problem affects other pop-up dialogs, as well, not just AISTargetList.

Let us leave this issue open as a reminder of a persistent problem.
Sorry Dave

rszemeti, 2021-09-19:
Noted and thanks for looking.  It's not a show stopper, but it does give
your night vision a bit of a kick.  The bright white right-click menu is
probably a worse offender, hopefully that is simpler and just a color
palette thing.

Maybe the issue can be raised in the wxWidgets/gtk GIT repos?