OpenCPN/OpenCPN #2365: Default colours for console legend/ console values

rszemeti, 2021-09-07:
The colours as default are close to black and invisible under normal conditions,  there are multiple questions on forums with users confused why they can't see track/waypoint info in the console when a route is active, and it is always that they need to manually adjust the colours to something visible.  If a more visible set of default colours for the console could be set in the next release, that would solve this.

leamas, 2021-09-07:
There is one question to all your three issues filed today: What's your platform (Windows/Linux/MacOS/Rasbian)?

rszemeti, 2021-09-07:
Raspbian,  5.2.4 from the openplotter distro

bc38, 2021-09-07:
Will there be a refinement of the night mode for windows and the context menu under Windows? There have been no changes in this direction since 15 years!
https://www.cruisersforum.com/forums/f134/opencpn-night-mode-is-not-really-a-night-156580-2.html#post1973256

commit ef262763 (2021-09-12): Improve default console legend/value font colors. Closes #2365
closed 2021-09-12 by bdbcat