OpenCPN/OpenCPN #2366: clang-format configuration is too new

leamas, 2021-09-07:
The checked in  clang-format file  leads to some problems as described in #2360. There is also reports (douwe) that it breaks his Visual Studio setup.

We should replace the current version which is for clang-format vers 1.12, going back to at least 1.11 which is the version in Debian stable.  I plan to make a PR  as I write this.  If we need to go to an older version we need to check that it does produce the same formatting as 1.12.

nohal, 2021-09-07:
We should also perhaps tell people to use a git hook to format their code before submitting automagically, my `.git/hooks/pre-commit` looks like this for example

```
#!/bin/bash

# find out where clang-format is
exe=$(which clang-format)

if [ -n "$exe" ]
then
  # set the field separator to new line
  IFS=$'\n'

  for line in $(git status -s)
  do
    # if the file is added or modified
    if [[ $line == A* || $line == M* ]]
    then
      # and the file is a c or cpp file
      if [[ $line == *.c || $line == *.cc || $line == *.h || $line == *.cpp ]]
      then
        # format the file
        clang-format -i -style=file $(pwd)/${line:3}

        # and then add the file (so that any formatting changes get committed)
        git add $(pwd)/${line:3}
      fi
    fi
  done
else
  echo "WARNING: clang-format was not found, cannot format files!"
fi
```

leamas, 2021-09-07:
Yes, we should. However, I think that hook could need some polish: Using the modern _ diff filter_ one can write things like

```
for f in $(git diff --cached --name-only --diff-filter=ACMR); do
    stuff; 
done
```
which is cleaner. Here is also the question on formatting in a hook vs checking if  it is formatted, fail if not.

leamas, 2021-09-07:
.clang-format is downgraded to version 11 in #2367.  Leaving bug open for some time if there is a need for an even older version, and it's feasible.

leamas, 2021-09-21:
Nothing more here. No news is good news. Closing

closed 2021-09-21 by leamas