OpenCPN/OpenCPN #2370: ci: macos: broken builds.

leamas, 2021-09-10:
Since some time the MacOS CI builds fails.

nohal, 2021-09-10:
```
In file included from /Users/distiller/project/libs/wxsvg/src/SVGDocument.cpp:17:
/Users/distiller/project/libs/wxsvg/src/cairo/SVGCanvasCairo.h:15:10: fatal error: 'cairo/cairo.h' file not found
#include <cairo/cairo.h>
         ^~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [libs/wxsvg/CMakeFiles/WXSVG.dir/src/SVGDocument.cpp.o] Error 1
make[1]: *** [libs/wxsvg/CMakeFiles/WXSVG.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

but Cairo is installed and found by `cmake`, so to start with we should probably switch to use the verbose build in CI environments to see what actually happens.

leamas, 2021-09-10:
I have actually a patch in the oven... stay tuned (need to to other tings now, probably for the rest of the day).

leamas, 2021-09-10:
CircleCI has this killer feature that you can log in to the build box using ssh, which makes it possible to hunt down these things quite fast. The build log is [here](https://app.circleci.com/pipelines/github/leamas/OpenCPN/1528/workflows/34f91c7b-dc4c-49e8-86bf-0fd246235387/jobs/2696)

bdbcat, 2021-09-10:
I am in active hack/patch mode on Mac build.  Expect failures for a little bit.
I use ssh login a lot....

leamas, 2021-09-10:
anyway. #2372  is a patch which makes it build for now...

PR #2372 (2021-09-10): ci: MacOS: Handle cairo library update (#2370)
#2370 was caused by the combination of an hardcoded library path and  an updated version. Don't hardcode that path to avoid future problems.

Closes: #2370
closed 2021-09-10 by nohal