OpenCPN/OpenCPN #2373: 3 clicks to switch out of Night Mode

rszemeti, 2021-09-11:
OpenCPN 5.2.4, Raspbian.

Entering night mode needs just 1 click on the "moon" icon ... leaving night mode needs 3 more clicks (with appropriate waits for the screen to reconfigure)  .. its as if it is a 4 state switch and 0 is day mode, 1,2 and 3 states are all night mode.

nohal, 2021-09-11:
It does have 4 states as there are 4 color schemes defined by the ECDIS standard, each having a different color palette: DAY, DUSK, NIGHT and RGB.

closed 2021-09-11 by nohal

rszemeti, 2021-09-11:
On OpenCPN they appear to be Day, Night, Night, Night ... I can detect no
difference between the 3 "non day" states on my display ...

nohal, 2021-09-11:
I would say the difference is clearly visible on my systems
[image]
[image]

nohal, 2021-09-11:
And the behavior seems totally logical to me as it changes from bright to darker to darkest to darker to bright, which is exactly what I need when sailing overnight.

rszemeti, 2021-09-11:
Well the fact that I, as a user could detect no change should tell you
something ... maybe it is only on Raspbian/Pi?   I'll have a closer look
when I am next at the boat and see if I can detect any differences, but if
there are any, they are very subtle.

transmitterdan, 2021-09-11:
It's only your display or eyes. There are definitely 4 modes. And none are the same. It is easier to see the differences when there is little or no light. Possibly your display does not have sufficient contrast.

bdbcat, 2021-09-12:
Also depends on the chart.  Some chart types do not have embedded color palettes, so they will indeed show no differences.
Perhaps some screenshots would help....

rszemeti, 2021-09-12:
I'm using the oeSENC charts for the UK, I can try screenshots when I am
back at the boat, which as we have just got back after a month of sailing
may not be for a couple of weeks.