OpenCPN/OpenCPN #2376: ci: Debian: broken.

leamas, 2021-09-12:
The ci debian builds are broken, stops at cmake error in plugins/chartdldr_pi/CmakeLists.txt line 159.

This is related to recent work for an overhaul of the MacOS builds. Still, we need IMHO a tracker bug for this.

leamas, 2021-09-12:
Right now, master Debian builds with this patch (MacOS still broken)

```
index d8f6c5ab6..6bd550030 100644
--- a/plugins/chartdldr_pi/CMakeLists.txt
+++ b/plugins/chartdldr_pi/CMakeLists.txt
@@ -134,6 +134,8 @@ ENDIF()
 
 ADD_DEFINITIONS(-D_FILE_OFFSET_BITS=64)
 
+ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC_UNARR} ${SRC_CHARTDLDR})
+
 if (QT_ANDROID)
   target_include_directories(${PACKAGE_NAME}
     PRIVATE ${CMAKE_SOURCE_DIR}/buildandroid/libarchive/include)
@@ -189,7 +191,6 @@ IF(QT_ANDROID)
   ADD_DEFINITIONS("-Wno-inconsistent-missing-override -Wno-potentially-evaluated-expression")
 ENDIF(QT_ANDROID)
 
-ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC_UNARR} ${SRC_CHARTDLDR})
 
 if ("${CMAKE_CXX_COMPILER_ID}" MATCHES "Clang|AppleClang")
     # FIXME: This is to broad, needs better handling after 5.0.0

```

bdbcat, 2021-09-13:
Debian should build without patch now.
MacOS is still WIP.

bdbcat, 2021-09-14:
MacOS build under control now, builds correctly.
Does not use CCI cache for new macports modifications.  So, build takes a ...long... time.   CCI refuses to restore a cache image to /opt, which is where macports lives and does its work.  Investigating workarounds...

leamas, 2021-09-14:
Symlink /opt/local  elsewhere?

Anyway, the fact that builds are slow does not qualify them as "broken", so this issue should be closed. 

If someone could give #2378  some love we would  have the CI builds running again :)