OpenCPN/OpenCPN #2380: RFE: Better integration of Grib plugin with Pat WinLink of raspberry pi needed

mgrouch, 2021-09-15:
RFE: Better integration of Grib plugin with Pat WinLink of raspberry pi needed.

New option is needed to send email via pat to WinLink with a grib request.

Thanks,
--MG

rgleason, 2021-09-15:
https://www.winlink.org/   I have an call numbers and  account that I use occasionally on other's radios at sea.
This might be very useful, but it should d work with other OS too.  I know WinLink, what is PAT?
Can you point to the necessary resources or is this some un researched request?

mgrouch, 2021-09-15:
It’s only WinLink client that
runs on pi afaik. 

https://getpat.io/

rgleason, 2021-09-15:
So what exactly do you want?  Winlink already supports attachment of small grib files.

mgrouch, 2021-09-16:
User needs to see WinLink as an option. And OpenCPN grib plugin needs to prepare content of the request for grib for WinLink and address where it needs to be sent. and possibly send it too my invoking pat command. 

Currently the option is SailDoc only.

rgleason, 2021-09-16:
Grib_pi  menu   "Write and Send Email Request"  currently has  saildocs and zygrib selections.
Add  winlink selection with an entry form for winlink

also make sure that the limitations on size etc are correct for Winlink as they quite restrictive.
Do you know where the necessary winlink information is found?

Also Winlink is not a grib service, just an excellent radio based email system, therefore some other grib service will be needed for the gribs.

Can't you just enter your Winlink email under saildocs?
I believe I might have done that during a race some time ago, but perhaps I simply made saildoc requests through WinLink!

So what is it that you really wnat?
Maybe this should be closed.

mgrouch, 2021-09-16:
Is Grib plugin usable with WinLink as-is?

nohal, 2021-09-16:
> Is Grib plugin usable with WinLink as-is?

You obviously think it is not, so be so kind and try to describe exactly what you demand us to implement so that it was.

rgleason, 2021-09-16:
I used grib_pi formatted emails and simply entered my winlink express email address and it works for only small files but they are big enough to get at least 3-4 days.  I forget the exact size limits!  I did that many times during the race.

You should try it and close this. IMHO

rgleason, 2021-09-18:
Time to close and if there is something more specific, just reopen.

rgleason, 2021-09-18:
@mgrouch 

Happened upon this
https://weather.mailasail.com/Franks-Weather/Receiving-Saildocs-Gribs-Over-Hf-Radio

rgleason, 2021-09-18:
From Frank's website

"Look at the Grib Parameters section. Check the approx file size - make sure it's less than 10K so the file is not too big for SailMail. (The SailMail server will chop it off if over 10K.)"

Time to close this.

closed 2021-09-18 by rgleason

mgrouch, 2021-09-18:
> @mgrouch
> 
> Happened upon this
> https://weather.mailasail.com/Franks-Weather/Receiving-Saildocs-Gribs-Over-Hf-Radio

There is nothing about Raspberry Pi and pat and WinLink in this link. 

It mentions Sailmail, AirMail. 
AirMail is not available on Pi. 

This JIRA was about WinLink and pat support on raspberry pi. 

Thanks

rgleason, 2021-09-18:
I explains how to receive gribs through HF radio.
Isn't that your interest??    I had to be sure they were less than 10k and used what I call "walking or moving gribs"
I believe I prepared a walking grib format using Grib_pi,,  then copied those parameters into WinLink and issued the request hoping the size did not get bigger than the estimated size.  The attachment then camee back eventually as an email and the attachment was  downloaded slowly to a grib directory which was then used by grib_pi to do the routing.

nohal, 2021-09-20:
@mgrouch please take the time you spend complaining about people wrongly refusing your demands on explaining what the functionality you demand to be implemented actually is.
It would be nice if everybody knew what `pat` is and what is missing to "support" it, but unfortunately it is not the case and I'm afraid the vague specification you provided has very little potential to generate any developer interest.
Of course unless there is a developer having the same unknown to us problem you have, in which case we would probably already see a pull request implementing the missing functionality though.

mgrouch, 2021-09-20:
There was no complaint. 
It was just proposed improvement.

nohal, 2021-09-20:
Well, while I am absolutely sure I can write that code, I have absolutely no idea what it should be doing or where to try to find out what it should be doing if you are not going to explain it.
And no, looking at https://getpat.io and even reading through https://github.com/la5nta/pat/wiki/The-command-line-interface did not give me any clue what it is you want us to do.
So even after wasting all this time here, you got nowhere closer to having that stuff implemented. Please try to consider it when you ask for stuff to be implemented next time.

rgleason, 2021-09-20:
Pavel, I think Mike has a wicked sense of humor   m"grouch"  (:-))