OpenCPN/OpenCPN #2382: Filter AIVDO sentence

rgleason, 2021-09-17:
https://www.cruisersforum.com/forums/f134/aivdm-aivdo-and-own-ship-gps-data-255576.html#post3483969

`In any case, I think that the AIVDO sentence should always be filtered 
without the need for the user to configure it. 
I think it is quite useless information, and as we are seeing, it is also annoying.`

bdbcat, 2021-09-18:
There is good reason NOT to always filter AIVDO.
Some users prefer to use AIVDO, instead of some other GPS receiver's messages.  They may be using an older receiver with low sensitivity, slow capture, etc.
The way I see it is this:  Whether you know it or not, if you have an AIS transponder, then you have two GPS receivers.  Only you (the user) can know which is to be preferred in your own environment.

rgleason, 2021-09-18:
Thanks.  Will add to docs.

closed 2021-09-18 by rgleason

rgleason, 2021-09-18:
Dave It might be worthwhile to consider what tehani reports in this thread.

https://www.cruisersforum.com/forums/showthread.php?t=255576

rgleason, 2021-09-21:
Related PR about bidirectional devices   https://github.com/OpenCPN/OpenCPN/pull/1476