OpenCPN/OpenCPN #2383: Polar plugin icon appears many times and keeps multiplying in plugin toolbar when Polar plugin is enabled

mgrouch, 2021-09-18:
Polar plugin icon appears many times and keeps multiplying in plugin toolbar when Polar plugin is enabled.

I had it on raspberry pi. Not sure about other set ups.

Thanks,
--MG

nohal, 2021-09-18:
There is no polar plugin being part of OpenCPN, if a 3rd party plugin erroneously asks OpenCPN to add icons to the toolbar multiple times, it is an error in that plugin, not in OpenCPN.

closed 2021-09-18 by nohal

mgrouch, 2021-09-18:
OpenCPN shouldn’t allow a plugin to do that. It has to protect itself from malfunctioning plugins. 
Or I’m wrong?

nohal, 2021-09-18:
Yes, you are, OpenCPN can't know why a plugin calls the API and whether the author of the plugin is serious about it. If you found a broken 3rd party plugin, please open an issue for it where it belongs.

mgrouch, 2021-09-18:
I think it’s also OpenCPN issue. Plugin has to have one icon. 
It’s a contract which OpenCPN has to enforce.

mgrouch, 2021-09-18:
I think this issue needs to be reopened to get a chance to be discussed. 
thanks

nohal, 2021-09-18:
Please open an issue where it belongs. And it does not belong here.
If you have a problem with the OpenCPN API design, open an issue for that, preferably specifying a real design problem it has and if possible outlining a design solution for it.
I will certainly not reopen this issue as it is not an OpenCPN issue at all, but, if an issue at all, an issue in a 3rd party plugin.

nohal, 2021-09-18:
And before we start to go in circles, no, a plugin does not have to have one icon, look at the VDR plugin, for example.

rgleason, 2021-09-18:
This pi needs some work. Report in issues. Polar_pi Thanks.