OpenCPN/OpenCPN #2396: Update  wxWidgets [post 5.6.0]

leamas, 2021-09-25:
During the next cycle, we IMHO need to investigate paths to update wxWidgets to 3.1.5 The blocker here is that there are no updated packages for Debian and hence for Ubuntu. Furthermore, there will not be any such packages until the ABI-stable 3.2 is released.

Looking at the wxWidgets project I see a real risk that we won't have any new version available even for next release in a year or so. There is almost no activity in the [remaining bugs](https://trac.wxwidgets.org/roadmap), and the overall impression is that the project has stalled.

Given all these factors I think we should investigate the route to create a Debian wxWidgets 3.1.5 package and use that in the Linux builds. We must then be extremely careful with applying any patches besides those already present in the upstream wxWidgets project.

Such a package could be published for Ubuntu in  a PPA. For Debian, we could build amd64 binary packages and distribute them using for example Cloudsmith. With this package in place, updating wxWidgets for other platforms should be reasonably simple since we here already build from source. It will also be usable in the plugin build pipelines.

The best point to do such a move is probably as soon as 5.6.0 is released.

transmitterdan, 2021-09-25:
Can we build a non-shared version and link it directly?

What is the down side besides executable size?

rgleason, 2021-09-25:
Maybe you could build your own version of 3.2.0 and use it for all environment, or would that be a mess?

rgleason, 2021-09-25:
Alec when I look at wxWidgets github account things look better and there has been activity, so maybe that page is not up to date?

https://github.com/wxWidgets/wxWidgets/milestones

transmitterdan, 2021-09-26:
I have discussed some issues in the past with the lead wx developer. If you like I can query the status of 3.2.

There is a regular development going on in their repository.

nohal, 2021-09-26:
The list of pending issues in the wx Trac historically has had very little to do with when the release was announced. From what I see, they seem to be pretty close. Which of course could still mean another year.

leamas, 2021-09-26:
> Which of course could still mean another year.

Indeed. We were in a similar situation around the 5.2.4 release some 15 months ago: 3.0 is almost here. We need to plan for that 3.0 isn't available for next OpenCPN release.

IMHO, even if the 3.2 release will happen before our next release, we still will get  advantages by moving early instead of just waiting for the 3.2 Debian package (the packaging might introduce further delays):
  - We will have a common code base on all platforms again (at least MacOS has already moved to 3.1.5).
  - We will have more time to explore the new API to handle for example gestures (#2057) and HiDPI screens.
  - There is a window where we test 3.1.5/3.1.6 and if required can upstream patches to 3.2 (which, BTW, is how the development cycle is supposed to work).

There is also the official package to handle in this process; this is a more narrow path. I'm having a dialog in debian-devel about this.  We need the outcome from that before making more decisions.

leamas, 2021-09-26:
> Can we build a non-shared version and link it directly?

Possibly, but there is no need. If we make a reasonable 3.1.5 packaging based on the 3.0 package it can co-exist with 3.0. OpenCPN will run just fine anyway, the runtime linker will resolve the versioned library dependencies.

leamas, 2021-09-26:
> If you like I can query the status of 3.2.

This is a complicated mess, and all input is certainly valuable. Please do!

leamas, 2021-09-26:
The [debian-devel discussion](https://lists.debian.org/debian-devel/2021/09/msg00356.html) seems  finished. It basically boils down to a plan like this:

1. We make an internal wxWidgets 3.1.5 package which we use in the project as described [above](https://github.com/OpenCPN/OpenCPN/issues/2396#issue-1007113325).
2. If wxWidgets  3.2 is available when we are closing in to the next OpenCPN release we use that. Case closed.
3. If not, we approach the wxWidgets maintainer about releasing our 3.1.5 package into the experimental branch.  If he agrees, we use that package until wxWidgets 3.2 is available.
4. If the wxWidgets maintainer don't want to release a 3.1.5 package even into experimental, we bundle 3.1.5 (3.1.6?) in our OpenCPN package.

Alternatives 3) and 4) means that also OpenCPN will live in the experimental branch until we can make a proper release based on wxWidgets 3.2. Basically a pain, but probably less so than maintaining a fork in the sources, see the mailing list archives.

Any input from @transmitterdan  about hints for the schedule would certainly be helpful here...

tbultel, 2021-09-27:
My 2 cents. I recently logged a bug in wxWidgets bug tracker:  https://trac.wxwidgets.org/ticket/19265
Without a fix for that bug (I am working on it and should push a patch soon) , wxGesture do not work on systems that have touch-only devices.
What I point out, is that maybe being dependent on the 3.1.x coming release(s) is a big constraint.
So I would also vote for bundling the wxWidgets with OpenCPN, yes. That is is for the packaging point of view.
And for the daily build point of view, with not setting it as a submodule of the project ?

leamas, 2021-09-27:
Bundling wxWidgets in  a stable Debian release will not happen with me as maintainer even if I could persuade my sponsor. It means a multi-year undertaking to maintain a large library like wxWidgets and must IMHO be avoided at (almost) all costs.

The proper route here is to see what happens with the reported bug upstream. If/when there is a patch available in the upstream tree (and it thus will be part of wx3.2) , we could apply the patch to our provisionary  3.1.5 package, which might be bundled or not. We should not apply patches which are not accepted upstream, it means that we will have to maintain a independent fork which just is a bad thing to do.

In other words: What we shall do here is to keep ourselves in the wxWidgets 3.2 loop where we test by using 3.1.5, report problems and possibly also provides patches which are sent upstream for review. When patches are available upstream, we apply them ourselves and continues our testing. This is how this process is supposed to work, and I see no reason to not adhere to it.

leamas, 2021-09-27:
@tbultel : Looking at that bug you filed upstream would be happy to apply a patch according to your analysis. Providing that patch after testing it is the necessary next step in this process. If you can do that, please do it.

If not, please file a new issue here at OpenCPN to see if you can get some help with it. It's in our interest to get this sorted out. @transmitterdan  is our liaison officer at the wxWidgets project, he knows more about how the  project's workflows.

transmitterdan, 2021-09-27:
The wx maintainer has agreed to accept a patch for this issue once it is fully tested.

The wx3.2 release is not years away but maybe months. I think it is likely that when the next major release of O happens sometime in 2022 then wx3.2 will be already released. Obviously, nothing is certain when speaking about open source projects.

leamas, 2021-09-27:
@transmitterdan 

> The wx3.2 release is not years away but maybe months.

Right, thanks!!

All this said, do you agree about the overall plan to move to 3.1.5 early in next cycle instead of just waiting for the 3.2 Debian package?

transmitterdan, 2021-09-27:
I don't have any opinion as yet.

What OS support will we drop when adopting wx3.2?  I think the 3.2 release of wx will deprecate some older OS releases.

leamas, 2021-09-27:
> What OS support will we drop when adopting wx3.2? I think the 3.2 release of wx will deprecate some older OS releases.

This is indeed a good question... OTOH, we *will* have to adopt to 3.2, question is just when, and the related problems will need to be dealt with anyway. I cannot fathom that we will release OpenCPN 5.8.0 using wxWidgets 3.0 if 3.2 is available.

And since we will adopt 3.2, it's IMHO better to do it early so we can better engage in the bug fixing loop, and otherwise prepare ourselves to the new version.

Or?

leamas, 2021-09-27:
That said, a quick look at  the details:

  - The official Debian wxWidgets 3.2  will likely land in upcoming Bookworm, hopefully together with next OpenCPN (5.8.0?). For earlier releases we might need to backport to Bullseye. I havn't been through this process, plan to make a try when 5.6.0 is out.

  - Raspbian is basically Debian in this context. A backported OpenCPN would make it also here.

  - All Ubuntu versions should be possible to fix  **iff** we can  backport wxWidgets 3.2 to Xenial+ in the PPA repo and build against that. It's a question of wxWidgets dependencies. We should be able to explore these things once we have a wxWidgets 3.1.5 package available -- we could then try to rebuild it on older platforms.

  - Oher linuxes uses Flatpak where this isn't an issue at all (we bundle wxWidgets in the Flatpak builds anyway).

  - MacOS already uses 3.1.5 so no problems there.

  - Windows is supposed to be API stable (knock, knock...)

TL:DR:  as always, the old unmaintained Xenial might be problematic, but otherwise we should be fine with some work. And the alternative is more problems due to the ageing 3.0: touch screens, gtlk3, HiDPI and what not.

tbultel, 2021-09-27:
> The wx maintainer has agreed to accept a patch for this issue once it is fully tested.
>
> The wx3.2 release is not years away but maybe months. I think it is likely that when the next major release of [...]

I just made the pull request : https://github.com/wxWidgets/wxWidgets/pull/2539

leamas, 2021-09-28:
Let's now put this on ice until 5.6.0 is out of the doors.

leamas, 2021-10-08:
@tbultel : To start a workflow I have made some 3.1.5 packages which includes  the patch in your PR at https://github.com/wxWidgets/wxWidgets/pull/2539

Debian packages are available at https://cloudsmith.io/~alec-leamas/repos/wxwidgets/packages - only amd64, but there is a source package which can be rebuilt on other platforms.

Ubuntu packages (currently only Focal) are at https://launchpad.net/~leamas-alec/+archive/ubuntu/wxwidgets/

Hakansv, 2021-10-08:
OK, sorry :)

leamas, 2021-10-08:
@Hakansv said:

> Mind the caption for this issue.

Indeed. The issue here is about if/how/when updating to next version of wxWidgets 3.2.  Unless there is info on the Win XP support  in wx 3.2 this is not related to the need to update drop XP or not., so please don't hijack this bug for such discussions.

leamas, 2022-01-02:
Found a bug in the wxWidgets packaging which made the webview library unavailable in a typical install. Now fixed in 3.1.5+dfsg-2.1~ubuntu3, available in the [PPA](https://cloudsmith.io/~alec-leamas/repos/wxwidgets/packages)

rgleason, 2022-01-03:
Thats good. Maybe Petri's newer Dashboard-tactics will work.

leamas, 2022-01-24:
I have created opencpn builds using wxWidgets 3.1.5 for Focal and Flatpak. They can be used to test plugins to check differences compared to current 3.0.0. Details and how-to [here](https://github.com/OpenCPN/OpenCPN/pull/2506#issuecomment-1020189377)

Basic smoke test looks fine, at least some plugins loads and "seems" to work.

rgleason, 2022-01-24:
Alec, please see this question for Jon
https://github.com/jongough/testplugin_pi/issues/243

Does Testplugin need to change accordingly?  Or is this just a test?
If it is just a test, if it works, how soon will we need to change testplugins templates?

leamas, 2022-01-24:
Not now. This is something we will have to do, but in an orderly fashion.  The PR is about core OpenCPN, and it's certainly not accepted. As I see it, the core issue is rather when than if.

However, we can already now test the two versions (Focal/Flatpak) and, more important. we can test plugins against them.  See it as a preparations step. We will have to land at least o-charts and probably also OpenCPN 5.6.2 before  looking deeper into how to plan this.

That said, we are sooner or later facing an ABI break when moving to wxWidgets 3.2 (which this prepares us for). One aspect is to check if there are other things we might want to change since we have an ABI break anyway.

Another thing is next Ubuntu LTS release Jammy which is upon us in April - should we then use wxWidgets 3.0 or 3.1.5 (assuming 3.2 is yet not released at that point). We tried to support two versions on Bionic (gtk2 and gtk3). Lessons from that is that it was a bad idea,  better to make a clean break.

rgleason, 2022-01-25:
Thanks very much. Alec. I know that it difficult to keep all the parts working together and moving forward at the same rate. This does clarify what needs to be done with the TP pi.

jongough, 2022-01-28:
What in the xml/xsd distinguishes between wxWidgets 3.0 & wxWidgets 3.1/3.2 so that plugins can be built and installed appropriately? There is no wxWidgets information in the xml.

leamas, 2022-01-28:
Nothing, so far. We need to add -wx31 suffixed targets, a PR against OpenCON/plugins is required. On my TODO...

jongough, 2022-01-28:
How about using xml and creating a <build-wxWidgetsVersion></build-wxWidgetsVersion> rather than creating a convoluted extension to the <target></target> element?

leamas, 2022-01-28:
> How about using xml and creating a rather than creating a convoluted extension to the element?

?

jongough, 2022-01-28:
should have read, and did when I edited it, but I missed the 'ticks' to indicate `code`
How about using xml and creating a `<build-wxWidgetsVersion></build-wxWidgetsVersion>` (or something similar) rather than creating a convoluted extension to the `<target></target>` element? If you need to use the `<target></target>` value then make it a list of sub-values.

leamas, 2022-01-28:
We now have very simple semantics: plugin is compatible if the plugin abi matches host's abi. This is a single  string comparison. It's not just about keeping code simple, it's about overall simplicity.  That the string gets a suffix is a far smaller problem compared to  new XML elements

Any changes in this direction should thus IMHO be avoided if possible.

jongough, 2022-01-28:
Simple like '-gtk3' needed for ubuntu but not for debian in target?

So why use xml in the first place if plugin manager is going to go its own way. It may be simple for the person doing plugin manager but it is a complete mess when looking at it from the plugin perspective. It is obscure and now there is no link between the xsd and the plugin manger it makes it very difficult to work out what is going on. Dave informs me that there is no link between the xsd, xml and plugin manager and that anything can be put into the xsd as it is not used except for a cursory, often wrong, check of what is valid.

This is not a change in direction, but correcting a deviation from standards.

leamas, 2022-01-28:
The -gtk3 suffix is a IMHO a mistake. At the time I was against it,  it was Dave's decision.  We should drop it when next Ubunte LTS release Jamyy is upon us in April.

IT will be sorted out when we get wx-31 which actually makes sense. There will be n o debian-gtk3-wx31, there will be debian-wx31

That said, I cannot see that having a ABI specification in a single string is a problem, neither in plugins nor main OpenCPN. On  the contrary, it's a simple and a uniform way. What is the actual problem in the plugins?

jongough, 2022-01-28:
The ABI should have picked up that the wrong version of wxWidgets was in use with the changes made to the flatpak environment for 3.1.5, but it did not. Instead we are relying on text strings to see if it is correct then the log to find out why things don't work and that is most time opaque.

I did raise the issue with wxWidgets at the same time as gtk in the early days, but.....

This is not the end of the changes as you are already talking about wxWidgets 3.2.x so we will then have three versions of text 'extensions'. This will also mean that the xsd which is used to validate an pushes of plugins xml is always kept up to date, and backwards compatible, in all three environments, alpha, beta and prod.

bdbcat, 2022-01-29:
Jon...
Just so we will know:
What exact plugin build problem are you working now, that is causing you specific issues?

jongough, 2022-01-29:
Flatpak builds changed the version of wxWidgets that got used in the 'Beta' branch. I only found this out when moving OD into prod where the build succeeded but the install failed with an error message about not finding libwx_base-3.1 in the log. Apparently the 'beta' branch which was being used for the 2008 version was upgraded between me releasing to beta and through to prod. Following the links it now appears that we have moved the 2008 build to stable and the beta version is for wxWidgets 3.1.5 . I have now reworked frontend2 (testplugin) to handle this, BUT the flatpak version of OCPN is still on 3.0.5. So now I have two builds for the same flatpak enviornment, one with widgets 3.0.x and one for 3.1.x (soon to also be 3.2.x), but there is no way to distinguish between the different widgets versions in the xsd or the plugin manager.

This shows that there is little backwards compatibility and there is a need to consider how many plugins are going to have to be rebuilt for the new version of OCPN and in particular if the underlying version of widgets changes. It also has to be 'transparent' to the end users and not require them to look in the log. The issue being that the install worked, according to plugin manager, but failed to run. See [ Open Flatpak: preferences dialog not active, plugin dead? #241 ](https://github.com/jongough/testplugin_pi/issues/241#issuecomment-1021982963).

bdbcat, 2022-01-29:
Jon...
I get it.
We are on track to use extensions to the "target" string in the XML, exactly as we have done for MacOS.
So, as I see it simply, we probably need a new target definition as "flatpak-x86_64-wx315".
We can add this to the XSD, for all branches.  When OCPN(flatpak-wx315) is ready for this plugin, it will be recognized and installed.

This process is working more-or-less as it should, with some bumps in the road. viz.
1.  A new platform is defined and described (alec):  flatpak using wx315
2.  A plugin builder (jon) wants to builds a plugin against this platform.
3. We agree on the wording of the "target" in the plugin's metadata ("flatpak-x86_64-wx315")  This name depends on how the actual build of OCPN defines it's HOST ABI in the new platform, and is established by the author/sponsor of the new platform build. (alec)
4. I accept PRs against github plugins repo, on any/all branches, to update the XSD.  Or I simply do the update myself.
5. And then the new plugin can be published.

This is all just normal system maintenance.  It is what we do.
We are currently on step 3.  
Any disagreement with "flatpak-x86_64-wx315"? @leamas??

leamas, 2022-01-29:
> Any disagreement with "flatpak-x86_64-wx315"? @leamas??

Fine with me.

leamas, 2022-01-29:
> Flatpak builds changed the version of wxWidgets that got used in the 'Beta' branch. I only found this out [...]

Indeed, one of those things. During previous cycle we were using the Flatpak beta branch, initially for experimenting with aarch64 builds. Plugins were built from this branch to also experiment with flatpak-aarch plugins.

And when we  released 5.6.0 the changes in the beta branch was merged into stable.  From that point plugins should build from stable, exactly as we build plugins against released versions on Mac/Win/Debian rather than against the master branch. 

I think this basically is a communication problem. Other projects I work always have some "unstructured" channel like a mailing list and/or a IRC channel for  this kind of info.  If there was such a channel, this change would of course have been announced there. However, there is really no such place.

leamas, 2022-01-29:
@bdbcat :

> We are on track to use extensions to the "target" string in the XML

This is an important statement from the maintainer. Are you saying that you agree on the general principles in this PR, obviously without any commitments about timing?

leamas, 2022-01-29:
@jongough :

As you see above, you might be a little ahead of actual decisions. This PR is just a PR, it's definitely not merged and Dave has not given his approval or blessing. So, it might be a little early to make wx31 plugins.

Last time (the Ubuntu -> Debian transition), we discussed things and filed bugs against the templates when there was a common understanding.  Perhaps this would work here as well?

jongough, 2022-01-29:
Because of the change to the 20.08 beta flatpak environment it meant the any plugin built using frontend2 would fail when using it in the current process. I understand about upgrades and needing to test new features, but I was not aware that this change was going to be made.

It must be remembered that there are many plugins being generated, semi automatically, for a multitude of environments which makes testing difficult and keeping up with the changes challenging. This change to the build environment caused a runtime issue not a build issue, so would only be picked up if someone were testing flatpak for some reason. Luckily this time I was testing to show that frontend2 was not impacting the display of 'preferences' from the settings dialog. To do this I had to get OD (one of my plugins with a preferences dialog) running on flatpak on a compatible virtual machine. This showed up the error, hopefully, before anyone used frontend2 for build plugins.

Backwards compatibly must be considered when updating such environments or there is going to be confusion amongst users. So a plan needs to be constructed and then distributed so that others know what is happening. This is true for the wxWidgets 3.0->3.1 upgrade and for the 3.1->3.2 upgrade. Further, it would seem appropriate to find a way to cause a build error with this type of upgrade rather than leaving it to be a runtime error. Perhaps changing a name, version, environment variable or something else would help.

leamas, 2022-01-29:
> . I understand about upgrades and needing to test new features, but I was not aware that this change was going to be made.

Yes, as I said above: we have a communication problem: Where should such things be announced?

>  Further, it would seem appropriate to find a way to cause a build error with this type of upgrade

This problem is really hard for various reasons. I can't really see how it should be done.

jongough, 2022-01-29:
Could be done by passing an environment variable for the 'beta' flatpak (the only one that should/could be making breaking changes), giving a 'version number' and each time a breaking change is made change the 'version number'. If the version number passed does not match that in flatpak it fails with a version number message. The change to the build process would be minor to allow this. This is just one example, there would be others.

This is only addressing one side of the issue. The other is the runtime problems which only show up as errors in the OCPN log. It has been discussed before, but this highlights it, that the plugin manager could have a 'manifest' of requirements that the plugin expects and checks that. The current checks seem a little simplistic and small. Think what 'deb' files include to try and ensure the pre-requisites are installed before a package is. Plugin manager would not have to go that far, but could highlight missing capabilities or wrong versions.

The runtime issue does need addressing to handle backward/forward compatibility testing as part of the install process. It may also have to be done as part of the startup process so that changes to the system show up as incompatibilities as well. This would normally have been part of the package manager process, but is, or should be, part of the plugin manager process.

leamas, 2022-01-29:
you never give up, do you? I will not discuss this any more.

jongough, 2022-01-29:
I try to think of users as well as developers and suggest solutions that may be workable.

bdbcat, 2022-01-29:
Jon...
There was no _user_ in this particular scenario.  A developer found a fault, or deficiency, while working with and testing Beta branch code.  It got discussed and fixed. No harm, no foul....

bdbcat, 2022-01-29:
"flatpak-x86_64-wx315" added to XSD, three branches.

rgleason, 2022-01-29:
Thanks everyone for being "on top of it"!. I ask cautiously, is it ok to build using the new TP Templates now?

jongough, 2022-01-29:
Luckily there was no _user_ this time as I was testing for something quite different using a released, prod, version of OD. The build environment changed, the change was not picked up by the build process, the publish process or the install process for the plugin. It was the runtime that failed with no message to the user. Only looking through the logs identified the issue.

bdbcat, 2022-01-29:
"Luckily there was no user this time..."
That is why developers test all possible variations, surface issues, and fix them.

jongough, 2022-01-29:
I did test all I could and it worked, the next auto build, an hour later, generated the error as an underlying flatpak environment had changed without me knowing. I cannot test all 17 environments for every push I make to the repository. I don't have the time, the resources or the machines to do so. I test as much as I can in areas I know have changed, but I cannot test for every change to a docker image or flatpak environment and others would be in a worse position. So I am suggesting that thought be put into trying to avoid these issues and highlight when underlying changes have invalidated the plugin.

I am now in the process of making the changes to the xml and file names for wx315.

rgleason, 2022-01-30:
Jon you get a big thanks for catching it, from me.

jongough, 2022-01-30:
Interesting that for macOS it is `darwin-wx315-x86_64` and for flatpak it is `flatpak-x86_64-wx315`.

Testplugin (frontend2) 1.0.192.0 has a first pass at this change and is in the alpha cloudsmith repository.

rgleason, 2022-01-30:
Jon being the thorough character that you are, you would note that inconsistency. Should testplugin be switched to match?

jongough, 2022-01-30:
I noted it for its inconsistency. Having made the changes needed to support it the process is automatic and there is no human intervention, so it is not important. However, it shows the issue of mixing variable/names/contexts within a single xml element. If they were different elements, as xml is supposed to be, there would be no issue. That would leave the naming of the individual files, but that would, assuming no human intervention, not be an issue (have you looked at the actual file names that are stored in cloudsmith, they have build numbers added amongst other stuff, by circleci/cloudsmith).

rgleason, 2022-01-30:
Thanks will start updating slowly.

[70 more replies]

closed 2022-11-30 by leamas