OpenCPN/OpenCPN #2413: Grib_pi

rgleason, 2021-09-30:
A new Canadian user reports:  Any idea why Canadian grib2 files aren't loadable in here? This must be a very simple answer. Sourced from: https://dd.weather.gc.ca/model_gem_regional

I have tried one of the files and it does not load.
Trying to get more info about the CA website and the source of the files.

I believe Grib_pi should be compatible.

transmitterdan, 2021-10-01:
Rick,

Which file exactly? There seem to be many many files there.

I loaded this file:  [https://dd.weather.gc.ca/model_gem_regional/10km/grib2/00/008/CMC_reg_ABSV_ISBL_250_ps10km_2021093000_P008.grib2](https://dd.weather.gc.ca/model_gem_regional/10km/grib2/00/008/CMC_reg_ABSV_ISBL_250_ps10km_2021093000_P008.grib2)

It uses a projection type we don't support:  
<html>
<body>

20 | Polar Stereographic Projection (Can be North or South) (See Template 3.20)
-- | --

</body>
</html>

We support template numbers: 0, 10, 30 & 40

from this specification: https://www.nco.ncep.noaa.gov/pmb/docs/grib2/grib2_doc/grib2_table3-1.shtml

rgleason, 2021-10-01:
That is a huge help Dan. Thanks.  I wonder if we could support Polar Sterographic?  I suppost Didier could add that for our Canadian friends.

bdbcat, 2021-10-01:
" wonder if we could support Polar Sterographic? "
That is a pretty big ask. OCPN supports Mercator projections, mainly, since this is the main navigational projection used worldwide.  To support other projections, all the GRIB data would need to be re-projected (from Polar Sterographic to Mercator) before display.  This is do-able, but is a non-trivial graphics exercise.
Dave

rgleason, 2021-10-01:
Thanks Dave, that kind of answers the question, its a big ask, but I wonder if the world's new interest and awareness in the Poles might influence OpenCPN's charts at some point?

Should I close this now?

bdbcat, 2021-10-01:
Yes, with grib_pi documentation update specifying exactly the projections supported.

rgleason, 2021-10-01:
Added here 

https://github.com/opencpn-manuals/plugins/blob/main/manual/plugins/modules/grib_weather/pages/grib_weather.adoc
 
Closing

closed 2021-10-01 by rgleason