OpenCPN/OpenCPN #2414: O crash when receiving certain NMEA messages

rgleason, 2021-10-01:
OpenCPN 5.2.4 used by person reporting

Hakan checked & found O failed.
https://www.cruisersforum.com/forums/f134/opencpn-crashes-when-receiving-certain-nmea-messages-256118.html

- If I send the following 3 NMEA messages to OpenCPN, it freezes and I have to restart it:
- $HEHDT,90.0,T*16
- $HEGGA,000000.00,0.0,N,0.0,E,2,11,0.8,-0.5,M,44.1,M,,*64
 
- Here comes the even weirder part:
- If I only sends one of those messages there are no problems.
 
- If I send both messages, but change the heading in the HDT message to 89.9 degrees or 90.1 degrees, there are no problems.
- It is only when I send both messages and keep the heading at 90 degrees that OpenCPN crashes.

In weather-routing_pi we have an Issues that has hung around for awhile, not exactly this, but similar.
https://github.com/rgleason/weather_routing_pi/issues/79

bdbcat, 2021-10-01:
Corrected now by 27b95d60b7536b4b754af9e04e389d2c9cd6a828

Hakansv, 2021-10-01:
Dave..
brg += 1e9 would be something like 1 billion and the resulting heading from 90.00° is  ~20°
brg += 1e-2 fits better in my tests? Or am I out in the dark?

bdbcat, 2021-10-01:
Oh, all for the lack of "-" symbol.  I meant "1e-9", but there is no way for the compiler to know that in advance.

Fixed in github
Thanks

rgleason, 2021-10-01:
Thanks,  Does this have anything to do with the Weather_routing issue?

bdbcat, 2021-10-01:
I don't know.
Only thing to do is try it...

Hakansv, 2021-10-02:
Yes, compilers are surprisingly silly. The minus sign would have been obvious? :) 
Works fine. Thanks