OpenCPN/OpenCPN #2415: JSON OCPN_ROUTELIST_REQUEST generates a "Select Route" dialogue using Linux

Rasbats, 2021-10-02:
Ubuntu 20.04, OpenCPN 5.2.4 and 5.5.

Sending the json request message results in a "Select Route" dialogue appearing in the viewport, with OK and cancel buttons.

The plugin (JavaScript_pi) needs the JSON response, not a dialogue. Pressing OK crashes OpenCPN.

antipole2, 2021-10-02:
The following script can be run in the JavaScript plugin to demonstrate this issue.
It works correctly under MacOS and Windows but, if there is at least one route present, erroneously displays the Select Route dialogue under Ubuntu

```
// test OCPN_ROUTELIST_REQUEST
OCPNonMessageName(handleRL, "OCPN_ROUTELIST_RESPONSE");
OCPNsendMessage("OCPN_ROUTELIST_REQUEST", JSON.stringify({"mode": "Not track"}));

function handleRL(routeListJS){	// receive list of routes in JSON and build index
	if (routeListJS == null) print("No routes exist\n");	// no routes
	else {
		routeList = JSON.parse(routeListJS);
		if (routeList != null) {
			if (routeList.length > 1){  // (changed from 0 because null entry when no routes)
				if (!routeList[0]) routeList.shift(); // drop first null entry - don't know why it is there
				}
			}
		print(routeList.length, " routes found\n");
		}
	}
```

bdbcat, 2021-10-02:
Seems to work for me.
What am I doing wrong?
[image]

Rasbats, 2021-10-02:
Dave ... thanks for testing. Latest version of JavaScript_pi is 0.4. Tarball here:
[https://cloudsmith.io/~opencpn/repos/javascript-beta/packages/](https://cloudsmith.io/~opencpn/repos/javascript-beta/packages/)
I will test v0.3.

Rasbats, 2021-10-02:
Did a clean install of OpenCPN and the problem went away (using v0.4). All is well.
Sorry for a false alarm.

closed 2021-10-02 by Rasbats