OpenCPN/OpenCPN #2416: Color Correction: Menu Bar, Status Bar, Context Menu in Night Mode (Windows)

bc38, 2021-10-04:
Now the colors of backgrounds and fonts are not configured correctly, black font on a dark background is not readable.

I drew optimal colors for backgrounds and fonts for Menu Bar, Status Bar, Context Menu in night mode.
RGB color codes are marked.
It would be nice to take these colors as the standard for coloring UI elements for night mode

[image]
[image]
[image]
.

bc38, 2021-10-04:
on the images redrawn Сompass/GPS Status indicator for the existing style.
If this is interesting, I could continue in a separate topic on the forum

transmitterdan, 2021-10-05:
Some of this is controlled by the Windows theme.  For me, it is easy to see the pop-up menu in night mode.

[image]

transmitterdan, 2021-10-05:
You have some control of the text in the status bar.  I prefer the "Aqua" color for the status bar when it is not in daylight mode.  It is not a hardship to change this color using the Options->User Interface menu.  In the Fonts section select "StatusBar" and then "Choose Font" .  There you can select a color.

[image]

Hakansv, 2021-10-09:
On Windows I use to switch OCPN to full screen mode, F11, to cover all OS windows. What's tricky is frames on small windows like our Dashboard. The frame color when having focus is controlled by the screen themes but the white frame when out of focus is not easy.

mgrouch, 2021-10-09:
That’s half of the solution. 
A menu on right click will be bright.

transmitterdan, 2021-10-09:
Some big OS like Windows has features like automatically changing the theme based on time of day. Maybe you can look into that. Maybe in Linux you can find a way as well. 

But an app that unilaterally mucks with the desktop will not be well accepted. It is considered bad behavior.

rgleason, 2021-10-09:
https://www.faqforge.com/windows/windows-10/how-to-enable-dark-theme-at-night-on-windows-10-automatically/
https://www.howtogeek.com/356087/how-to-automatically-enable-windows-10s-dark-theme-at-night/
https://www.makeuseof.com/tag/auto-toggle-dark-mode-windows-10/

bdbcat, 2021-10-09:
"But an app that unilaterally mucks with the desktop will not be well accepted. It is considered bad behavior."
I agree completely.  We will not do that.
That, by the way, is why it is not possible to change some desktop element colors programmatically.  Like it or not, we must play nice as a Desktop member.  We do not "own" the Desktop UI, in any sense.

chanmckearn, 2022-02-25:
The problem I have is that the status bar for Open CPN on my Raspberry Pi is black on a fairly dark grey background and in direct sunlight it's almost impossible to read.  If I could change either the background in the status bar or the font it would really help, but I went to the Options/User Interface/Choose Fonts and tried to change the color, but when I hit apply and OK, the font just went back to its default.  The ability to change it doesn't seem to work.

transmitterdan, 2022-02-25:
What operating system are you using on the RPi?  I just tried this on my Pi and the font color changes as expected.

chanmckearn, 2022-02-25:
I'm sorry, I misspoke regarding my Raspberry Pi.  I'm running system 5.2.0 on that and my status bar font does change colors as it should.  I should have said on my Mac Book Pro running Open CPN 5.6.0 it does not work.  I go through the same steps as for on the RPi but when I hit "apply" nothing changes.  Also on the RPi I could not make the font bigger than 18.  I wish I could change the background color on the status bar as well as the font to make it more readable in the sunlight.  Do you know if that is possible.  I don't see it anywhere in the manuals or forum.

bdbcat, 2022-02-26:
I see two distinct issues here:
1.  MacOS, 5.6.0, status bar font does not change color.  Yes, that looks like a bug.
2. rPI:  Max font size is 18.  You wish it were bigger.  That is a Feature Request.

Is this correct?
For future, items like (2), mentioned in passing on another issue, are quite often lost in the noise.  If the behavior/bug issue is truly significant to you, open a fresh issue report on it.  That way, it won't be lost.

rahra, 2023-05-07:
The font colors are a mess on Linux as well which means that the colors of OpenCPN and the desktop theme somehow interfere but in a non consistent way. 
I run OpenCPN on Ubuntu Jammy (22.04) in the Kubuntu flavor (KDE Plasma desktop) and I typically have the dark theme active (which is default BTW).

Some windows a hardly readable, e.g. "Object Query" and "Target Query" from the context menu. When I change the color mode of OpenCPN (F5) the colors change and some are more readable than others.
When I change the desktop theme to light, all these windows (the text in it) seem to be very well readable in all OpenCPN color modes (as rotated with F5).
Interestingly when I change back the desktop theme to dark, the font colors of these sub windows do not change again, the stay in the colors as when I switched from dark to light scheme.

bc38, 2023-05-07:
By version CPN 5.10, you can [try to implement dark theme support for win32 applications](https://www.cruisersforum.com/forums/f134/win32-dark-mode-solved-270684.html)

bc38, 2023-05-08:
can you make screenshots of what it looks like?

rahra, 2023-05-08:
Ok, here we go. In the following you can see 4 different screenshots. Each screenshot shows an AIS target query (left) and an object query (right). Further the following color settings were used:

1. dark desktop theme OpenCPN day mode
2. dark desktop theme OpenCPN night mode
3. light desktop theme OpenCPN day mode
4. light desktop theme OpenCPN night mode

[image]
[image]
[image]
[image]

My personal opinion is that the target window in the night mode is hardly readable in both desktop themes and the object window in dark desktop OpenCPN daylight mode is impossible to read.

nohal, 2023-05-08:
The object query font is user configurable, it is advisable to set it light in case the system is using dark theme.
[image]
Anyway, this all needs a thorough revision for the next release as all the platforms already implement dark themes to some extent so we don't need to fight with them anymore (Or we do, but differently).

rahra, 2023-05-09:
Yes. I don't know how this is internally handled but the desktop usually offers many different settings (at least on Linux and Mac) from the theme. Probably it could be an option to take this as is and just give options to the user to tweak it a little bit. Although IMO the latter is not so important.
In the settings there are options to change fonts/colors but I think this is only chart related and does not change anything in the UI.

nohal, 2023-05-09:
> In the settings there are options to change fonts/colors but I think this is only chart related and does not change anything in the UI.

You are not right in your thinking. Most of the fonts you can set up on the UI tab in the Toolbox  are related to the UI, not the chart.

rahra, 2025-02-10:
Is there any progress with the color scheme switching?
Meanwhile I'm on OpenCPN 5.10.0 and it's still the same.

If adapting to the system scheme is so difficult, maybe it is more simple to put the colors of OpenCPN's three color modes into a config file where one can add his own settings instead.