OpenCPN/OpenCPN #2418: O-charts  uncheck enable causes O to crash - O should protect itself

rgleason, 2021-10-06:
This also happens to many of the current plugins
ochart
statusbar
polar
logbookkonni
vfkaps
shipdriver
otcurrent

I have checked many of the plugins, but not all of them.
I think OpenCPN should protect itself in this case,
But there should be a flashing notice with a change in screen color  that the plugin did not shut down properly so that PI Devs know that there is a problem.

transmitterdan, 2021-10-06:
Rick,

Explain exactly the steps to cause crash using the Shipdriver plugin.  Also, what version plugin and any other helpful info.  I tried it and could not cause a crash.

Thanks,
Dan

bdbcat, 2021-10-06:
And how about o-charts_pi?  How to reproduce the crash?  On what OS?

rgleason, 2021-10-06:
Sorry, on Windows 10.  for ocharts it was the same actions.

rgleason, 2021-10-06:
Maybe this happens when there is a plugin that has caused a similar crash and O is restarted, and then it happens to another plugin,  but it seems to happen pretty reliably and multiple times on certain plugins.  It is a lot of O failing and restarting, but I could get the oones listed to fail and the other ones were consistantly good.

rgleason, 2021-10-06:
Isn't there some way to protect OpenCPN from this, so the plugin crashes but OpenCPN recovers?
The Plugin API is all one way isn't it?  I thought you have generally asked plugins to provide their own lib etc.
I am just thinking of the problem that one plugin can cause OpenCPN to crash does not seem like it is a reliable navigation device.
Am I being too concerned about this?

I have tested this a number of times, walking through the failure, rebooting and trying it again, and then repeat and find another plugin that does this. It's not pretty.

I just had a vfkaps failure again, select Safe Mode on startup, from PIM interface, after selecting enable, opening preferences and then selecting disable.  O failure.  I sent the error report.

rgleason, 2021-10-06:
Ochart just crashed OpenCPN when I enabled it. Sent the crash report.

rgleason, 2021-10-06:
Using   OpenCPN 5.5.928+9271ee0  Win10    Second O crash

rgleason, 2021-10-06:
Can a "bad" plugin that is not enabled, cause O to fail due to disabling another plugin?
Seems unlikely.  I did these tests with all plugins disabled, and enabled one plugin at a time.

rgleason, 2021-10-06:
Polar  just crashed O, sent report.
I think this enough for tracing the problem.  
The list of plugins above are the problems, along with OpenCPN not being resilient to this problem.

rgleason, 2021-10-06:
Also please see https://github.com/jongough/testplugin_pi/issues/181

rgleason, 2021-10-06:
That is the intention, but I had hoped O could be more resilient, but I guess not.

What did you find out about ochart and my reports?

Is there some easy way to validate a plugin for this problem?
However there are many places where a plugin could crash.
So I guess this is a futile idea.

rgleason, 2021-10-06:
Dave,
I am sorry about this. I will do some more fishing around to determine this, but I think it happens both
when in Safe Mode and normal mode (without other plugins enabled).

I can install another version if you would like.  I will be moving to a new Dell Laptop soon with 1TB and blazing fast!
but I want to get all PI up to date first.

rgleason, 2021-10-06:
Sent another report, Aisradar 1.2.10.0   OpenCPN in normal mode, PIM, update catalog, enable, pref on/off, disable, crash O
( I noticed that PIM, update catalog, enable, disable, did not crash O on this PI)

AISradar also has an anonymous  Preferences selection  "Show Icon" left over from earlier days.  This needs to be expunged from the plugin, but it gets pretty intertwined in aisradar_pi.cpp.

rgleason, 2021-10-06:
Same procedure as Aisradar no other plugins enabled, same as above, open & close preferences, disable  O fails.   Sent report.
I think its pretty clear that under normal or safe mode this occurs.

rgleason, 2021-10-06:
Dave, 
Have you succeeded in making O fail with any of these plugins?

rgleason, 2021-10-06:
If I enable AISradar and then open Preferences and exercise the two ridiculous checkboxes   "Show Icon" and "Show Ais as radar" and then close preferences after awhile and then "disable"  OpenCPN does not crash.

bdbcat, 2021-10-06:
Rick...
I have never seen such a failure.
I have asked Hakan to try to reproduce this.

rgleason, 2021-10-06:
Do I have the right version of O?

rgleason, 2021-10-06:
I renamed my opencpn.ini and restarted to make a new opencpn.ini.
The problem still exists.

bdbcat, 2021-10-06:
"Do I have the right version of O?"
yes.

rgleason, 2021-10-06:
I've uninstalled and reinstalled   version  opencpn-msvc_5.5.928+2965.7965c69_setup
and the bugs still remain.
I guess I've done what is necessary to convince myself that it is not a problem with my installation.

rgleason, 2021-10-07:
Mike (Rasbats) has tested logbook and finds that O does not fail after open and close of preferences and then uncheck of the Enable checkbox.   However this is still happening in my machine.  
Since others seem to think this is not a problem, I am inclined to build and deploy to the Beta catalog..
Can someone confirm please?

bdbcat, 2021-10-07:
Rick...
OK, stop there.  No need to look at the entire fireworks display.  We'll focus on o-charts_pi only.
I re-iterate the crash instructions, to be clear.
1.  Start OCPN in SAFE mode.  (how exactly do you do that?)
2. open the PIM
3. You see a lot of plugins listed.  All are disabled.
4. Enable o-charts_pi (what o-charts_pi version is reported?)
5. Immediate crash occurs.

All correct?
Dave

rgleason, 2021-10-07:
I had to add CatalogExpert=1 to Plugins section due to new opencpn.ini.
Using Beta I see ochart and it is something like  v0.9.0  and I tried to update and get a checksum error.
The try to check enable and O crash.

rgleason, 2021-10-07:
I cleared the cache to and removed most of the other PI that were causing problems. 
Went to Beta, clicked on ochart  enabled and no crash.  Then disabled and no crash.
Did that 3 times. 
Did not see ocharts in the master catalog.

Then back to master and tried AISradar. It still fails.   I believe I deleted it.

rgleason, 2021-10-07:
I am headed for bed. Thanks.

bdbcat, 2021-10-07:
Rick...
So, o-charts_pi is OK.
Moving to AISradar:  I finally experienced the crash as reported.  I corrected the fault in AISradar source code, submitted PR to your github repo..  Tested locally, all is well.

I am now convinced that these crashes are caused by the plugins failing to close and delete their Preferences dialogs correctly. I suppose we will go thru them one by one.  Which one is next?

Hakansv, 2021-10-07:
FYI:
Looking in the plugin folder of my RPi the jail function seems to work. The date is from when I first updated from 5.2.4 to .5.5xx

[image]

bdbcat, 2021-10-07:
In status bar:
After line 141:

delete m_PreferencesDialog;
m_PreferencesDialog = NULL;

rgleason, 2021-10-07:
both for Logbook?    Thankk you

Rasbats, 2021-10-07:
>vfkaps
shipdriver
otcurrent

Deleting and nullifying all these in the present releases.

bdbcat, 2021-10-07:
OK, I'll fork it and do a PR later tonight.

rgleason, 2021-10-08:
Thanks Dave, that is a huge help. I think I should be able to do that now.

rgleason, 2021-10-08:
Making progress https://github.com/jongough/testplugin_pi/issues/177

Rasbats, 2021-10-08:
Do what?

rgleason, 2021-10-08:
Mike it is necessary so that Opencpn is does not crash!

rgleason, 2021-10-08:
That's good.  I can understand why you want to wait until SD template is updated.  Perhaps I should wait for Logbook too?
Flatpak isn't building for that, but we've made the change for closing preferences so O does not crash.  I wasnt going to push it to Plugins untill I had flatpak working...

Hakansv, 2021-10-08:
For info:
Win10 yesterday's master code.
I updated the master catalog and there was a o-charts update from 0.9.0.40 to 0.9.0.42
Installation went smooth but when I tried to deactivate o-charts O crashed.
Actions:
1. Update catalog master.
2. Selected o-charts and accepted update.
3. Installation OK. 
4. Deselected the o-charts row.
5. Clicked to deactivate o-charts. -- Crash.
6. I got a crash report and exported the files. (log attached)
7. Restarted O, not safe mode, and then I could deactivate and activate o-charts wo problem.

O stopped here:
17:29:13.675 MESSAGE OCPNPlatform.cpp:907 Getting DefaultSystemLocale...
17:29:13.676 MESSAGE OCPNPlatform.cpp:928 Found LanguageInfo for: Swedish
17:29:15.286 MESSAGE fpr.cpp:99 oexserverd results:
17:29:15.286 MESSAGE fpr.cpp:102 0
17:29:19.345 MESSAGE pluginmanager.cpp:1732 PlugInManager: Deactivating PlugIn: C:\Users\HAS-WIN\AppData\Local\opencpn\plugins\o-charts_pi.dll
17:29:19.345 MESSAGE chartdb.cpp:282 Chart cache PlugIn purge

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/7312582/opencpn.log)

rgleason, 2021-10-09:
Tomorrow we may have 11-12 PI to deploy if flatpak builds and Polar is fixed. Good progress today.
https://github.com/jongough/testplugin_pi/issues/177

Rasbats, 2021-10-10:
As mentioned in the CruisersForum:
An issue with included and other plugins:
To reproduce:  

- Open the preferences for a plugin
- Change a preference
- Disable the plugin/Apply/Ok

When Options -> Plugins is opened again the icon for the plugin shows as the jigsaw one.

This was found with oesenc/WMM and an updated version of squiddio so far.

rgleason, 2021-10-10:
Thanld. Thats old. The new ones are in beta.
I have no idea when I should push them to beta because I have no idea how many times eavh pi os version has been downloaded or used without a failure report. And i dont test each and every environment as that is impossible.

rgleason, 2021-10-10:
I meant master

rgleason, 2021-10-10:
Thanks, this appears to be another problem, but not so fatal.

Rasbats, 2021-10-10:
No ... please keep open. The issue with the jigsaw icon is mentioned here. It could be linked with this issue.

rgleason, 2021-10-10:
Polar_pi is working now, Thanks Dave.  
https://github.com/rgleason/polar_pi/issues/9

Now just need to get svg icons working.

bdbcat, 2021-10-11:
Rick:
"Now just need to get svg icons working."
I would drop this for now.  Move on to truly broken plugins/platforms, please.

Closing this issue now.  It has (over)run its course.
Mike:  Open another issue for "jigsaw" icon if necessary, please.