OpenCPN/OpenCPN #2427: RFE: option for automatic night mode switch

mgrouch, 2021-10-08:
option in OpenCPN to automate night/day/twilight modes of the screen based on gps local time and local sunset/sunrise

thx

sebastien-rosset, 2025-03-22:
Related to this, in the Weather Routing plugin, I am adding efficiency factors that are used during isochron propagation.
1. Upwind efficiency modifies boat speed when sailing toward the wind source.
2. Downwind efficiency adjusts performance when sailing with the wind.
3. **Night efficiency** applies a cumulative performance factor during darkness hours.

For night efficiency, that means the isochron propagation algorithm needs to know whether it's day or night at each specific geographical position and time being evaluated, requiring accurate sunrise, sunset, and twilight calculations based on coordinates and UTC time. Without this capability, the algorithm cannot properly apply the configured night efficiency percentage that accounts for reduced sailing performance during darkness.

The WR plugin could do the sunrise/sunset calculation on its own, but this would create unnecessary code duplication. A centralized API in OpenCPN's core ensures consistent sun position calculations across all plugins and allows improvements in calculation accuracy to benefit the entire application.

In addition, plugins may benefit from being able to query and receive events related to day/night transitions, enabling them to automatically respond to changing light conditions. An event-based system would allow plugins to register for sunrise, sunset, or twilight notifications at the current location, triggering automatic actions like switching chart modes, adjusting instrument backlighting, or modifying routing parameters without requiring continuous polling.

# Proposed Sun Position and Day/Night State API

I propose adding a set of functions to OpenCPN's plugin API that would provide standardized access to sun position data and day/night state information. These functions would:

1. Calculate sun position (elevation and azimuth) for any given location and time
2. Determine sunrise, sunset, and solar noon times for any location on a given date
3. Provide day/night/twilight state determination with configurable twilight definitions
4. Optionally offer an event notification system for day/night transitions

These functions would benefit not just the Weather Routing plugin but also dashboard displays, navigation tools, and the proposed automatic night mode switching. The API would handle all astronomical calculations internally, ensuring accuracy and consistency across the application.

# Proposed API Code

```c++
/**
 * Calculate sun position for a specific location and time
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param datetime Date and time in UTC
 * @param elevation Output: Sun elevation angle in degrees above horizon (negative when below)
 * @param azimuth Output: Sun azimuth in degrees from true north (0-360)
 * @return true if calculation was successful
 */
bool GetSunPosition(double lat, double lon, wxDateTime datetime, 
                    double* elevation, double* azimuth);

/**
 * Calculate times of sunrise, sunset and solar noon
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param date Date in UTC (time component is ignored for calculation)
 * @param sunrise Output: Time of sunrise as seconds since midnight UTC (or -1 if no sunrise)
 * @param sunset Output: Time of sunset as seconds since midnight UTC (or -1 if no sunset)
 * @param noon Output: Time of solar noon as seconds since midnight UTC
 * @param twilight_angle Angle of sun below horizon to use for twilight definition in degrees
 *                      (6° for civil, 12° for nautical, 18° for astronomical)
 * @return true if calculation was successful
 */
bool GetSunEvents(double lat, double lon, wxDateTime date,
                  double* sunrise, double* sunset, double* noon,
                  double twilight_angle = 6.0);

/**
 * Determine the day/night state at a specific location and time
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param datetime Date and time in UTC
 * @param twilight_angle Angle of sun below horizon to use for twilight definition in degrees
 * @return DAY_NIGHT_STATE enum value (DAY, TWILIGHT, or NIGHT)
 */
enum DAY_NIGHT_STATE {
    DAY_NIGHT_STATE_DAY = 0,
    DAY_NIGHT_STATE_TWILIGHT = 1,
    DAY_NIGHT_STATE_NIGHT = 2
};

DAY_NIGHT_STATE GetDayNightState(double lat, double lon, wxDateTime datetime,
                                 double twilight_angle = 6.0);
```

sebastien-rosset, 2025-03-22:
I've created #4415 to track the proposal separately.