OpenCPN/OpenCPN #2429: Signal K / dashboard error at start or when satellites lost

LennartG-Sve, 2021-10-10:
There is something odd going on when Signal K is reading NMEA 2000 before hooking up to any satellites, see also the attached picture. SK is sending a speed although no satellites are detected. The GPS unnit is a Garmin GPS24xd NMEA 2000 (probably the best they have). I've also tested with another GPS giving the same result.

- HorizontalDilution equals '-0.01'  (perhaps correct but should IMHO here be a high popsitive number)
- MethodQuality equals 'no GPS'  (no satellites in view so this is correct)
- satellites equals '0' (also correct)
- speedOverGround equals 655.34 m/s (this is definitely wrong and always the same value)

As neither OpenCPN nor dashboard does any sanity check on data from SK dashboard displays a speed of 1273.89 knots in NNE direction (direction varies slightly). This should be taken care of somehow.

This is also fully repeatable, I've used a tin can covering the GPS to force a test case  ;-)

[image]

.

Hakansv, 2021-10-10:
I discussed this on the SK portal: https://app.slack.com/client/T02ENM6QA/C03F1MKQG/thread/C03F1MKQG-1633896838.359500
A short answer so far:
"The fix should probably be in n2k-signalk. It should probably not report any data when methodQuality is “no GPS”?"
Let's see what's coming up..

nohal, 2024-02-16:
Is there any actionable item for us here? Perhaps actively disregarding the data when `navigation.position.methodQuality` is `no GPS`?

Hakansv, 2024-02-16:
Pavel..
I have not seen any quality value in my stream. Is it implemented in SK? Or if it's seldom transferred by the GNSS receiver it would be for less use here?

nohal, 2024-02-16:
I see it on the screenshot above.

Hakansv, 2024-02-16:
Yes, but it's not always transmitted from the GNSS receiver. 
So we shouldn't rely on it's presence. But if received and the value is "no GPS" we would act as suggested, agree.
BTW: value: "no GPS"! What if OK "Glonass"? :-)

nohal, 2024-02-16:
I suppose it might be transmitted in the delta message only when it changes. Maybe, really don't know the internals of the SK server. That can be handed by a simple state machine, no problem.