OpenCPN/OpenCPN #2430: Jigsaw icon for installed plugin, when using preferences

Rasbats, 2021-10-11:
An issue with included and other plugins:
To reproduce:

-  Open the preferences for a plugin
-  Change a preference
-  Disable the plugin/Apply/Ok

When Options -> Plugins is opened again the icon for the plugin shows as the jigsaw one.

Restarting OpenCPN restores the correct icon for the plugin.

This was found with oesenc/WMM and an updated version of squiddio so far.

Not common to use this combination of menu options. Early indication of whether a fix will be deferred to the next version of OpenCPN would be appreciated.

rgleason, 2021-10-11:
Using  Beta Catalog   I just tried Grib_pi 0.0.0  enabled it, closed and got the icon in the taskbar. The PIM, select  grib_pi, preferences, checked  use gradual colors  and OK and O crashed immediately.  When I reopened all the uninstalled folders were at the top of PIM with green puzzle icons.   When I diid the same to disable  Grib_pi crashed.

This seems to be pretty common, even included plugins.....

rgleason, 2021-10-11:
Watchdog and Polar are OK.

rgleason, 2021-10-11:
GPS-odometer  v0.6.2.3 has this problem..  not available yet

leamas, 2021-10-12:
Since some plugins work and some not this is most likely a plugin issue rather that an OpenCPN one.

Also likely to be related to #2418, perhaps just different symptoms for the same basic bug: the preferences dialog not handled correctly when closed/deleted.

rgleason, 2021-10-12:
Thanks Alec. That is helpful & good to know that Mr. O is ok.

closed 2021-10-14 by bdbcat