OpenCPN/OpenCPN #2437: Build on RPi400

Hakansv, 2021-10-15:
I bought me the new RPi400. It's really recommended not least for new Linux users. A little bit faster than RPi4 and surprisingly low CPU temp despite no cooler fan.
A Swedish keyboard and a very fine book about LInux and Raspberry fundamentals. Here in Swedish!
I've installed OpenPlotter to the included SD card and all is running wo issues.

Now, I tried to set up the build environment but failed and may need some help from our Linux experts. 
The old method can't find "libgtk3.0-gtk3-dev". The new method lacks info. See below extracts.

I'm posting here since it may come more questions about this machine.
Thanks
Håkan

>> System:
pi@RPi400a:~ $ hostnamectl
   Static hostname: RPi400a
         Icon name: computer
        Machine ID: f06ab3a473dc4acfb2922bd400893992
           Boot ID: 6d47ec52754e46758f4706e02ac8341d
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.63-v7l+
      Architecture: arm

pi@RPi400a:~ $ lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1800.0000
CPU min MHz:         600.0000
BogoMIPS:            324.00
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32

>>Installing dependencies acc to the "old" method failed in gtk3...

pi@RPi400a:~ $ sudo apt-get install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev wx-common libgtk3.0-gtk3-dev libglu1-mesa-dev libgtk2.0-dev libgtk3-dev wx3.0-headers libbz2-dev libtinyxml-dev libsndfile1-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'git' instead of 'git-core'
E: Unable to locate package libgtk3.0-gtk3-dev
E: Couldn't find any package by glob 'libgtk3.0-gtk3-dev'
E: Couldn't find any package by regex 'libgtk3.0-gtk3-dev'
E: Unable to locate package libgtk3-dev
pi@RPi400a:~ $ 

>>Installing dependencies acc. the "new" method wants some more info?

pi@RPi400a:~ $ sudo apt install devscripts equivs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
devscripts is already the newest version (2.19.5+deb10u1).
equivs is already the newest version (2.2.0).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@RPi400a:~ $ sudo mk-build-deps -i -r ci/control
E: You must put some 'source' URIs in your sources.list
mk-build-deps: Unable to find package name in `apt-cache showsrc ci/control'
pi@RPi400a:~ $

bdbcat, 2021-10-15:
Hakan....
The "old" method:
If the version of wxWidgets in the "official" RPi400 repo (which is what?) is built for GTK2, then you should not need any part of GTK3.
So, try the "sudo apt-get..." without any GTK3 references at all.  Then see if the normal "camake" succeeds.

Hakansv, 2021-10-15:
Dave..
I tried that but cmake didn't found wxWidget and other things. I'll come back.
Thanks

mgrouch, 2021-10-15:
@Hakansv 

OpenCPN for pi is build currently with gtk2 not gtk3

Hakansv, 2021-10-15:
@leamas 
re: "That list is basically obsoleted and should not be used. The method in the manual, which uses ci/control, is tested since it is part of the ci builds."
Yes could be. I'll check further what should be included in the "control"
When I run your(?) command:
sudo mk-build-deps -i -r ci/control
I got this::
E: You must put some 'source' URIs in your sources.list
mk-build-deps: Unable to find package name in `apt-cache showsrc ci/control'

leamas, 2021-10-15:
> When I run your(?) command:

Certainly not mine. It's just what's used in the ci builds which we know works (and maintain together as soon as anything breaks).

Hakansv, 2021-10-17:
It turns out if I put in a SD card with the the OpenPlotter image using NOOBS the "old" dependencies are working fine and cmake is happy. 
Now the Raspberry official site recommend not to use NOOBS for RPi4 and RPi400 anymore. Reason unclear so far.
Now we're outside OCPN so I close this issue. It could also be the RPi400 will not be the first choice as a build environment. For daily OCPN navigation the board will though be an attractive candidate.
Thanks for all advises!

closed 2021-10-17 by Hakansv