OpenCPN/OpenCPN #2444: GRIB plugin - bugs and improvements

bc38, 2021-10-20:
1. In the properties, the option "Zoom to file center when opened" is always enabled, even when the checkbox is cleared
2. The wind direction arrows have become very thin, merges with the chart, a little thicker would be done as it was in version 5.
3. Can the plugin automatically open the latest grib file by date and time when enabled? Now this only works when the CPN is restarted. This is relevant for those who have the Grib files in the directory are updated automatically.

Hakansv, 2021-10-20:
bc38..
I'm on Windows and can't reproduce all of your remarks. What's your system?

1. My setting, see pict, is saved and valid all time I start.
2. Also here I see no difference.  See pict2
3. I can agree on this. Last opened file is loaded and stays there even if grib_pi is closed. So to grab a new file you've to open the grib file manager or restart O.

[image]

[image]

bc38, 2021-10-21:
Windows 10, video nvidia GTX 1070 ti

1. I think I saw the logic of the "Zoom to file center when opened" option when it is enabled, the grib is centered on the screen when
[image]
 the plugin is first enabled. BUT when the "Zoom to file center when opened" option is off, when the "open new file" command is selected, the grib is centered on the screen, which contradicts the option. If the option "Zoom to file center when opened" is off, then logically there should be no centering.
2. See attachment
3. Let's discuss whether it is possible to make this improvement.