OpenCPN/OpenCPN #2446: Display error

OgreTransporter, 2021-10-21:
I have found a possible display error. When using the map https://www.elwis.de/DE/dynamisch/IENC/enc-dateien/WW12_2021-06-22_0_226.zip the end i.e. the harbor in Dortmund is not displayed as water area. On the official website (https://atlas.wsv.bund.de/clients/desktop/?parameter=visible&value=iencwms) and in other map plotters like WinGPS the display is correct.

OpenCPN 5.2.4+6b314e6
[image]
[image]
[image]

OgreTransporter, 2021-11-11:
I tried this on the current release candidate (https://www.cruisersforum.com/forums/f134/opencpn-beta-5-5-1109-rc1-released-257723.html):
[image]

transmitterdan, 2021-11-11:
If you magnify (+) more does it change the color?

nohal, 2022-05-04:
Not reproducible with current IENC set and current OpenCPN for me, closing.

closed 2022-05-04 by nohal