OpenCPN/OpenCPN #2459: Untranslatable strings in localization

bc38, 2021-11-01:
Crowdin translated all new lines, no translations applied to CPN
more on screenshots
[image]

plugin state statuses are not translated. Although in Crowdin the translation is made
[image]

did not find the ETA string to translate
noticed that the name of the month does not fit into the field
maybe leave the month and day on the top line and the hour and minutes on the bottom?
[image]
[image]

bdbcat, 2021-11-02:
1.  String translations corrected.
2.  ETA console field width and layout improved.
3. In settings->Fonts, you should use the items that appear translated into your preferred language.  So, if you are set for language RU, then ignore the EN entries that appear on the list..  This is a long term bug, but no real harm....

Please try this version:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-msvc_5.5.1031+3051.54b46b6_setup.exe

Same link as https://github.com/OpenCPN/OpenCPN/issues/2460

bc38, 2021-11-02:
Works great
Thanks Dave!

closed 2021-11-02 by bc38