OpenCPN/OpenCPN #2460: Different font sizes in the menu

bc38, 2021-11-01:
The problem is long-standing! Has risen repeatedly!
I will raise the question again!
When changing the font, the Menu is not applied to some labels, in particular plugins.
In Russian localization, the font is not applied to the Measure function.

Maybe changing the default font size will solve the problem? It is quite small as standard.
[image]
[image]

bdbcat, 2021-11-01:
I have corrected in github master.  Please test if possible.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-msvc_5.5.1031+3046.b35f185_setup.exe

bc38, 2021-11-01:
> I have corrected in github master. Please test if possible. https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-msvc_5.5.1031+3046.b35f185_setup.exe

Finally!
Everything looks great in EN!
[image]

but in RU
[image]

bdbcat, 2021-11-02:
Corrected here:

https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn-msvc_5.5.1031+3051.54b46b6_setup.exe

bc38, 2021-11-02:
Wish for the Menu for the future
When the Touch Screen function is active, the line spacing in the Menu should be slightly wider for a comfortable finger touch

closed 2021-11-02 by bc38