OpenCPN/OpenCPN #2463: Options Charts list dialog Unused area

Hakansv, 2021-11-01:
Win10 
Sorry for a late note but the charts pathway are truncated and there is a "unused" column in the list control. Could we maybe use the whole area so If the dialog width is adjusted the available space is used for the path string?

[image]

Hakansv, 2021-11-01:
Dave..
with this change I get rid of the empty column. Works fine in Win but I'm not sure for Unix so no PR. You may test?
Håkan
```
@@ -3640,9 +3640,9 @@ void options::CreatePanel_ChartsLoad(size_t parent, int border_size,
   chartPanel->Add(activeSizer, 1, wxALL | wxEXPAND, border_size);
 
   pActiveChartsList =
       new wxListCtrl(chartPanelWin, ID_LISTBOX, wxDefaultPosition,
-                     wxSize(100, -1), wxLC_REPORT | wxLC_NO_HEADER);
+                     wxSize(100, -1), wxLC_LIST | wxLC_NO_HEADER);
 #ifdef __OCPN__ANDROID__
   pActiveChartsList->GetHandle()->setStyleSheet(getAdjustedDialogStyleSheet());
 #endif
```

transmitterdan, 2021-11-02:
This change looks like this on RPi Linux (Ubuntu 21.04).

[image]

Hakansv, 2021-11-02:
Dan..
Yes, looks OK I suppose.
Also on RPi Debian it's fine. If the path is to long to fit it's not truncated but a slider is present.

[image]

bdbcat, 2021-11-03:
A slight additional patch required for linux x86_64.  Otherwise nice.
Thanks, Hakan.

commit 443fc44d (2021-11-03): Improve Chart Directory list presentation format.  Closes #2463
closed 2021-11-03 by bdbcat