OpenCPN/OpenCPN #2464: Improvements to C code for robustness

nhasabni, 2021-11-01:
Hi,

I work at Intel, and we have developed a tool that detects anomalous programming language expressions that can possibly lead to bugs. We scanned the code repository for this project as it has considerably high number of stars!

We found a couple of places where the code/expressions are confusing and seem to implement the logic in a rather convoluted manner. We think that the expressions could be rewritten to capture the logic accurately and precisely.

Line 404 in [libs/wxservdisc/mdnsd.c#L404](https://github.com/OpenCPN/OpenCPN/blob/master/libs/wxservdisc/mdnsd.c#L404) has expression `if(c->q = _q_next(d, 0, r->name, r->type))`, which seems to be missing `()` around the assignment.

Any thoughts on the findings? If this looks acceptable, I'm happy to send a pull request with the changes.

rgleason, 2021-11-01:
https://github.com/IntelLabs/control-flag

bdbcat, 2021-11-02:
Improvements in clarity, while following our present coding style, are always welcome.
PR's at the moment are on hold, as we are in the very late stages of a release cycle.  After the release of OCPN 5.6.0, new PRs are encouraged.
Thanks

leamas, 2021-11-02:
This warning is just the tip of the iceberg.  wxserverdisc gives a really large amount of compiler warnings, larger than any other part of our main software. In next cycle it might be necessary to take a broader approach here,  perhaps re-considering the use of the library (?) or start some serious code cleanup.

mgrouch, 2021-11-14:
Have you tried setting up OpenCPN and plugins in

https://sonarcloud.io/

for code inspections?

Thanks

rgleason, 2021-11-14:
That did not work for me, but I found
https://sonarcloud.io/documentation
https://sonarcloud.io/documentation
For some strange reason

rgleason, 2021-11-14:
Is this the same as  https://www.sonarqube.org/  ?  Looks like it.

mgrouch, 2021-11-14:
> That did not work for me, but I found
> https://sonarcloud.io/documentation
> https://sonarcloud.io/documentation
> For some strange reason
> This just worked https://sonarcloud.io/ very strange.
> Opensource is free https://sonarcloud.io/pricing
> I am chicken to do it.

You can just login to it by picking login using GitHub. And it will guide you how to set it up. There are only 3 simple steps. One add api key to GitHub project and second add build.yml file for cmake. (It’s copy paste action)

rgleason, 2021-11-14:
Mike have you done this with your OpenCPN repository?  How did it go?

mgrouch, 2021-11-14:
I’ve done it. But it failed during compilation. My fork is in disarray. 
That’s why it failed. But it run something

nohal, 2022-05-04:
Closing for lack of activity, @nhasabni if you feel like submitting PRs improving the quality of the code, please do so directly without opening issues asking if they are viable, they always  are if they make sense.

closed 2022-05-04 by nohal