OpenCPN/OpenCPN #2468: Question:  O Load check ini for actual PI files & dir

rgleason, 2021-11-05:
Example:
When I clean out dashboard_tactics  dll files and directories from 

- %localappdata%/opencpn/plugins/
- programdata/opencpn/plugins/cache
- programdata/opencpn/plugins/

and Start Opencpn, I find that the PIM still shows two installations that do not exist.

Shouldn't this be corrected by having a pre-check as Opencpn reads the opencpn.ini file?

I suppose I could just go into opencpn.ini and remove those references, if that is where they are located...???

However it seems to me that the PIM should represent what is actually present.

rgleason, 2021-11-05:
I found that by updating the catalog, these two install instances disappear.  However the question remains. Should Opencpn check PI files and directories for existance, as it loads the opencpn.ini file?

Two Phantom Installation Files
[image]

Try to load one
[image]

After update catalog
[image]

rgleason, 2021-11-05:
I found this directory 
C:\ProgramData\opencpn\plugins\install_data
maybe we just have to remove the 3  dashboardtactics.*   files?

rgleason, 2021-11-05:
Actually, the problem is the existing catalog which has the incorrect metadata files with the wrong common name.
When this is corrected, the problem will probably go away.

bdbcat, 2021-11-05:
Yes

rgleason, 2021-11-05:
Ok, but shouldn't O when loading opencpn.ini check to see if the PI files exist?

bdbcat, 2021-11-05:
Please tell me again what problem you want to solve?

rgleason, 2021-11-05:
https://github.com/OpenCPN/OpenCPN/issues/2468#issuecomment-961884023
Look at "Two Phantom Installation Files"  and  "Try to Load one"  

This is what happens when I clean out dashboard_tactics dll files and directories from

    %localappdata%/opencpn/plugins/
    programdata/opencpn/plugins/cache
    programdata/opencpn/plugins/

Which made me realize that OpenCPN assumes when it loads opencpn.ini, that nothing has changed!
Shouldn't Opencpn be checking to confirm that those installation files exist and if they don't, just remove them?

rgleason, 2021-11-05:
I suppose that Opencpn could just update the catalog on its own every time it starts, but that would tend to drive users nuts, I think.

bdbcat, 2021-11-05:
Rick...
If you want/need to remove a truly bogus plugin, you should first "Uninstall" it with the PIM.
Then you may clean the three directories you mention, if you like.  The most important, and probably only one that needs cleaning, is "%localappdata%/opencpn/plugins/"
Finally, download a new master catalog.

All this comes about from publishing a master catalog containing an error in a plugin definition.  Let's try not to do that again.

OpenCPN .ini file has very little to do with plugin detection and loading.  It is all controlled by the contents of the important directories, and the catalog.

rgleason, 2021-11-05:
Ok, understood.

I still have some questions.  I could keep the incorrect commonname  "DashboardTactics"  and change the 

Line 98 of dashboard_tactics_pi.cpp

```
#ifdef _TACTICSPI_H_
const char *dashboard_pi::s_common_name = _("Dashboard_Tactics");  <---To "DashboardTactics"
#endif // _TACTICSPI_H_

```

rgleason, 2021-11-05:
OK, I've removed both DashboardTactics 1.6.6.0 and Dashboard_Tactics 1.6.6.13  and I see what you mean.
I will keep it as DashboardTactics and "change the common_name"

closed 2021-11-05 by rgleason