OpenCPN/OpenCPN #2469: Draw a route, some confusion

rgleason, 2021-11-06:
1. Select Create a Route
2. Hamberger rolls up, causing some distraction.
3. It looks like the cursor has gone away and disappeared.
4. Finally after awhile messing around, I discover the cursor shows at the edges.
5. Then I realize that the Pencil is brown and so small that it is hardly noticed.
6. And by then trying to create a route I am able, but I cannot see the pencil and probably would not see it in bright light.
7. Why? because laptop has a high res screen 3800x2400
8.  Admittedly most older screens are far less resolution.

Suggestion: Have the cursor scale up with the User Interface Scale Factor because that would be naturally appropriately responsive to the screen resolution.
The alternative is to have the "cursor size scale setting" as a new scaling bar added to the similar bars in Options > User Interface >

bdbcat, 2021-11-07:
Rick...
Are you using Windows display scaling other than 100%?
Gonna need a screenshot....

rgleason, 2021-11-07:
[image]
Pencil is between the route and the bullseye, just above the bullseye. Hard to see.
[image]

This is using User Interface Scale Factor = 5
[image]

[image]

Enable Scaled Graphics Interface is checked
[image]

[image]

Display Settings
Scale and Layout   =  250% (Recommended)
[image]

I can't find the pencil in these screenshots, the act of taking a screenshot must eliminate it.
If it is there it is just above the bullseye, very faint.
[image]

bdbcat, 2021-11-07:
Thanks for the screenshots.
1.  Which shot represents the "most likely" use settings for you, in actual navigation?
2. What does increasing the "Chart Object" scale slider do, if anything?

rgleason, 2021-11-07:
It is a new computer and I've been using it basically to build plugins and test, so the toolbar can be smaller than if I were using it on the boat.  I currently have "User Interface Scaled Graphics set to 5 everything else 0 and  "Enable Graphics Scaled Interface" = Off.  

I could set  "Enable Graphics Scaled Interface " to ON and  lower "User Interface Scaled Graphics" to -1 or so for being on the boat.

Neither of these settings seem to affect the cursor or pencil size. Since there are two settings that affect the User/Graphics interface  I don't know which is most appropriate to attach the pencil too.  It might be better as a separate setting like the ship.

I have not used charts yet on this machine, it is about 10 days old to me, so that may be a factor too.

Hakansv, 2021-11-07:
For info and comparison:
Win10 "Scale and layout" = 175%
I can see the route pencil rather well. It's kind of "thin" but since the route line end is following the cursor I see no problem using the route tool.
No screen shot since the cursor disappears as soon the screen shot function is active.

bdbcat, 2021-11-07:
Sadly, as it turns out, the maximum cursor size on Windows is limited to 32 x32 pixels.  I expect this will be very small on hi-def displays.
There may be some way to increase this limit, involving the Windows registry.  Online info suggests that this would be a fragile hack, not recommended. So, suffice to say, this will not be included in O56.
But possible for future consideration.
Sorry
Dave

rgleason, 2021-11-07:
Dave,
Why not Fatten up the damn pencil so we can actually see it? 
In daylight it will be impossible otherwise. As it is it is pretty bad.

rgleason, 2021-11-07:
Also my white cursor (Arrow) is probably bigger than 32x32, what function is that using I wonder?  That can be scaled up and down. I find it is the "mouse cursor"  
[image]

Also there is a text cursor that can be scaled up and down
[image]

bdbcat, 2021-11-07:
The windows control panel cursor sizes are for so-called "system" cursors.  The pencil cursor is a wxWidgets custom cursor, and from the wx documentation:
`In wxMSW the foreground will be white and the background black. If the cursor is larger than 32x32 it is resized.`
Only clear solution would be to stop using OCPN custom cursors for hi-def displays on Windows.
Thoughts on that?

rgleason, 2021-11-07:
Yes, that would be fine I think.

rgleason, 2021-11-07:
If I set it at 175% instead of 250% text will be impossible to see. I suppose I could run the computer screen at less resolution and it is an Nvidia graphics card so that might work, but it will be less than recommended. Don't know it it does anything to the hardware screen.

rgleason, 2021-11-07:
The wxMSW will probably have to change at some point, but when? and when will it get to O?

bdbcat, 2021-11-07:
"The wxMSW will probably have to change "
Not clear.  I would not hold my breath.
I guess not too many people using wx custom cursors on modern machines.

rgleason, 2021-11-07:
So can you change the cursor to the regular cursor for HIgh Res screens without messing up OpenCPN? I think that would be most preferable.

rgleason, 2021-11-07:
Thank you Dave. I will try to build it now and try it.

https://github.com/OpenCPN/OpenCPN/commit/16ae5b70ecdc724506edb499a225a02326530c82

rgleason, 2021-11-07:
That works great!  Thank you Dave!!!

Can't see the cursor here but it is the same as what is set for the mouse and is adjustible in size!
[image]

closed 2021-11-07 by rgleason