OpenCPN/OpenCPN #2470: VDR filtering

rgleason, 2021-11-08:
Hans and I are discussing Dashboard_tactics and testing using his nmea file.
We would like to filter MWV, MWD, AIVDM, AIVDO from the incoming stream and are having trouble doing that, as the NMEA data window does not change to show colored filtered output.
I have tried both serial and Network. What are we doing wrong here? I havent done this for awhile.
Also inserting these filters in the Nmea data window seems to permanently stop the scrolling and I don't know why.
[image]

rgleason, 2021-11-08:
Running a VDR file does not require a Connection.
The Nmea0183 debug window shows the file being read.
The filter at the bottom of the NMEA window is a "Viewing Filter" and only works as a highlight filter to exclude everything else from the screen which is still running in the background. It does not filter out sentences.

How would we filter out sentences from a VDR file?
Do we have to edit the file?

rgleason, 2021-11-08:
If so, I think that VDR could be improved considerably by having a built in filter to exclude or include.

CarCode, 2021-11-08:
You are working with OpenCPN a long time now. Finally you should know the difference between the main software OCPN and its plugins now.

rgleason, 2021-11-08:
Ignored

transmitterdan, 2021-11-08:
Rick, filtering is applied to incoming streams as you suspected. VDR does not generate a stream. You can use my VDRplayer tool to play a NMEA file. Then filters will work. 

Filters are applied on a per-stream basis. Filters are not global. This is by design because we need the ability to filter one message from a stream but allow that message from another stream. 

VDR is a plug-in and so doesn't appear to O as a stream. It just kicks out API messages.

Use my Python tool and you will be simulating very close to the real world.

rgleason, 2021-11-08:
Another good reason to setup VDRplayer again. Thank you Dan.

> You can use my VDRplayer tool to play a NMEA file. Then filters will work.... Use my Python tool and you will be simulating very close to the real world.

closed 2021-11-08 by rgleason

rgleason, 2021-11-08:
@transmitterdan   I realize you have a Readme and I've created some documentation in the [User Manual here](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_software:nmea_instruments&s[]=vdrplayer#vdrplayer)  but I think it would be much better to have an up to date folder in your github repository for "manual" and to make asciidoc/antora files so that we can include VDRplayer in the new [Plugin Manual.](https://opencpn-manuals.github.io/main/opencpn-plugins/0.1/index.html) so that users know more about it.

rgleason, 2021-11-08:
That is on my rather long list some time in the next 2 months