OpenCPN/OpenCPN #2474: PIM oversized Image problem

rgleason, 2021-11-12:
After cleaning out  climatology data and dll, and dashboard_tactics data and dll, in %localappdata%/opencpn/plugins  and then cleaning out all occurances of climiatology and dashboard_tacticsin in the cache/tarballs  and also clean out entire metadata directory,  in the process of doing a final test for some changes.  Then started Opencpn, and Updated plugins, the PIM screen changed and there was an enlarged green puzzle.
[image]
[image]
[image]
[image]

[image]

Tried updating the catalog and closiing O and opening, it still persisted.

rgleason, 2021-11-12:
I turned off Enable Scaled graphics and closed options and then opened several times and this problem seems to have gone away for the moment.

rgleason, 2021-11-13:
I think this can be closed. Probably due to a plugin.

closed 2021-11-13 by rgleason

rgleason, 2021-11-13:
I am going to reopen this because [ Findit_pi on branch  master-update](https://github.com/rgleason/FindIt_pi/tree/master-update)  has this exact problem.  The plugin installs and works but the findit_panel.png icon does not seem to work properly and we get the overzized panel image. I thought I had it fixed when I found the  png file was 37x37 but reducing it to 32x32 did not fix it. 

By finding the fix for findit_pi we will have a better idea for what causes the oversize images in PIM.  I am not able to figure this out.   Rasbats has been helping a lot with suggestions and small but important fixes.

I think that if for some reason the Listing Panel Image for PI Manager PIM is not working properly, then OpenCPN should be showing the green puzzle image at the proper scale in any case!!  ..as it is quite disorienting to users and needn't be, also the plugin likely works fine, the panel image is just not working.

Bad PI Manager Listing Panel Icon

[image]

New version installs fine

[image]

The new version works fine, as did old one,

[image]

1. The  PIM Listing Panel Icon is not being found  and the green puzzle is not being rendered properly by Opencpn. (since it is green puzzle, )
2. The fix to this plugin will provide some insight to what causes this to happen. (I've tried and have given up)

 [ Findit_pi on branch  master-update](https://github.com/rgleason/FindIt_pi/tree/master-update) 

Sorry. Many thanks to Rasbats who has also helped a lot with this particular problem.

rgleason, 2021-11-13:
Also, after this, I cleaned out polar_pi and findit_pi from %localappdata% and then emptied the cache metadata and tarballs.
Then started opencpn again, the overzied green images were still there for several plugins including findit.
I updated the catalog twice. Closed Options and opened, sitll green oversized puzzle. 
Closed Opencpn and reopened and the green puzzles were gone.

rgleason, 2021-11-13:
I take that back. There is a large green puzzle still for Polar_pi even though I removed that entirely. I found that when opened, I could "Downgrade"  when opened there was only version 20 available which is in the catalog and works fine. I downgraded and the large green icon went away.  Here are some screenshots. Of note is the condition remains even after removal of the plugin and reboot.  It appears that Opencpn  ini or something else is still in the state where the enlarged green puzzle occured.

[image]
[image]

[image]

rgleason, 2021-11-13:
PS: I am trying to get Polar_pi  toolbar svg icons working without much success. Then I was going to change the Listing Panel Icon.
The Polar_pi toolbar svg icon work in the[ master-update branch  ](https://github.com/rgleason/polar_pi/tree/master-update)

The Findit_pi listing panel icon work is in the [master-update branch](https://github.com/rgleason/FindIt_pi/tree/master-update)   

They both seem to be creating oversized green puzzle images.

rgleason, 2021-11-13:
Well, Findit was doing the same thing, It worked but the PIM  Listing or Panel PNG file was not found so the green puzzle became giant.  Rasbats identified that problem, as using the capitalized plugin repository name rather than all lower case!
Using all lower case for the plugin name fixed it in this statement

```
  // Create the PlugIn icons
    initialize_images();

 // From shipdriver to read panel icon file
    wxFileName fn;

    auto path = GetPluginDataDir("findit_pi");
    fn.SetPath(path);
    fn.AppendDir("data");
    fn.SetFullName("findit_panel.png");

```

bdbcat, 2021-11-13:
Rick
You evidently missed the post on CF:
https://www.cruisersforum.com/forums/f134/opencpn-beta-5-5-1109-rc1-released-257723-4.html#post3518619
This error is corrected in github.
Had nothing to do with specific plugins.
Rebuild OCPN locally, and all should be well.

rgleason, 2021-11-13:
Thanks, I think I saw the post but did not comprehend it I guess.   Now findit is working well and the big green is not occurring, now just have polar_pi to sort out.