OpenCPN/OpenCPN #2476: Grib plugin overlay and current time

mgrouch, 2021-11-13:
Does Grib plugin adjust forecast time to current time when it displays overlay over chart?

Thanks

rgleason, 2021-11-13:
No Grib is set at UTC.  I believe this is a duplicate Issue and it should be closed.

https://github.com/OpenCPN/OpenCPN/issues/2453

closed 2021-11-13 by rgleason

mgrouch, 2021-11-13:
I meant as I move through the day
grib overlay should also change to give up to date forecast automatically. I know there is ‘now’ button. But I think automation is better. It can show past weather data which would be irrelevant.

rgleason, 2021-11-13:
I see, can you explain this clearly on the other issue please, it is basically the same topic. Time, but another kind of timekeeping.

rgleason, 2021-11-14:
We learn from history. History is a very important component in making decisions.  It is not irrelevant.

transmitterdan, 2021-11-15:
Almost no one leaves the Grib viewer on for hours at a time in my experience.  When you open a grib file it will automatically align with the current UTC time but it does not automatically advance if you leave it up for 6 hours.  Remember that the hourly feature of the plugin uses interpolated data and so is a combination of data.

I will look at whether it is easy to implement this feature.  It should not be hard. But I guess no else one cares because it would require leaving the grib viewer on all the time.

rgleason, 2021-11-15:
Dan, while at it https://github.com/OpenCPN/OpenCPN/issues/2453