OpenCPN/OpenCPN #2477: RFE: Boat parameters set up wizard

mgrouch, 2021-11-13:
I think it would be nice to have boat parameters set up wizard, which would need to include

1. Draft
2. Vertical clearance
3. Boat Polars
4. Leeway

All these parameters are needed for routing algorithms (Weather Routing/Dashboard Tactics, etc plugins) which can benefit from these data.

I think current integration of weather routing, grib, climatology, polars, tactics is too hard to use for an average user.

There is also lack of routing like in Navionics app (based on vertical clearance to pass bridges, tides and currents and draft)

What do you think?
Thanks

rgleason, 2021-11-13:
Mike, I wish you would vote for some of the Flyspray items such as these under "Vetted"

https://opencpn.org/flyspray/index.php?do=details&task_id=920&order=summary&sort=desc
https://opencpn.org/flyspray/index.php?do=details&task_id=2267&order=summary&sort=desc
https://opencpn.org/flyspray/index.php?do=details&task_id=1102&order=summary&sort=desc

mgrouch, 2021-11-13:
I think there are plenty of routing features built in into plugins right now already. 

What is needed is general integration. 

You have a boat. You know it’s parameters. Draft, Vertical clearance, leeway, polar performance. It’s one input to routing. 

You have data for tides, currents, wind (via grib), climatology. Another input to routing. 

You have chart. Depth, obstructions, bridges heights, etc

You have AIS to see other vessels and their course, speed, etc

You have tactics plugin which builds laylines for tacking. 

It would be great to have an 
algorithm which takes all that into account to build a route to a destination. and be able to recalculate it periodically. and ask for 
missing data if it needs it. 

At least making this as a strategic
goal for OpenCPN development. 

thanks,
—MG

mgrouch, 2021-11-13:
And it’s better to split OpenCPN into
several independent pieces with well defined interfaces. 

Weather provider,
Chart provider,
Router algorithm provider,
UI canvas,
Alert manager,
Boat manager

and so on. It will help it to evolve. 
One monolithic piece in C++ is hard to manage.