OpenCPN/OpenCPN #2478: Bug in OCPN_OSDetail

transmitterdan, 2021-11-18:
In MyApp::OnInit() there is a bug.  This code that came in with commit 6c172d1daa195329655a667c6d02495166db6bdc

 msgplat.Printf( "OCPN_OSDetail:  %s ; %s ; %s ; %s ; %s",
              detail->osd_arch.c_str(),
              detail->osd_name.c_str(),
              detail->osd_version.c_str(),
              detail->osd_ID.c_str(),
              detail->osd_names_like[0].c_str() );
  wxLogMessage(msgplat);

It accesses osd_names_like which is never assigned on MSW.  Maybe on other platforms as well.  I looked and only saw it initialized in linux.

leamas, 2021-11-18:
Indeed. But is this actually a problem? The variables displayed are std::string which has a sane default constructor. That means that on Windows or other nonlinux platforms this should just print empty strings. Or?

transmitterdan, 2021-11-18:
The code dereferences a non-existent element of an array. The array is size zero and so causes a trap at least in debug mode. The simple fix is to populate all of these at creation with a string with a space or two to avoid the traps.

leamas, 2021-11-18:
Ah, right,  missed that one. Seems fixed in current master at 7ab946d7d20

bdbcat, 2022-04-07:
Close?

closed 2022-12-06 by transmitterdan