OpenCPN/OpenCPN #2480: RFE: Add rate of turn to dashboard options

mgrouch, 2021-11-24:
Add rate of turn to dashboard options 

In some cases it can be derived 
from other parameters 

Thanks

rgleason, 2021-11-24:
Dashboard has normally not shown derived data (I believe this is dashboard's original intent), Just the nmea0183 data stream. Hakan has modified this somewhat with respect to true wind.

Hakansv, 2021-11-24:
This is not easy and probably nothing for Dashboard. There was a good attempt for this function here: https://github.com/OpenCPN/OpenCPN/pull/2039
but didn't worked out.
Better to trust a real calculation from a gyro or as nowadays available from a satellite compass. The latter also suitable for a small boat if you have the money. If the ROT info will be more frequent on pleasure boats it could be added as source for a Dashboard instrument. Either as NMEA0183 or N2k via SignalK? 
But all this is more of a future request than a issue concerning present code?

mgrouch, 2021-11-24:
From autopilot you have 
pitch, roll, yawn. I think it’s not difficult to calculate ROT and should be quite accurate using some kind of moving average.

mgrouch, 2022-01-06:
Looks like latest pypilot sends ROT as NMEA sentence