OpenCPN/OpenCPN #2484: RFE: Chart Status Bar on high res displays

rgleason, 2021-11-29:
Dave is it possible to make an adustment in the Chart Status Bar height for high res displays?
It is a very small line right now.
[image]

bdbcat, 2021-11-30:
Rick...
Sorry, too late for OCPN 5.6
This kind of thing looks small, but would affect every platform, every display resolution.  Would reset the test process dramatically.
I will have to wait for the (inevitable) first update to O56

rgleason, 2021-11-30:
Understood.  In thinking about this issue more, I think that when the time comes to do it, perhaps there should just be a user setting that can be changed in the ini or config.

rgleason, 2021-12-09:
In Ocpn-Draw due to high res screen the Ocpn-Draw Icons are very small and the drawing tool is miniscule. 
I asked Jon if he could fix these and he responded:

```
The cursors used by OD are the ones supplied by OCPN. OD does change the cursor depending on what the user is doing, but they are still the cursors supplied by OCPN though the "GetIcon_PlugIn" and "SetCursor_PlugIn" API calls. So I am not sure what you expect OD to do with this. It is a problem for OCPN.

I hope any change made by OCPN is reflected in the calls by plugins for the cursor icons.

```
https://github.com/jongough/ocpn_draw_pi/issues/496#issuecomment-990232149

rgleason, 2023-01-16:
See https://github.com/OpenCPN/OpenCPN/issues/2893

rgleason, 2023-01-16:
see https://github.com/OpenCPN/OpenCPN/issues/2893

rgleason, 2023-01-29:
Fixed. Closing

closed 2023-01-29 by rgleason