OpenCPN/OpenCPN #2486: universal binary for macOS

tcurdt, 2021-12-03:
Is there any chance for the macOS build scripts to create a universal binary that also works on Apple Silicon M1 (ARM)?

nohal, 2021-12-03:
It is not that easy as also all the dependencies must be built as universal binaries and as some minor code changes are required. I have it done locally, although tested only on x86 due to lack of HW, but it will definitely not be part of the upcoming 5.6 release.

Do you have any problem using the x86 version on M1? So far all the reports we have is it works just fine on Apple Silicon.

tcurdt, 2021-12-03:
> It is not that easy as also all the dependencies must be built as universal binaries and as some minor code changes are required. I have it done locally, although tested only on x86 due to lack of HW,

Could you outline the procedure? Or is there a script that should work?
Would maybe a homebrew based build be possible?

> but it will definitely not be part of the upcoming 5.6 release.

OK. Thanks for the clarification.

> Do you have any problem using the x86 version on M1? So far all the reports we have is it works just fine on Apple Silicon.

Certainly sounds like a good interim solution.
But if I could help to have native support - that would be even nicer.

Thanks for the quick reply!

nohal, 2021-12-03:
Basically you just need to add `-DCMAKE_OSX_ARCHITECTURES="x86_64;arm64"` whenever you invoke `cmake` or add `-arch arm64 -arch x86_64` compiler flags when the dependency you want to build uses autotools.

I will be making a pull request with the changes rebased on the current master and detailed instructions to build all the required dependenvies as I get to it.

caesar, 2021-12-18:
Pavel, have you managed to build on an M1 system, or did you build the universal binary from an Intel machine?
I'm trying to build on an M1 Mac and having no luck so far…

nohal, 2021-12-18:
Nope, the only Apple machine I have still is the same 2015 Intel Macbook. Does the M1 system complain about something in particular?
And how are you doing? Wanted to send you an e-mail when I saw Ocean Tramp in AFASyN two weeks ago...

caesar, 2021-12-18:
Hmm, it complains about everything 😆 but basically:
- First it was missing 32-bit builtins (I don't recall what exaclty). So I set `-DCMAKE_OSX_ARCHITECTURES="x86_64;arm64"` which resolved that particualr issue, but:
- Then it builds about ~35% and then when it gets to jeeps it complains that it can't use precompiled headers with multiple `-arch` flags. Even If I only set one architecture in `CMAKE_OSX_ARCHITECTURES` it is the same. I couldn't work out how to disable precompiled headers so I set `-DOCPN_USE_GARMINHOST=OFF` just to skip that part for now, but:
- Now I'm getting a load of invalid calls to builtins such as:
    ```
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/13.0.0/include/mmintrin.h:394:19: error: first argument of MTE builtin function must be a pointer ('__v2si' (vector of 2 'int' values) invalid)
    return (__m64)__builtin_ia32_paddd((__v2si)__m1, (__v2si)__m2);
    ```
    (yeah that looks like 32 bit stuff still, idk what's up there)
- Just for kicks I tried using GCC instead of clang with `-DCMAKE_C_COMPILER=/usr/bin/gcc -DCMAKE_CXX_COMPILER=/usr/bin/g++` but it still calls clang at some point and I get the same errors. (I read that GCC doesn't have the same issue with precompiled headers.)

I haven't got any further yet but probably my next step when I have time will be to check I can build ok on my old Intel machine just to re-familiarise myself with the process because I haven't built OCPN for a couple of years now.

Possibly worth noting:
- I didn't do a custom build of wxwidgets; I just used the homebrew version
- There are some hardcoded paths in `CMakeLists.txt` that point to `/usr/local/`, but the homebrew prefix on Arm Macs is `/opt/homebrew/`. Setting a few env vars and `-DCMAKE_PREFIX_PATH="/opt/homebrew" -DLibArchive_INCLUDE_DIR="/opt/homebrew/opt/libarchive/include" -DLibArchive_LIBRARIES="/opt/homebrew/opt/libarchive/lib"` _seemed_ to solve that for now.

---

As for me, I'm doing well thanks, I hope you are too – I'll drop you an email 😀

nohal, 2021-12-18:
Homebrew libs are unfortunately no good at least on x86_64 as they contain only single architecture, I had to build all the dependencies myself (except wxWidgets master at the time I used brotli-1.0.9, fontconfig-2.13.1, gettext-0.21, libb2-0.98.1, libpng-1.6.37, pixman-0.40.0, zstd-1.5.0, cairo-1.16.0, freetype-2.11.0, libarchive-3.5.1, libexif-0.6.22, lz4-1.9.3, xz-5.2.5), but unfortunately did can't find any notes on what exactly each of them required to produce usable result. It was not that hard though, only minor problems, if any...

You can check what's inside a binary like this

```
# lipo -archs /usr/local/lib/libfreetype.6.18.0.dylib
x86_64 arm64
```

I've pushed the OpenCPN code as I left it a couple of months ago to https://github.com/nohal/OpenCPN/tree/macos_universal for reference. This was building and producing an usable result on x86_64 back then.

caesar, 2021-12-18:
> Homebrew libs are unfortunately no good at least on x86_64 as they contain only single architecture, I had to build all the dependencies myself

Ugh, silly me, I didn't think of that. I wonder if a separate, parallel, x86_64 build of homebrew would work. Probably no use anyway as I guess there's probably no way to give separate include paths for each architecture when building the universal binary. Hmm.

But tbh I'd be happy if I could just build a single-arch arm64 build to start with – but it seems that even with just `-DCMAKE_OSX_ARCHITECTURES="arm64"` it's still trying to build for x86_64 as well.

> I've pushed the OpenCPN code as I left it a couple of months ago to https://github.com/nohal/OpenCPN/tree/macos_universal for reference. This was building and producing an usable result on x86_64 back then.

Thanks, I'll try building that branch and see how it goes. I'll try it on my old Intel machine too if I can (it's dying, hence the purchase of the expensive new one…)

caesar, 2021-12-18:
re https://github.com/nohal/OpenCPN/commit/fc10d8c7c80cdb44efd0a237a4d017188db50773, do you remember why you had to remove the dark mode stuff? It doesn't look related, but I guess there was a reason ;-)

re https://github.com/nohal/OpenCPN/commit/c6244273cc02f75d3ad32af89ac04579007a556a, I guess you had the same issue as me with jeeps not compiling…

nohal, 2021-12-18:
> re nohal@fc10d8c, do you remember why you had to remove the dark mode stuff? It doesn't look related, but I guess there was a reason ;-)

It does not build (If I remeber correctly the stuff used for the trick is not visible anymore), but we would be removing it anyway as macOS now has real dark mode support so on first sighting of a problem I just took the chainsaw and performed the amputation...

> re nohal@c624427, I guess you had the same issue as me with jeeps not compiling…

Yes, pretty sure.

caesar, 2021-12-18:
Hmm, I recall last year reimplementing the dark mode stuff to actually use the native dark mode. But maybe the stuff you removed wasn't part of that. I'll have to look into it.

In the mean time: I didn't have any more luck building your branch, but I realised that the problems I was having with callls to missing 32-bit builtins are coming from the bundled texcmp library. It seems like that's only bundled if OpenGL is enabled so I thought I'd try with `-DOCPN_USE_GL=OFF`, but it seems OCPN doesn't actually build without OpenGL because there are hardcoded calls to `GetglCanvas` etc in various places.

Hmm, I'll get there in the end 😆

nohal, 2021-12-19:
I think nobody tried to build with `DOCPN_USE_GL=OFF` for quite some time...

The only problem with `texcmp` I remember was we build it with SSE2 enabled and that obviously can't be used on ARM, so I ended up disabling it for the whole universal build in https://github.com/nohal/OpenCPN/blob/macos_universal/libs/texcmp/CMakeLists.txt#L86-L91
But it will still be broken if you build just ARM64 as the condition won't match and you end up in the `else` branch where it is enabled.

caesar, 2021-12-19:
> I think nobody tried to build with `DOCPN_USE_GL=OFF` for quite some time...

Haha no, I started to realise that as I got deeper into the weeds... I should probably start a new issue for that, but: is it intended to still be a possibility? It should either be fixed or the option should be removed...
I added a ton of ifdefs and did get a working build without OpenGL in the end, so I could submit a PR for that. It'll need some testing and cleanup first though.

> The only problem with `texcmp` I remember was we build it with SSE2 enabled and that obviously can't be used on ARM, so I ended up disabling it for the whole universal build in [...]

D'oh, I should have gone through your changes in more detail. As I say, I did get a working build without OpenGL in the end last night, bit just disabling SSE would be a far preferable solution so I'll try again this evening.
Also, that would be why I didn't have any luck building your branch with both architectures too, because I had them in the other order (Arm first) so again the condition wouldn't have matched.

The thing I still don't understand though is why it's insisting on also building Arm even if I try to only build x86_64... Oh well, I feel like I'm making progress, anyway!

caesar, 2021-12-19:
By the way, perhaps only a partially related question, but has anyone had any luck with builds for Windows on ARM? Or Linux on ARM other than Raspberry Pi?

nohal, 2021-12-19:
That match could certainly be changed to only `arm64` substring to cover all the cases, but the real solution should be to change the `texcmp` code to actually allow compiling `x86` with SSE2 and `arm64` without in the multiarch build.

We have been building ARMHF and ARM64 in the Ubuntu PPA, but I have no idea if someone actually uses ARM OpenCPN on anything but RPi, less if OpenGL is working there.

tcurdt, 2021-12-19:
> We have been building ARMHF and ARM64 in the Ubuntu PPA, but I have no idea if someone actually uses ARM OpenCPN on anything but RPi, less if OpenGL is working there.

Besides macOS there now also is Windows ARM64. So far it's in beta but seems pretty solid. How much the latter will spread is of course yet to be seen. If it spreads there will probably also be more Linux ARM64, too. But I guess that's all still up in the air.

caesar, 2021-12-19:
Looks like there's a longstanding ticket for fixing the build with `-DOCPN_USE_GL=OFF` at #698, so I'll take that discussion there and hopefully submit a PR as I do have a working build now (haven't tested fully yet though).

But now, back to trying to actually do a proper build *with* OpenGL… :-)

dwkaiser, 2021-12-20:
I currently have the task, "Run/Build OpenCPN for BeagleBone" on my list.  This will be armhf on "non-RPi".

caesar, 2021-12-20:
I did (fairly easily in the end) get an arm64-only build working on macOS yesterday. Disabling SSE in texcmp for all Arm builds (since it's an Intel-only feature anyway) solved most of the issues.

My build seems to be missing the built-in plugins (they were built, but not copied into the app bundle, so they don't show up when I run OCPN) – but maybe that's because of the recent changes in the plugin system? I guess there's somehting I should be doing to get the plugins copied into the final app bundle.

Now, back to trying to get a universal binary to compile…

> but the real solution should be to change the texcmp code to actually allow compiling x86 with SSE2 and arm64 without in the multiarch build

Yes, that would be ideal… I'm not sure how to go about it, but if I can get everything else working, I'll look into it.

caesar, 2021-12-20:
> I currently have the task, "Run/Build OpenCPN for BeagleBone" on my list. This will be armhf on "non-RPi".

Great. I assume armhf on other systems shouldn't be much different from armhf on RPi. I will also be trying an arm64 Linux build later. I think the 8GB Pi 4 is arm64, do the current builds work on that?

caesar, 2021-12-21:
It wasn't too hard to get a universal build working in the end – the far easer solution instead of building all dependencies manually is to use macports, which (unlike homebrew) has an option to install universal versions of most formulae.

The only thing I haven't solved is the precompiled headers issue with the garmin library.

Oh, and the built-in plugins not being bundled, but I haven't looked at that at all yet.

When I get a chance I'll document the steps to build the universal binary and send a PR with the few changes I had to make. Hopefully then we can get a universal build working in CI.

bdbcat, 2021-12-21:
caesar...
Some things to think about:

It took some effort building custom macports formulae to allow building a bundle that will run on MacOS 10.10, on a Catalina CI host.  Then, at run-time, several libraries required for a 10.10 bundle faulted when executed on 10.11+.  Also required some ugly hacks to macport formulae.  Not an uncommon occurrence in the MacOS ecosystem, I understand.
Then, there were issues with the CI environment.  Rebuilding all of the required macports libraries added massive time to the build cycle, so that the CI timed out after one hour.  I requested more time from CCI  for this platform, to no avail.  We could of course buy more time from CCI, using real money.  Anyway, I ended up using the CCI caching system, and downloading a bundle of prebuilt libraries, in order to get the build time back to something reasonable.  Not really sustainable in a universal build model, I'm afraid.

So, all this makes me wonder if a new universal build should be created as a totally new target, for MacOS 11+.  We would keep and maintain as long as reasonable the existing x86 build as a "legacy" release, covering 10.10-10.15.

Thoughts?

caesar, 2021-12-21:
Dave, all good things to think about, indeed. I'm not familiar with how you have the CI set up, but I did notice that the CI build script for macOS seems to use Macports.
I'm surprised about the build times, though. When I installed the dependencied from Macports yesterday, prebuilt binaries were downloaded for everything – even for the "universal" (multiarch arm64/x86_64). I didn't have to build anything, except our custom build of wxWidgets (a binary was also available, if we didn't need a custom build).

All that said…

> So, all this makes me wonder if a new universal build should be created as a totally new target, for MacOS 11+. We would keep and maintain as long as reasonable the existing x86 build as a "legacy" release, covering 10.10-10.15.

This could be a valid option, but I was also thinking perhaps another approach would be not to do a Universal build at all – instead, do an Intel build and an Apple Silicon build. The Intel build would target macOS 10.10+ on x86_64, and the Apple Silicon build would target macOS 11.0+ on arm64.

What do you guys think about that approach?

I still think a Universal build would be better _from a user point of view_ if we can do it easily, but only if the issues Dave mentions can be sorted.

Either way, I'd like to get to a state where the build tooling works for producing both Universal and single-architecture builds, regardless of what we end up doing for CI / release versions.

tcurdt, 2021-12-21:
> This could be a valid option, but I was also thinking perhaps another approach would be not to do a Universal build at all – instead, do an Intel build and an Apple Silicon build. The Intel build [...]

I assume a universal build would be transitory anyway - just like it was with PPC.
I personally never liked the overhead anyway. I am not sure that UX is really affected that much.

I'd be more than fine with a single arch build.

bdbcat, 2021-12-21:
caesar...
"When I installed the dependencies from Macports yesterday, prebuilt binaries were downloaded for everything – even for the "universal" (multiarch arm64/x86_64)."
True enough.  But those libraries will not work on MacOS 10.10, either due to missing dependencies, or frank crashing due to some API variance between the build environment _vs_ the runtime.

Background:
1.  dependencies installed by macports default formulae on MacOS 10.15 will not run on MacOS 10.10.
2.  dependencies installed by macports default formulae on MacOS 10.10 will not run on MacOS 10.15.

In order to run on any legacy MacOS version prior to 10.15, we needed to build some dependent libraries using macports on a real MacOS 10.10 system using custom formulae. These libraries are bundled into a zip file, and stored in the cloud.  During the CI build process, the zip file is downloaded, fixed up, and bundled with the app package.

This was a PIA to setup and test.  And it still is somewhat fragile.  I think combining a universal build requirement with the existing CI will be a rats nest.  So a parallel CI build, using the same codebase (ifdefs allowed)  for arm64 on MacOS 11+ makes sense to me.

bdbcat, 2021-12-21:
I wonder if CircleCI offers a build environment supporting MacOS 11 and arm64 natively?  Or any other CI provider, for that matter?

caesar, 2021-12-22:
Hmm, I see. That does sound like a mess.

> dependencies installed by macports default formulae on MacOS 10.15 will not run on MacOS 10.10.

This doesn't surprise me I guess, as there would be no real reason for the default builds to be built with compatibilty for older systems.

> dependencies installed by macports default formulae on MacOS 10.10 will not run on MacOS 10.15.

I find this very surprising. I've never known an older program to not work on a newer system. Not saying it doesn't happen, but it's something I've never experienced, and it certainly *shouldn't* be the case except in very exceptional circumstances.

> I wonder if CircleCI offers a build environment supporting MacOS 11 and arm64 natively? Or any other CI provider, for that matter?

I don't know, but it *shouldn't* (in theory) be necessary to build on an arm64 machine, because macOS 11+ should be able to cross compile for arm64 (I think Pavel has done this already on his own machine).

> So a parallel CI build, using the same codebase (ifdefs allowed) for arm64 on MacOS 11+ makes sense to me.

I agree that this sounds like the way to go, at least in the short term. That particular bild *should* (haha) be easy to set up, with no custom dependency builds needed (we'll see…)

dwkaiser, 2021-12-28:
nohal and caesar,

I have OpenCPN, 5.6.0~ubuntu20.04.1 armhf, installed and working on Debian Bullseye running on a Beaglebone Black.

I am happy to do much more testing, and even building/testing of plug-ins, if someone wants to provide some direction/priorities.

Do we have any established test procedures and/or test code?

Thanks,

Dave K.

nohal, 2022-12-12:
Current state:

Native Apple Silicon ARM64-only build should work out of the box without any modifications

Universal binary has the following prerequisities:
  - The dependencies instead of from Homebrew built universal, usually `cmake -DCMAKE_OSX_ARCHITECTURES="x86_64;arm64" ..; make; sudo make install`
  - OpenSSL 3.0 built using an updated version of https://github.com/BlueP2P/openssl-prebuilt for mongoose (To do that, I used this dirty patch)
```diff
diff --git a/libs/mongoose/CMakeLists.txt b/libs/mongoose/CMakeLists.txt
index 986671429..1aefcfb32 100644
--- a/libs/mongoose/CMakeLists.txt
+++ b/libs/mongoose/CMakeLists.txt
@@ -24,12 +24,12 @@ if (NOT WIN32)
     message(STATUS "Mongoose: APPLE: Setting OpenSSL...")
     target_include_directories(
       MONGOOSE
-      PUBLIC  /usr/local/opt/openssl@3/include
+      PUBLIC  /tmp/openssl/openssl-3.0.7-x86_64/include/
     )
     target_link_libraries(
       MONGOOSE
-      PRIVATE /usr/local/opt/openssl@3/lib/libcrypto.3.dylib
-      /usr/local/opt/openssl@3/lib/libssl.3.dylib
+      PRIVATE /tmp/openssl/openssl-mac/libcrypto.a
+      /tmp/openssl/openssl-mac/libssl.a
     )
  endif (NOT APPLE)
 else (NOT WIN32)
```

bdbcat, 2022-12-12:
1.  Assuming the dependencies are rebuilt for multi arch, can we create a CI script that will do the right things and build a universal image?
2. I assume that plugins will need to be rebuilt for Apple silicon arm64 as well?

nohal, 2022-12-13:
The short answer is yes and yes.

Apple did a very good job on the crosscompilation toolchain, so basically `-DCMAKE_OSX_ARCHITECTURES="x86_64;arm64"` really is all that's needed to add support. Of course the build time more or less doubles as the compilation is really done twice. But once the CI build works, it should be a trivial task even for the plugin developers to switch to producing universal dylibs.
The issue of course can be the more obscure dependencies, like the SG-lock SDK (haven't checked whether it is linked to the plugin itself, for `oexserverd` it would not be a problem as it is a separate process and can run on Rosetta) or S-63.

We would have to maintain the build of all the dependencies (Much easier now since we got rid of Cairo, so the only "weird" one is OpenSSL) or switch from Homebrew to MacPorts, which as @caesar said support multiarch.

I would probably leave the "official" universal bundle (or separate ARM64/x86_64) for the next cycle as there is more than enough other stuff on the table, neither me nor you have the ARM hardware constantly available and Rosetta seems to be doing a tremendous job at x86_64 emulation.

May oppinion, @keesverruijt  or @caesar may have a different perception on this.

bdbcat, 2022-12-13:
"like the SG-lock SDK (haven't checked whether it is linked to the plugin itself, for oexserverd it would not be a problem as it is a separate process and can run on Rosetta)"  No problem.  Runs as separate process.  Not linked to plugin dylib.

MacPorts could help us here, since the builds can be more accurately controlled.  But it is a can of worms...
I agree that we should leave for the next cycle.  Our main focus on O58 has to be on linux, to support the debian plan.

And then there is Metal graphics....

nohal, 2023-09-28:
Universal binaries are now produced given the correct dependencies are available, see https://github.com/OpenCPN/OpenCPN/pull/3442

closed 2023-09-28 by nohal