OpenCPN/OpenCPN #2488: map shown only the lower left qudrant of the display area

arnaudlb, 2021-12-06:
with the version 5.6.0 of open CPN ( and the previous 3 release candidate on ubuntu)

I experience that the display area is 3/4 black, showing the map in only the lower left of the drawing area, as shown in the following screenshot

[image]

(this laptop computer has a rather high screen resolution (3300x2200) and I am therefore using a scaling ratio of 200% for the dispaly in the windows manager)

let me know how I should investigate further

leamas, 2021-12-06:
I'm no position to handle this. That said, what operating system are you on?

rgleason, 2021-12-06:
Try updating your chart database. And also turn off and on opengl and zoom in and out.

nohal, 2021-12-06:
Start OpenCPN using `GDK_SCALE=1 opencpn` Is the problem gone?

arnaudlb, 2021-12-06:
@leamas I am using GNU/Linux, Ubuntu 20.04

@nohal using `GDK_SCALE=1 opencpn` worked although the icons are really small. However, increasing the `user interface scale factor` to 5 gives me usable icons. Thank you

rgleason, 2021-12-14:
@arnaudlb  As OP,  Can you please close this?

arnaudlb, 2021-12-14:
@rgleason sure, the proposed solution worked on my system. Thanks

rgleason, 2021-12-14:
T

closed 2021-12-14 by rgleason