OpenCPN/OpenCPN #2492: PluginHandler::isCompatible(): Strange runtime system identification

leamas, 2021-12-08:
In PluginHandler.cpp, around line 229
```
 // Check the plugin for host run-time ID match.
  // This test is necessary for cross-built hosts.
  // For example, Ubuntu PPA builds for armhf define PKG_TARGET as ubuntu-armhf
  // But run-time reports as raspbian on "Raspberry Pi OS".
```

\+ related code. 

I have problems to wrap my head around this. If I understand the problem correctly, the PPA builds generates a _ubuntu-armhf_ target for the plugins which is rejected by opencpn which only accepts _raspbian-armhf_.

Now, _ubuntu-armhf_ is plain wrong, and is not listed in ocpn-plugin.xsd. Isn't then the solution to force the target in the PPA build to be the correct value using OCPN_TARGET_TUPLE rather than to patch isCompatible()?

bdbcat, 2021-12-08:
You understand correctly.
"Isn't then the solution to force the target in the PPA build to be the correct value using OCPN_TARGET_TUPLE"
Maybe.   Probably.  But consider that there may be other environments where the ubuntu-armhf build would be useful, such as Ubuntu/armhf itself.  Some are running this. So, this may require another build in the PPA matrix, if we can figure out how to do this (two separate armhf builds for each distro?).
On the other hand, this may become a "policy" decision, more than a "code" decision.  That is to say: 
"What is our default target for PPA armhf builds?  Should it be ubuntu, or should it be raspbian?  Might not be able to have both..."

Anyway, in the code, I took the pragmatic solution that works, at least for O56.

leamas, 2021-12-08:
First: 056 is out, and it works. That's a good starting point. Question is then how we proceed towards next release (5.8?).

I share your view that this is more about policy than code. For better or worse the runtime test breaks the underlying policy which was that the correct platform should be determined in build time. And that's why I filed this issue.

As a starter, I presume that the really important armhf platform is RPi-3. And on RPi-3, Raspbian is the dominating OS, right? That is, raspbian:10 and :11  is if not the default armhf target (whatever that would mean) the main target we build plugins for.

Yes, there are other OS:es used on RPi-3: Ubuntu, Debian and what not. Part of the situation is also  that from now on RPi-3 is probably fading away, slowly being replaced by RPi-4 and aarch-64. 

Taking all this into consideration, do we see a need to start building armhf plugins and main OpenCPN  for other OS:es than  Raspbian 10 and 11? I have so far assumed no. Am I wrong?

transmitterdan, 2021-12-08:
My 2 cents. The Ubuntu team has excellent support of RPi 4 with x86 support and now 64-bit is what many will want. So, yes, I think we have to support Raspbian and Ubuntu 32/64 at least. I realize it requires building more plugin varieties but assuming it can be automated.

If it's too great a burden, then Raspbian is dominant and we have to support that. Ubuntu users may be able to build for themselves. But I think Raspbian will decrease its share fairly rapidly given the robust Ubuntu support for RPi 4 variants of the hardware. Ubuntu just runs out of the box on RPi 4.

leamas, 2021-12-08:
Here is really two things: RPi-3 and RPi-4.

As for RPi-3, my current understanding is that Raspbian is dominating.  I have no solid figures, though. Anyone?

When it comes to RPi-4 and aarch64,  the market here seems to be much more more fragmented. Our current position here is that we build flatpak only which runs on all the OS:es actually used  here. If this should be discussed it's kind of another issue, the issue here is the armhf packages.

Hakansv, 2021-12-08:
For OpenCPN my view from here is Openplotter, with Rasbian, is the main image for a new user. First it's a convenient way to use Linux for a user coming from Windows. All updates are taken care of by the "OpenCPN installer" and "Openplotter settings" apps. (No sudo apt -nothing.)  And, not less, since that set up with SignalK and all connection apps make it possible to include both NMEA0183 and N2k devices in a, not straight forward but possible, understandable way.
Since the Openplotter's OpenCPN installer, on the Buster OS, now use some Ubuntu xx.xx for both OCPN and the plugins it may made things easier?
RPi3 and/or RPi4? A new unit may be RPi4. Many existing installations use RPi3.
If Openplotter may be based on BullsEye? Possible but I don't know. Please include Roberto..
Håkan

leamas, 2021-12-08:
Googling around it seems that OpenPlotter is based on Raspbian 10 a. k. a. Buster. Have pinged Roberto, he is the one who can confirm or falsify this. Let's hope he can chime in .

rgleason, 2021-12-08:
You are getting into Packaged RPI Setups here.
Openplotter is one.
[Bareboat Necessities is another](https://www.cruisersforum.com/forums/f134/macos-pim-plugins-testing-258613.html)
[RPI in Manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:raspberrypi_rpi2)
[Packaged OpenCPN which needs updating](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_software:opencpn-packages)

Be careful about going down the rabbit hole here.

bdbcat, 2021-12-08:
Roberto is on holiday this week.
But he is known to be working on Bullseye now, for future release.

leamas, 2021-12-08:
I happen to know that Bullseye in this context refers to Raspbian 11/Bullseye, not vanilla Debian 11.  I also find evidence in my personal email logs that current version is based on Raspbian 10/Buster.

The more I look into this it seems that the armhf builds we should do is Raspbian 10 and 11.  It is what all integrated solutions including OpenPlotter seems to use, and is also the #1 listed when there is talk about RPi-3 OS:es.

leamas, 2021-12-08:
The remaining question is then if there is a need to plan for other armhf OS:es during the 5.6 cycle. I don't see the need. Does anyone else?

bdbcat, 2021-12-09:
O56 is done.  It runs well on RaspiOS/Bullseye, using the Focal/armhf build.

For plugins, all we need do is to build raspbian11 versions for all plugins.  I have tested O56(focal/armhf) on RaspiOS, with a test plugin targeting raspbian11.  It is detected and installed properly.  So we are good to go.

And there is no time to waste.  The recommended version for new installs on Rpi3/4 is now RaspiOS 11.
https://www.raspberrypi.com/software/operating-systems/

mgrouch, 2021-12-09:
There is no existing prepackaged distribution with opencpn for Bullseye yet.

Both OpenPlotter and Bareboat Necessities OS (BBN OS) are based in current releases on Buster.

So there is a need for buster/bionic plugins releases for (at least armhf, arm64 in addition would be better) 5.6.0. If users of BBN OS or OpenPlotter upgrade
to 5.6.0 a lot of plugins are going to be broken for them.

Bullseye (or just released Raspberry OS) has a share of issues.
It's not really fully ready for production. Cameras support has issues,
hardware acceleration in chromium, wxWidgets with dark themes and many other
software which used to be part of OpenPlotter. SignalK has issues due to nodeJS upgrade with number of plugins on Bullseye too.

You might see some crippled prepackaged distros with OpenCPN on Bullseye but they will not be really as complete as Buster/Bionic based due to Bullseye released just recently.

Thanks

mgrouch, 2021-12-09:
Another idea is try to automate a build of every git push on every platform.
I think this can be done using circleci. It allows executing almost arbitrary scripts. If you need
to build on Launchpad for Ubuntu/arm/Raspberry you can just invoke upload of packaging to
launchpad from circleci. Basically invoking one continuous build system from another for the reason that single
one doesn't serve all cases.

leamas, 2021-12-09:
@mgrouch:
> Another idea is try to automate a build of every git push on every platform for every plugin too

That is what we are doing. Still, there are limits on the number of platforms we can support, each one adds build time,  dependencies and overall complexity.

leamas, 2021-12-09:
@bdbcat:

> Both OpenPlotter and Bareboat Necessities OS (BBN OS) are based in current releases on Buster.

As I understand it, OpenPlotter is based on Raspbian 10 which is downstream from Debian Buster. Am I wrong?

@mgrouch:

Same question for BBN: Is it based on vanilla Debian Buster or Raspbian/Buster?

sailoog, 2021-12-09:
Hi all,

You are right, the current stable OpenPlotter version 2 is based on Raspbian10/Buster and OpenCPN is installed from PPA Bionic/armhf build. This method has been working like a charm from Raspbian 9/stretch and the transition to Raspbian 10/buster was easy.

We are already working on OpenPlotter 3 based on Raspbian 11/bullseye. Transition from Raspbian 10 to 11 is being more problematic this time and they have even released a legacy version. Besides this, we have to add 64bit to the equation and things are getting a bit messy. So this time we want to make the OpenCPN installer app in OpenPlotter 3 ready for any scenario and the wildcard is the flatpak installation.

In OpenPlotter 3 based on Raspbian 11/bullseye, we will offer installing OpenCPN from PPA focal/armhf build and optionally also from flatpak (they can even coexist in the same system). As usual we will use a GUI for both options, no terminals involved. This way we will be ready for any decision is taken about plugins and ready for the 64bits transition.

Said this, I share Dave's opinion, the easiest way should be to start building plugins for Raspbian 11/bullseye to be installed from a PPA focal/armhf installation as usual and see what finally happens with Raspbian 64bits that seems to be far away yet.

sailoog, 2021-12-09:
Sorry I forgot: In my opinion we could stop building OpenCPN plugins for Raspbian 9/stretch, keep plugins for Raspbian 10/buster for some time and start building plugins for Raspbian 11/bullseye

leamas, 2021-12-09:
> the current stable OpenPlotter version 2 is based on Raspbian10/Buster 

Assuming the same is true for BBN (I would be surprised otherwise) this sort of settles it.

Where plugins are created and distributed is not really important here. Bullseye plugins are already in the pipe and will be available shortly (some already are).

Once again: this is about about the path after the 5.6 release. Here, would it not be possible to produce  plugins which has the correct target `raspbian:10` (or 11). And in next release drop the runtime system identification which is the very core of this issue thus reverting to a simpler and more robust state?

leamas, 2021-12-09:
Quoting @bdbcat  [long ago](https://github.com/OpenCPN/OpenCPN/issues/2492#issuecomment-988407312)
>  But consider that there may be other environments where the ubuntu-armhf build would be useful, such as Ubuntu/armhf itself.

The sound preparations for such a scenario is IMHO that we don't (ab)use the `ubuntu-armhf` target with other usage. That is, we use the `raspbian-armhf` target for Raspbian builds, `ubuntu-armhf`for Ubuntu builds and no runtime code which mix them together.

rgleason, 2021-12-09:
Mgrouch
 Building plugins continuously is impractocal for several reasons and is not a good solution in my opinion. . Wastes limited opensource computing resources. We are on the edge as it is.

rgleason, 2021-12-09:
I think OpenCpn, OpenPlotter and BBN should discuss, and chart a simple course and adhere to it, one 32bit and one 64 bit  perhaps flatpak and whatever rasbian, but keep the number of separate builds low, because this affects everything including plugins and support becomes even worse.

bdbcat, 2021-12-09:
"If users of BBN OS or OpenPlotter upgrade to 5.6.0 a lot of plugins are going to be broken for them."

Let's get specific here, and not general.  What plugins are broken on rPI with OCPN56?
Is there a list?

bdbcat, 2021-12-09:
"Here, would it not be possible to produce plugins which has the correct target raspbian-armhf:10 (or 11)."
Is that not exactly what we are doing now?

" we use the raspbian-armhf target for Raspbian builds, ubuntu-armhffor Ubuntu builds"
It is not obvious yet (to me) how to convince Launchpad to build two flavors of armhf build for each Ubuntu distro.  Further, it is not obvious (to me) how the canonical linux package managers will handle two flavors of the same $ARCH at install time.

leamas, 2021-12-09:
@bdbcat: OK, this is indeed the core issue. 

First, I think here has been some confusion, at least for me. This is about the core OpenCPN packages, not the plugins. The plugins uses the correct target.

Is there any need for Ubuntu/armhf package packages? We  don't build plugins for this today, right?

bdbcat, 2021-12-09:
core OCPN:
"Is there any need for Ubuntu/armhf packages?"
IMHO, yes.  There are users running vanilla Ubuntu on rPi3/4.

Plugins:
"We don't build plugins for this today, right?"
Right.  But I expect that Raspbian plugins will load, with the correct "compatibility" setting in settings->Plugins->Settings. Though I have not tested specifically.

leamas, 2021-12-09:
> IMHO, yes. There are users running vanilla Ubuntu on rPi3/4.

On RPi-4 we have Flatpak, it might make sense to stick to that so that build matrix is kept sane

On RPi-3? Really? A large enough user base to care about?

bdbcat, 2021-12-09:
Let's stick to armhf discussion first.
It appears (to me) that most rPI4 systems are still running 32 bit kernels.  Either Ubuntu, or raspbian. There has been some success in convincing users that 64 bits (for kernel or apps)  brings no detectable performance benefits for the apps commonly run on rPi.
Also, there is general resistance to flatpak in the average rPi user base, when there is a native alternative.

Moving to arm64:
It's true that we have no plan to create plugins specifically for rPi-64, either Ubuntu or raspbian/64(whenever it comes).  So flatpak is clearly the way forward here.  Message:   "On rPi/arm64, if you want OCPN with plugins, then you must use flatpak".

leamas, 2021-12-09:
OK. So there is/might be a need for armhf Ubuntu packages.

Here is also the problem that current 5.6 OpenCPN package on Raspbian loads both ubuntu-armhf and raspbian plugins. This is to overcome the fact that we are using packages built for Ubuntu on Raspbian. This happens to work, but carries risks since there is no ABI compatibility promises between the two.

What would bring some order into this is if we somehow could create a new build chain which builds Raspbian OpenCPN on a Raspbian host.

mgrouch, 2021-12-10:
> 

> @bdbcat:
> 
> > Both OpenPlotter and Bareboat Necessities OS (BBN OS) are based in current releases on Buster.
> 
> As I understand it, OpenPlotter is based on Raspbian 10/Buster which is downstream from Debian 10/Buster. Am I wrong?
> 
> @mgrouch:
> 
> Same question for BBN: Is it based on vanilla Debian Buster or Raspbian/Buster?

BBN is based on Raspberry Pi OS/ Buster

mgrouch, 2021-12-10:
> "If users of BBN OS or OpenPlotter upgrade to 5.6.0 a lot of plugins are going to be broken for them."
> 
> Let's get specific here, and not general. What plugins are broken on rPI with OCPN56? Is there a list?

I think it's better to go from opposite direction and start building a list of the ones which work.

bdbcat, 2021-12-10:
"Here is also the problem that current 5.6 OpenCPN package on Raspbian loads both ubuntu-armhf and raspbian plugins"
It probably does.  But we do not build nor publish any ubuntu-armhf plugins by our current CI.  Only Raspbian.
So if a ubuntu-armhf user wants a plugin, the rasbpian-armhf build will very likely run.  But you are right, no guarantees.

bdbcat, 2021-12-10:
"What would bring some order into this is if we somehow could create a new build chain which builds Raspbian OpenCPN on a Raspbian host."
Even if we did that, I still do not see how we integrate .deb file this into the package management tools (apt, etc..) on the rPi device, without a truly private repo of some sort.

mgrouch, 2021-12-10:
I've started combining the list 

Here: https://github.com/bareboat-necessities/opencpn-plugins-bundle/blob/main/plugins-armhf-compat-5_6_0.txt

mgrouch, 2021-12-10:
I've marked with '-' the ones which break OpenCPN on armhf after upgrade to 5.6.0 and update of plugins from PIM.
Check here:

https://github.com/bareboat-necessities/opencpn-plugins-bundle/blob/main/plugins-armhf-compat-5_6_0.txt

mgrouch, 2021-12-10:
> "What would bring some order into this is if we somehow could create a new build chain which builds Raspbian OpenCPN on a Raspbian host." Even if we did that, I still do not see how we integrate the resulting .deb file into the package management tools (apt, etc..) on the rPi device, without a truly private repo of some sort.

Ideally OpenCPN should be pushed via official Debian channel. Then it would cover raspberry and probably Ubuntu distros

leamas, 2021-12-10:
> Ideally OpenCPN should be pushed via official Debian channel. Then it would cover raspberry and probably Ubuntu distros

I have initiated the process to get 5.6.0 into Debian. The problem is the sluggish Debian release cycle. I will probably get into Sid in a reasonable time. But that just means 5.6.0 will hit upcoming Debian Bookworm/12 and Ubuntu 22.04.

Raspbian is downstream from Debian, I have no idea if anything is updated after being released. Raspbian Bullseye should have 5.2.4 packaged, havn't checked..

Once 5.6.0 is in Sid, I will try to make it available also in Bullseye/backports. If this succeeds  a reasonable updated version should be available for Debian users. However, using Debian backports is not officially supported on Raspbian. That said, it should be less of a gamble than using Ubuntu which is longer from Raspbian so to speak.

leamas, 2021-12-10:
> Even if we did that, I still do not see how we integrate the resulting .deb file into the package management tools (apt, etc..) on the rPi device, without a truly private repo of some sort.

Indeed. The alternative is to just download it from somewhere and use apt to install it (so it pulls the deps). 

Seems that here are four options:
  - Use the PPA, risks with potential Raspbian incompatibilities and clashes if there is a need to build Ubuntu/armhf packages.
  - Use an official channel like Debian backports. Longer delays to release, still ABI risks but less so than using Ubuntu.
  - Use the official packaging. No ABI risks, but roughly one release behind. 
  - Build a  .deb package, having it available to be downloaded and installed "manually" using apt. No ABI risks, worse UX.

None of these alternatives shines, for sure.

rgleason, 2021-12-10:
Why not just build a deb package. RPI users understand this and expect it even.

sailoog, 2021-12-10:
I do not like the manually installable deb package solution because this means problems with dependencies and lots of non updated installations. The first reaction after downloading a deb for non linux users will be double-click on it and dependencies problems will start. We can solve all this with the openplotter-opencpn-installer app but non openplotter users will have these problems.

If we are building Raspbian packages to distribute as installable deb files why not adding a specific repo to cloudsmith for Debian/Raspbian. Users will only have to add the source to apt sources just like they do now with the PPA solution and install and update it just using "apt install opencpn" as usual.

leamas, 2021-12-10:
> I do not like the manually installable deb package solution because this means problems with dependencies and lots of non updated installations. 

Agreed. As I said, none of the alternatives really shines...

rgleason, 2021-12-10:
```
I've started combining the list

Here: https://github.com/bareboat-necessities/opencpn-plugins-bundle/blob/main/plugins-armhf-compat-5_6_0.txt
```

You're going to need an automatic way to keep this up to date as they change quite frequently.

rgleason, 2021-12-10:
`If we are building Raspbian packages to distribute as installable deb files why not adding a specific repo to cloudsmith for Debian/Raspbian. Users will only have to add the source to apt sources just like they do now with the PPA solution and install and update it just using "apt install opencpn" as usual.`
The two templates (Testplugin and ShipDriver) will need upload- script modifications to upload to the 3 dedicated cloudsmith OpenCPN/debian-raspian repositories  (-alpha, -beta, -prod).  The cmakelists.txt files may need adjustment.

bdbcat, 2021-12-10:
Rick....
Hold your horses.  No decisions have been made yet.

leamas, 2021-12-10:
@rick: 

> The two templates (Testplugin and ShipDriver) will [...]

As for plugins, everything already works as it should, we build and deliver Raspbian plugins through the normal build chains, no changes are required as far as I know.

This issue is about building and distributing the Raspbian OpenCPN package in a more robust way. 

@sailoog : 
> If we are building Raspbian packages to distribute as installable deb files why not adding a specific repo to cloudsmith for Debian/Raspbian [...]

Yes, agreed, if we go for .deb file generation this is probably a good solution.

Thinking about it, I'm a bit mystified by the missing dependencies you mention when a user clicks on a .deb file in the browser. I would expect that when the .deb file is 'opened' it would be installed by apt. And apt will no doubt pull in whatever dependencies needed. 

Are you saying that  when a user clicks on a .deb package it is installed without dependencies in Raspbian?

rgleason, 2021-12-10:
Yes, Alec, what you say is true, but I thought Sailoog wanted to process all raspian deb files somehow and package them with OpenCPN.  In that case wouldn't a CS raspbian repository that is up to date be useful?  I suppose sailoog could use Mike's plugin list if it were automatically kept up to date.... ah now I understand, that is all available in ocpn-plugins.xml!!!   Sorry for cluttering.

rgleason, 2021-12-10:
@bdbcat   Not going to do anything.

bdbcat, 2021-12-10:
mgrouch...
To make this easier, will you start a new issue with your plugin list?
Already I have questions.  (e.g. GRIB fails?  I think not likely, and not by my test 5 minutes ago)
Thanks

mgrouch, 2021-12-10:
> 

I’ve had few crashes with Grib, but it mostly works. Others marked ‘-‘ 
crash right away.

rgleason, 2021-12-10:
Back to the subject...

bdbcat, 2021-12-10:
Can we agree on the following?
There is **little** risk that the ubuntu-Bionic-armhf flavor of OCPN 5.6 as built by Launchpad will fail on Rasbpian-armhf.  After all, that is exactly what OpenPlotter has been shipping with O524 (i.e. OCPN 5.2.4 built on Launchpad as "Bionic-armhf") 
By little, I mean "acceptable", given the apparent success of OpenPlotter with O524.

leamas, 2021-12-10:
> There is little risk that the ubuntu-Bionic-armhf flavor of OCPN 5.6 as built by Launchpad will fail on Rasbpian-armhf.

Agreed. 

> By "little", I mean "acceptable"

I have no worries whatsoever with OpenCPN/5.6 and Raspbian/10. What worries me what happens when Raspbian and/or OpenCPN is updated. Since we have no guarantees, this will cast a shadow over any update.

sailoog, 2021-12-10:
> Thinking about it, I'm a bit mystified by the missing dependencies you mention when a user clicks on a .deb file in the browser. I would expect that when the .deb file is 'opened' it would be [...]

Yes, I can not confirm that behaviour right now but I am quite sure dpkg is used to install .debs when double-click on them and there is not dependencies management. You need to run "sudo apt-get -f install" to fix things after that.

> Yes, Alec, what you say is true, but I thought Sailoog wanted to process all raspian deb files somehow and package them with OpenCPN. In that case wouldn't a CS raspbian repository that is up to [...]
 no, no. I just proposed replacing the PPA repo by a cloudsmith repo but only for Raspbian and only if building specific packages for Raspbian is decided.

Actually my position on this matter is do nothing. We had this conversation by email short time ago with bdbcat and leamas because I proposed completely replacing PPA by cloudsmith but bdbcat made me see the pros of keeping the PPA. I also share leamas opinion, plugins are working right, and dbdcat has finally focused this debate, we should just evaluate the risk of keeping PPA and its ubuntu flavours as source for Raspbian.

If we agree that the risk is acceptable (I also agree) but we are not comfortable in this gray area (I am not), maybe we should prepare a plan B in case the sky falls on our heads someday. OpenPlotter 3 will be ready for any scenario and even it will be ready to change from PPA to cloudsmith at any time. We should be worried only for non OpenPlotter users.

leamas, 2021-12-10:
> Actually my position on this matter is do nothing

I actually agree.  There is just two gotcha's: 

  - Should we stick to this scheme also for next release, again gamble that it will work?
  -  If we do stick to current scheme we are basically saying that Raspbian plugins works in Ubuntu. This scheme is IMHO better implemented by hardcoding it explicitly rather than a  runtime test which tweaks the host target.

leamas, 2021-12-10:
A tentative patch which just treats Raspbian plugins as Ubuntu ones. Untested, and does  not handle versions although it should  (raspbian:10 -> Ubuntu 21.04 etc). For the sake of discussion...

---

    diff --git a/src/PluginHandler.cpp b/src/PluginHandler.cpp
    index 19f0037e0..2732d62e3 100644
    --- a/src/PluginHandler.cpp
    +++ b/src/PluginHandler.cpp
    @@ -147,6 +147,8 @@ class Plugin {
     public:
       Plugin(const PluginMetadata& metadata) {
         m_abi = metadata.target;
    +    // We treat raspbian plugins as ubuntu -- #2492
    +    ocpn::replace(m_abi, "raspbian", "ubuntu"); 
         m_abi_version = metadata.target_version;
         m_major_version = ocpn::split(m_abi_version.c_str(), ".")[0];
         m_name = metadata.name;
    @@ -223,25 +225,6 @@ class Host {
         return false;
       }
     
    -  // Check the plugin for host run-time ID match.
    -  // This test is necessary for cross-built hosts.
    -  // For example, Ubuntu PPA builds for armhf define PKG_TARGET as ubuntu-armhf
    -  // But run-time reports as raspbian on "Raspberry Pi OS".
    -  bool is_plugin_compatible_runtime(const Plugin& plugin) const {
    -    OCPN_OSDetail* os_detail = g_Platform->GetOSDetail();
    -    const std::string host_osd_abi = os_detail->osd_ID + "-" + os_detail->osd_arch;
    -    wxLogDebug("Checking for compatible run-time, host_osd_abi: %s : %s", host_osd_abi, os_detail->osd_version);
    -    wxLogDebug("   plugin_abi + version: %s %s", plugin.abi(), plugin.major_version());
    -    if (host_osd_abi == plugin.abi()) {
    -      wxLogDebug("   plugin_version: %s", major_version());
    -      if (os_detail->osd_version == plugin.major_version()) {
    -        return true;
    -      }
    -    }
    -    return false;
    -  }
    -
    -
       const std::string& abi() const { return m_abi; }
     
       const std::string& abi_version() const { return m_abi_version; }
    @@ -326,13 +309,6 @@ bool PluginHandler::isCompatible(const PluginMetadata& metadata, const char* os,
         rv = true;
         wxLogDebug("Found Ubuntu version matching Debian host");
       }
    -#ifdef ocpnARM
    -  //TODO  This conditional may not be needed.  Test on O57+
    -  else if(host.is_plugin_compatible_runtime(plugin)) {
    -    rv = true;
    -    wxLogDebug("Found host OCPN_OSDetail run-time match");
    -  }
    -#endif
       DEBUG_LOG << "Plugin compatibility check Final: "
                 << (rv ? "ACCEPTED: " : "REJECTED: ") << metadata.name;
       return rv;

bdbcat, 2021-12-11:
"This scheme is IMHO better..."
Convince me.  I'll listen to alternatives....

leamas, 2021-12-11:
It's no hurry here, the Swedish saying is that there is no cow on the ice.  I need some more nights sleep to land. Notes:

  - When we state that we can run Raspbian plugins on an Ubuntu host we are actually saying that both Ubuntu and Raspbian can run Debian plugins, the common upstream and denominator.
  -  There is nothing special with armhf. If Debian plugins works on Ubuntu, it's true on all platforms
  - The current code effectively changes the host target platform.  However, it is not coordinated with the otherwise used scheme in `CompatOs::CompatOs()` using hardcoded default, environment settings and configuration file settings. The result is hard to review but I think it breaks the code to override the host target.
  - What is the point building Raspbian plugins when there is no Raspbian host? What we have is an Ubuntu armhf host., we could as well build ubuntu-armhf plugins. Shouldn't change compatibility since plugins by definition have the same deps as main OpenCPN.

Not clear to me what all this means.

leamas, 2021-12-13:
OK, some nights sleep. I see two possible ways to clean up the current situation

The first is to continue using Ubuntu-armhf packages on Raspbian. Doing so, the simple solution is to build ubuntu-armhf plugins instead of Raspbian.  This works out of the box on both Raspbian and native Ubuntu-armhf hosts. If we follow this path we could then just remove the runtime test in next release. It is simple, but a gamble with respect to long-time Ubuntu - Raspbian compatibility.

The second option would be to use Debian plugins. This would become possible if/when the packaged version becomes available on Debian/11 and perhaps also 10.  There is no point to think about this until we  know how the official packaging of 5.6.0 turns out.

What I propose:
  - Stop building Raspbian plugins and start building ubuntu-armhf instead.
  - While on it, also add Debian 11/armhf to the plugin build matrix if the packaged version seems to land in time in Debian 11 (should be known in a week or two).
  - When the plugin Raspbian - Ubuntu/armhf switch is completed, the runtime test in main OpenCPN is removed

bdbcat, 2021-12-23:
@leamas...
Related to your proposal:
https://www.cruisersforum.com/forums/f134/opencpn-5-6-0-released-258634-19.html#post3542238

This is something of the inverse to your proposal.  That is, running raspbian plugins on generic ubuntu image.
However, I think your latest proposal makes better sense, with little risk.
But how to bridge to the new plan without breaking current useres?
More thinking....

leamas, 2021-12-23:
I actually think it's quite simple. Our current OpenCPN programs are Ubuntu, and they will accept ubuntu plugins. So, to repeat the transition schema:

  - We change the plugin build chains to build ubuntu/armhf plugins instead of Raspbian/armhf. These will be accepted buy current OpenCPN/armhf hosts as-is.
  - When all plugins have been switched to make ubuntu plugins, we just remove the runtime test which is the issue here. Practically, thi could be done in next OpenCPN release, there is no rush.

leamas, 2021-12-23:
On  a sidenote, we should probably try to build Debian plugins which are accepted by Ubuntu. This is the safe and natural route, but it's another issue.

[7 more replies]

closed 2023-02-04 by leamas