OpenCPN/OpenCPN #2494: Plugins for 5.6.0

mgrouch, 2021-12-10:
List of broken ones for armhf marked ‘-‘

see here:

https://github.com/bareboat-necessities/opencpn-plugins-bundle/blob/main/plugins-armhf-compat-5_6_0.txt

bdbcat, 2021-12-10:
Thanks for moving the issue.
I'll start some testing, also.  Since there was no ABI change on wxWidgets between O5.2.4, and O5.6, I'm pretty sure that many of the failures reported would also fail on O5.2.4.
But either way, if a plugin crashes immediately, it should and will be removed from the catalog until it is fixed.

rgleason, 2021-12-10:
Mike, 

Thank you for the helpful list and checking raspbian operation. The plugins mentioned below are ones I help to maintain.  I would like to get these working in raspbian., it makes no sense to have them if they do not work. Can you advise what the issue is with  AISradar, DashboardTactics, Logbook, Polar and Tactics?  Even better, if you clone my repos and can fix them and make a PR to one of my github  repositories that would be great.

The following failing plugins have not been updated in a long time and probably should not be used until they are
Admiralty Tide, rtlsdr, Opencpndebugger (I thought this was for windows debugging).  At some point I will update these.

mgrouch, 2021-12-10:
AIS Radar View issue when clicking on plugin toolbar icon

ASSERT INFO:
../src/gtk/dcclient.cpp(2038): assert "m_window" failed in DoGetSize(): GetSize() doesn't work without window

BACKTRACE:
[1] wxClientDCImpl::DoGetSize(int*, int*) const
[2] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[9] wxWindowBase::SendSizeEvent(int)
[10] wxTopLevelWindowGTK::Show(bool)
[11] wxDialog::Show(bool)
[12] MyFrame::OnToolLeftClick(wxCommandEvent&)
[13] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[14] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[15] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[16] wxEvtHandler::TryHereOnly(wxEvent&)
[17] wxEvtHandler::DoTryChain(wxEvent&)
[18] wxEvtHandler::ProcessEvent(wxEvent&)
[19] wxEvtHandler::ProcessPendingEvents()
[20] wxAppConsoleBase::ProcessPendingEvents()
[21] wxApp::DoIdle()

mgrouch, 2021-12-10:
DashboardTactics issue when enabling plugin:

ASSERT INFO:
../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime

BACKTRACE:
[1] wxDateTime::IsDST(wxDateTime::Country) const
[2] wxDateTime::MakeTimezone(wxDateTime::TimeZone const&, bool)
[3] PlugInManager::SendPositionFixToAllPlugIns(GenericPosDatEx*)
[4] MyFrame::OnFrameTimer1(wxTimerEvent&)
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::ProcessEventLocally(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[12] wxTimerImpl::SendEvent()

mgrouch, 2021-12-10:
Logbook crashes on restart:

ASSERT INFO:
/usr/include/wx-3.0/wx/arrstr.h(178): assert "nIndex < m_nCount" failed in Item(): wxArrayString: index out of bounds

BACKTRACE:
[1] logbookkonni_pi::LoadConfig()
[2] logbookkonni_pi::Init()
[3] PlugInManager::LoadPlugInDirectory(wxString const&, bool, bool)
[4] PlugInManager::LoadAllPlugIns(bool, bool)
[5] MyFrame::OnInitTimer(wxTimerEvent&)
[6] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[7] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[8] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[9] wxEvtHandler::TryHereOnly(wxEvent&)
[10] wxEvtHandler::ProcessEventLocally(wxEvent&)
[11] wxEvtHandler::ProcessEvent(wxEvent&)
[12] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[13] wxTimerImpl::SendEvent()

mgrouch, 2021-12-10:
Polar seems loads twice.

previous was 

/usr/lib/opencpn/libPolar_pi.so

and new one is 

./.local/lib/opencpn/libpolar_pi.so

Notice that in first case P is uppercase and in second it is lowercase

So OpenCPN thinks they are 2 different plugins. Icons are duplicated as so on.

mgrouch, 2021-12-10:
Tactics failed on enable:

ASSERT INFO:
../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime

BACKTRACE:
[1] wxDateTime::IsDST(wxDateTime::Country) const
[2] wxDateTime::MakeTimezone(wxDateTime::TimeZone const&, bool)
[3] PlugInManager::SendPositionFixToAllPlugIns(GenericPosDatEx*)
[4] MyFrame::OnFrameTimer1(wxTimerEvent&)
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::ProcessEventLocally(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[12] wxTimerImpl::SendEvent()

mgrouch, 2021-12-10:
OCPNDebugger fails on opening from toolbar:

ASSERT INFO:
../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime

BACKTRACE:
[1] wxDateTime::Format(wxString const&, wxDateTime::TimeZone const&) const
[2] ocpndebugger_pi::SetPositionFixEx(PlugIn_Position_Fix_Ex&)
[3] PlugInManager::SendPositionFixToAllPlugIns(GenericPosDatEx*)
[4] wxThreadHelperThread::~wxThreadHelperThread()
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::ProcessEventLocally(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[12] wxTimerImpl::SendEvent()

mgrouch, 2021-12-10:
vfkaps fails on starting from toolbar icon:

ASSERT INFO:
../src/common/sizer.cpp(1401): assert "Assert failure" failed in DoInsert(): too many items (7 > 1*6) in grid sizer (maybe you should omit the number of either rows or columns?)

BACKTRACE:
[1] wxGridSizer::DoInsert(unsigned int, wxSizerItem*)
[2] wxSizer::Insert(unsigned int, wxSizerItem*)
[3] MyFrame::OnToolLeftClick(wxCommandEvent&)
[4] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[5] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[6] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[7] wxEvtHandler::TryHereOnly(wxEvent&)
[8] wxEvtHandler::DoTryChain(wxEvent&)
[9] wxEvtHandler::ProcessEvent(wxEvent&)
[10] wxEvtHandler::ProcessPendingEvents()
[11] wxAppConsoleBase::ProcessPendingEvents()
[12] wxApp::DoIdle()

mgrouch, 2021-12-10:
EngineDashboard fails on enabling in PIM:

/usr/include/wx-3.0/wx/datetime.h(1886): assert "IsValid()" failed in Add(): invalid wxDateTime

with no stacktrace

rgleason, 2021-12-10:
Mike  I assume these Plugins are all sourced from PIM (Plugin Manager), correct? Also which Raspbian version are you using?  Do you know if the other version fails in the same way?

mgrouch, 2021-12-10:
@bdbcat 
@rgleason 
> Mike I assume these Plugins are all sourced from PIM (Plugin Manager), correct? Also which Raspbian version are you using? Do you know if the other version fails in the same way?

Raspberry Pi OS / Buster / armhf

before upgrading to 5.6.0 and updating everything from PIM they all worked fine.

mgrouch, 2021-12-10:
libwxbase3.0-0v5:armhf                 3.0.4+dfsg-8                            armhf        wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxgtk3.0-0v5:armhf                  3.0.4+dfsg-8                            armhf        wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)

wxWidgets are still 3.0.4 on Raspberry / Buster

mgrouch, 2021-12-10:
OpenCPN is linked against GTK2 on Raspberry / Buster

`
user@lysmarine:~ $ ldd /usr/bin/opencpn
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v8l.so (0xf72e8000)
	libX11.so.6 => /lib/arm-linux-gnueabihf/libX11.so.6 (0xf71a4000)
	libcurl.so.4 => /lib/arm-linux-gnueabihf/libcurl.so.4 (0xf711f000)
	libarchive.so.13 => /lib/arm-linux-gnueabihf/libarchive.so.13 (0xf7073000)
	liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0xf7047000)
	libelf.so.1 => /lib/arm-linux-gnueabihf/libelf.so.1 (0xf701d000)
	libwx_gtk2u_gl-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk2u_gl-3.0.so.0 (0xf6ffe000)
	libwx_baseu_net-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu_net-3.0.so.0 (0xf6fb6000)
	libwx_baseu_xml-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu_xml-3.0.so.0 (0xf6f99000)
	libwx_gtk2u_html-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk2u_html-3.0.so.0 (0xf6eea000)
	libwx_gtk2u_adv-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk2u_adv-3.0.so.0 (0xf6d7b000)
	libwx_gtk2u_aui-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk2u_aui-3.0.so.0 (0xf6cf9000)
	libwx_gtk2u_core-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0 (0xf684f000)
	libwx_baseu-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0 (0xf660f000)
	libGL.so.1 => /lib/arm-linux-gnueabihf/libGL.so.1 (0xf6582000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0xf6551000)
	libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0xf6521000)
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xf64fc000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xf64e5000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xf64bb000)
	libpango-1.0.so.0 => /lib/arm-linux-gnueabihf/libpango-1.0.so.0 (0xf646e000)
	libgobject-2.0.so.0 => /lib/arm-linux-gnueabihf/libgobject-2.0.so.0 (0xf6411000)
	libpangocairo-1.0.so.0 => /lib/arm-linux-gnueabihf/libpangocairo-1.0.so.0 (0xf63f6000)
	libexif.so.12 => /lib/arm-linux-gnueabihf/libexif.so.12 (0xf63b3000)
	libcairo.so.2 => /lib/arm-linux-gnueabihf/libcairo.so.2 (0xf62ba000)
	libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0xf6279000)
	libtinyxml.so.2.6.2 => /lib/arm-linux-gnueabihf/libtinyxml.so.2.6.2 (0xf6258000)
	libsqlite3.so.0 => /lib/arm-linux-gnueabihf/libsqlite3.so.0 (0xf6150000)
	libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0xf6009000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xf5f87000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xf5f5a000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xf5e0c000)
	/lib/ld-linux-armhf.so.3 (0xf7bea000)
	libxcb.so.1 => /lib/arm-linux-gnueabihf/libxcb.so.1 (0xf5ddd000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xf5dca000)
	libnghttp2.so.14 => /lib/arm-linux-gnueabihf/libnghttp2.so.14 (0xf5d9b000)
	libidn2.so.0 => /lib/arm-linux-gnueabihf/libidn2.so.0 (0xf5d6f000)
	librtmp.so.1 => /lib/arm-linux-gnueabihf/librtmp.so.1 (0xf5d46000)
	libssh2.so.1 => /lib/arm-linux-gnueabihf/libssh2.so.1 (0xf5d10000)
	libpsl.so.5 => /lib/arm-linux-gnueabihf/libpsl.so.5 (0xf5cf0000)
	libssl.so.1.1 => /lib/arm-linux-gnueabihf/libssl.so.1.1 (0xf5c71000)
	libcrypto.so.1.1 => /lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0xf5a60000)
	libgssapi_krb5.so.2 => /lib/arm-linux-gnueabihf/libgssapi_krb5.so.2 (0xf5a16000)
	libkrb5.so.3 => /lib/arm-linux-gnueabihf/libkrb5.so.3 (0xf595b000)
	libk5crypto.so.3 => /lib/arm-linux-gnueabihf/libk5crypto.so.3 (0xf591b000)
	libcom_err.so.2 => /lib/arm-linux-gnueabihf/libcom_err.so.2 (0xf5908000)
	libldap_r-2.4.so.2 => /lib/arm-linux-gnueabihf/libldap_r-2.4.so.2 (0xf58b3000)
	liblber-2.4.so.2 => /lib/arm-linux-gnueabihf/liblber-2.4.so.2 (0xf5897000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xf586c000)
	libnettle.so.6 => /lib/arm-linux-gnueabihf/libnettle.so.6 (0xf5828000)
	libacl.so.1 => /lib/arm-linux-gnueabihf/libacl.so.1 (0xf5811000)
	libzstd.so.1 => /lib/arm-linux-gnueabihf/libzstd.so.1 (0xf5778000)
	libbz2.so.1.0 => /lib/arm-linux-gnueabihf/libbz2.so.1.0 (0xf5758000)
	libxml2.so.2 => /lib/arm-linux-gnueabihf/libxml2.so.2 (0xf55db000)
	libgtk-x11-2.0.so.0 => /lib/arm-linux-gnueabihf/libgtk-x11-2.0.so.0 (0xf5221000)
	libgdk-x11-2.0.so.0 => /lib/arm-linux-gnueabihf/libgdk-x11-2.0.so.0 (0xf517e000)
	libgdk_pixbuf-2.0.so.0 => /lib/arm-linux-gnueabihf/libgdk_pixbuf-2.0.so.0 (0xf514c000)
	libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0xf503c000)
	libnotify.so.4 => /lib/arm-linux-gnueabihf/libnotify.so.4 (0xf5025000)
	libXxf86vm.so.1 => /lib/arm-linux-gnueabihf/libXxf86vm.so.1 (0xf5010000)
	libSM.so.6 => /lib/arm-linux-gnueabihf/libSM.so.6 (0xf4ff9000)
	libpng16.so.16 => /lib/arm-linux-gnueabihf/libpng16.so.16 (0xf4fbd000)
	libjpeg.so.62 => /lib/arm-linux-gnueabihf/libjpeg.so.62 (0xf4f77000)
	libtiff.so.5 => /lib/arm-linux-gnueabihf/libtiff.so.5 (0xf4ef7000)
	libGLdispatch.so.0 => /lib/arm-linux-gnueabihf/libGLdispatch.so.0 (0xf4e78000)
	libGLX.so.0 => /lib/arm-linux-gnueabihf/libGLX.so.0 (0xf4e59000)
[... 28 lines cut ...]
	libXdamage.so.1 => /lib/arm-linux-gnueabihf/libXdamage.so.1 (0xf2838000)
	libXfixes.so.3 => /lib/arm-linux-gnueabihf/libXfixes.so.3 (0xf2823000)
	libatk-1.0.so.0 => /lib/arm-linux-gnueabihf/libatk-1.0.so.0 (0xf27f5000)
	libgio-2.0.so.0 => /lib/arm-linux-gnueabihf/libgio-2.0.so.0 (0xf2679000)
	libXinerama.so.1 => /lib/arm-linux-gnueabihf/libXinerama.so.1 (0xf2666000)
	libXi.so.6 => /lib/arm-linux-gnueabihf/libXi.so.6 (0xf2648000)
[... cut ...]
	libp11-kit.so.0 => /lib/arm-linux-gnueabihf/libp11-kit.so.0 (0xf229e000)
	libtasn1.so.6 => /lib/arm-linux-gnueabihf/libtasn1.so.6 (0xf227e000)
	libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0xf2254000)
	libmount.so.1 => /lib/arm-linux-gnueabihf/libmount.so.1 (0xf21f1000)
	libselinux.so.1 => /lib/arm-linux-gnueabihf/libselinux.so.1 (0xf21bf000)
	libgraphite2.so.3 => /lib/arm-linux-gnueabihf/libgraphite2.so.3 (0xf218e000)
	libblkid.so.1 => /lib/arm-linux-gnueabihf/libblkid.so.1 (0xf2136000)

`

mgrouch, 2021-12-10:
Plugins with versions which worked with 5.2.4

`
"AIS Radar view"                   "1.2.13.0"                      "https://dl.cloudsmith.io/public/opencpn/aisradar-prod/raw/names/aisradar_pi-1.2.13.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.2.13.0/aisradar_pi-1.2.13.0-raspbian-armhf-10.tar.gz"
"Celestial Navigation"             "2.3.7.0"                       "https://dl.cloudsmith.io/public/opencpn/celestial-navigation-prod/raw/names/celestial_navigation_pi-2.3.7.0-raspbian-armhf-10-buster-armhf-tarball/versions/v2.3.7.0/celestial_navigation_pi-2.3.7.0-raspbian-armhf-10.tar.gz"
"Climatology"                      "1.4.31.0"                      "https://dl.cloudsmith.io/public/opencpn/climatology-prod/raw/names/climatology_pi-1.4.31.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.4.31.0/climatology_pi-1.4.31.0-raspbian-armhf-10.tar.gz"
"DR"                               "3.1.1.3"                       "https://dl.cloudsmith.io/public/opencpn/dr-prod/raw/names/dr_pi-3.1.1.3-raspbian-armhf-10-buster-armhf-tarball/versions/3.1.1.3/dr_pi-3.1.1.3-raspbian-armhf-10-buster-armhf.tar.gz"
"DashboardTactics"                 "1.6.8.0"                       "https://dl.cloudsmith.io/public/opencpn/dashboardtactics-beta/raw/names/dashboard_tactics_pi-1.6.8.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.6.8.0/dashboard_tactics_pi-1.6.8.0-raspbian-armhf-10.tar.gz"
"Engine Dashboard"                 "1.4.0.0"                       "https://dl.cloudsmith.io/public/steven-adler/engineplugin-prod/raw/names/engine_dashboard_pi-1.4.0.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.4/engine_dashboard_pi-1.4.0.0-raspbian-armhf-10.tar.gz"
"FindIt"                           "1.2.7.0"                       "https://dl.cloudsmith.io/public/opencpn/findit-prod/raw/names/findit_pi-1.2.7.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.2.7.0/findit_pi-1.2.7.0-raspbian-armhf-10.tar.gz"
"GPS Odometer"                     "0.6.4.0"                       "https://dl.cloudsmith.io/public/opencpn/gps-odometer-prod/raw/names/gpsodometer_pi-0.6.4.0-raspbian-armhf-10-buster-armhf-tarball/versions/v0.6.4.0/gpsodometer_pi-0.6.4.0-raspbian-armhf-10.tar.gz"
"Logbook"                          "1.4.22.0+20.v1.4.22.0"         "https://dl.cloudsmith.io/public/opencpn/logbookkonni-prod/raw/names/Logbook-v1.4.22.0-raspbian-10-tarball/versions/1.4.22.0+20.v1.4.22.0/Logbook-v1.4.22.0_raspbian-10-armhf.tar.gz"
"NmeaConverter_pi"                 "1.2.0.2"                       "https://dl.cloudsmith.io/public/dirk-smits/ocpn-plugins-unstable/raw/names/nmeaconverter_pi-1.2.0.2-raspbian-armhf-10-buster-armhf-tarball/versions/1.2.0.2+258.b5236e0/nmeaconverter_pi-1.2.0.2-raspbian-armhf-10-buster-armhf.tar.gz"
"OCPN Draw"                        "1.8.7.1"                       "https://dl.cloudsmith.io/public/opencpn/ocpn_draw-prod/raw/names/ocpn_draw_pi-1.8.7.1-raspbian-armhf-10-buster-armhf-tarball/versions/v1.8.7.1/ocpn_draw_pi-1.8.7.1-raspbian-armhf-10.tar.gz"
"oeRNC Charts"                     "1.2.5"                         "https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/oernc-raspbian-armhf-10-tarball/versions/1.2.5.0.fe52423/oernc_pi-1.2.5-0_raspbian-armhf-10-armhf.tar.gz"
"oeSENC"                           "4.3.0.0"                       "https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/oeSENC-4.3-raspbian-10-tarball/versions/4.3.0.0/oeSENC-4.3.0.0_raspbian-10-armhf.tar.gz"
"otcurrent"                        "2.8.7+844.v2.8.7"              "https://dl.cloudsmith.io/public/opencpn/otcurrent-prod/raw/names/otcurrent-2.8-raspbian-10-tarball/versions/2.8.7+844.v2.8.7/otcurrent-2.8.7+844.v2.8.7_raspbian-10-armhf.tar.gz"
"Plots"                            "2.2.7.0"                       "https://dl.cloudsmith.io/public/opencpn/plots-prod/raw/names/plots_pi-2.2.7.0-raspbian-armhf-10-buster-armhf-tarball/versions/v2.2.7.0/plots_pi-2.2.7.0-raspbian-armhf-10.tar.gz"
"Polar"                            "1.1.20.0"                      "https://dl.cloudsmith.io/public/opencpn/polar-prod/raw/names/Polar_pi-1.1.20.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.1.20.0/Polar_pi-1.1.20.0-raspbian-armhf-10.tar.gz"
"pypilot"                          "0.30.0.0"                      "https://dl.cloudsmith.io/public/opencpn/pypilot-prod/raw/names/pypilot_pi-0.30.0.0-raspbian-armhf-10-buster-armhf-tarball/versions/v0.30.0.0/pypilot_pi-0.30.0.0-raspbian-armhf-10.tar.gz"
"Race Start Display"               "1.1.0.0"                       "https://dl.cloudsmith.io/public/steven-adler/racingplugin-prod/raw/names/race_start_display_pi-1.1.0.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.1/race_start_display_pi-1.1.0.0-raspbian-armhf-10-buster-armhf.tar.gz"
"Radar"                            "5.2.2-+2107121208.be299ce"     "https://dl.cloudsmith.io/public/opencpn-radar-pi/opencpn-radar-pi-prod/raw/names/Radar-5.2-raspbian-10-tarball/versions/5.2.2-+2107121208.be299ce/Radar-5.2.2-+2107121208.be299ce_raspbian-10-armhf.tar.gz"
"S63"                              "1.17.1"                        "https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/s63-raspbian-armhf-10-tarball/versions/1.17.1.0.bcaad82/s63_pi-1.17.1-0_raspbian-armhf-10.tar.gz"
"sar"                              "2.8.2.0-+2108201407.8c9af67"   "https://dl.cloudsmith.io/public/opencpn/sar-alpha/raw/names/sar-2.8-raspbian-10-tarball/versions/2.8.2.0-+2108201407.8c9af67/sar-2.8.2.0-+2108201 [... cut ...]
"WeatherRouting"                   "1.13.32.0"                     "https://dl.cloudsmith.io/public/opencpn/weather-routing-prod/raw/names/weather_routing_pi-1.13.32.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.13.32.0/weather_routing_pi-1.13.32.0-raspbian-armhf-10.tar.gz"
"Windvane"                         "1.0.22.0"                      "https://dl.cloudsmith.io/public/opencpn/windvane-prod/raw/names/windvane_pi-1.0.22.0-raspbian-armhf-10-buster-armhf-tarball/versions/v1.0.22.0/windvane_pi-1.0.22.0-raspbian-armhf-10.tar.gz"
`

bdbcat, 2021-12-11:
Rick...
Pushed PR to AISRadar correcting fault.

rgleason, 2021-12-11:
Yes, thanks, merged.  Its all about extra spaces I think. How do I process files to remove those I wonder?  Perhaps my text editor Notebook++ has a command that will do that. Will check.  Will push again tomorrow.

bdbcat, 2021-12-11:
There is real code change here.
void AisFrame::OnSize ( wxSizeEvent& event )

rgleason, 2021-12-12:
Now I see these lines changed 

```
        m_pViewState->SetCanvasSize(dc.GetSize());    replaced  with
        wxSize cs = m_pCanvas->GetClientSize();
        m_pViewState->SetCanvasSize(cs);
```

rgleason, 2021-12-12:
```
DashboardTactics issue when enabling plugin:
ASSERT INFO:
../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime
```
This assert is coming from wxWidgets, what can be done?  Do you write around using that  datetime.h ?  or is it not formed properly?

rgleason, 2021-12-12:
All the same
DashboardTactics       ./include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime
OpenCPNdebugger  ../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime
Tactics                       ../include/wx/datetime.h(1740): assert "IsValid()" failed in GetTicks(): invalid wxDateTime

bdbcat, 2021-12-12:
I just posted a PR fixing it.
Check your email.

rgleason, 2021-12-12:
For Logbook  this assert
usr/include/wx-3.0/wx/arrstr.h(178): assert "nIndex < m_nCount" failed in Item(): wxArrayString: index out of bounds

Is untouchable and in wx.  Is this happening because Logbook code is quite old?

bdbcat, 2021-12-12:
Maybe...
I really prefer to work these issues one-at-a time.  I have other stuff going, and re-configuring to attack new bugs is not convenient sometimes...
Lets resolve Dash-T

rgleason, 2021-12-12:
Ok  merged, will build local for win and then push

rgleason, 2021-12-12:
They all completed.  https://github.com/rgleason/dashboard_tactics_pi/commits/mnt151-frontend2
Landed in CS alpha  https://cloudsmith.io/~opencpn/repos/dashboardtactics-alpha/packages/

bdbcat, 2021-12-12:
Dash-T import from Cloudsmith works OK on rPi Buster.
Ready for catalog update.

bdbcat, 2021-12-12:
Moving on to logbook.
Watch for a PR,please.

bdbcat, 2021-12-12:
Logbook....
Cannot reproduce fault on rpi or x86.
@mgrouch:  Please post the contents of your config file, the [logbook] section.
Thanks

bdbcat, 2021-12-12:
Polar is next.
Probably need to revert to "Polar" instead of "polar" in CMakeLists.txt:40.

set(PACKAGE "Polar")

That is the only way the a managed plugin can replace an old deb-installed plugin.
This will confuse some users who have already installed polar_pi from the PIM.  They will need to remove the  all  polar pis, and re-install.

rgleason, 2021-12-12:
```
Dash-T import from Cloudsmith works OK on rPi Buster.
Ready for catalog update.
```
OK will do, but I need to first apply TP 1.0.185 to fix MacOS again.

bdbcat, 2021-12-12:
Polar....
Re-thinking this:  I think the name has been "polar" for some time, including O5.2.4.  So it would be extremely disruptive to change that now.
Conclusion:
1.  Leave it as is, calling current PIM version "OK, no error"
2. @mgrouch:  You, and other affected users, should delete "/usr/lib/opencpn/libPolar_pi.so", most likely by removing/purging the previously installed .deb package.

Dave

rgleason, 2021-12-12:
Polar_pi: `Leave it as is, calling current PIM version "OK, no error"`
Thank you. I did make that change and tested it. I caused the icons not to be found and reverted back to the old icon image for unknown reasons. Then upon opening OpenCPN again, the green puzzle was prevelant until I reinstalled a version with the correct package name. This change would have wrecked havoc amoungst Windows users and Linux alike with regard to icons.

bdbcat, 2021-12-12:
Tactics_pi:
Pushed PR to rgleason repo, correcting wxDateTime ASSERT. Loads and enables now.
Still some trouble with toolbar icon.

rgleason, 2021-12-12:
RE:
```
Dash-T import from Cloudsmith works OK on rPi Buster.
Ready for catalog update.
```
For some reason my commit for Dashboard_Tactics  v1.6.10.0 has totally disappeared from [my repository branch mnt151-frontend2 ](https://github.com/rgleason/dashboard_tactics_pi/commits/mnt151-frontend2)  and I don't know why it would totally disappear.  Something has rewritten git history here.

Anyway, I've lost my updates for fixing MacOS and need to reapply the changes from TP 1.0.185, but when I do that it no longer builds, so I have to remember or figure out what I did to fix it.  That's where we are with Dashboard_Tactics right now. More later this afternoon. Have to paint.

rgleason, 2021-12-13:
Dash-Tactics  1.6.11 with bdbcat fix now pushed to Cloudsmith for test.
Tactics 1.0. 26  with bdbcat fix now pushed to CS for test.
https://cloudsmith.io/repos/

rgleason, 2021-12-13:
Dave merged dash-t and tactics  fixed for Raspbian.
https://github.com/OpenCPN/plugins/pull/537#event-5753992528

Mike, please test and advise, or close this if we are done. Thanks for reporting.

mgrouch, 2021-12-13:
> Dave merged dash-t and tactics fixed for Raspbian. [OpenCPN/plugins#537 (comment)](https://github.com/OpenCPN/plugins/pull/537#event-5753992528)
> 
> Mike, please test and advise, or close this if we are done. Thanks for reporting.

DashboardTactics seems loads fine now. 
Tactics is still missing icon on toolbar.

Thanks

mgrouch, 2021-12-13:
Also in DashboardTactics settings Tactics Performance Parameters tab
in radio button group second Radio button (under Use Heel Sensor) is missing text.
What does this second value mean?

Thanks

mgrouch, 2021-12-13:
> RE:
>
> ```
> Dash-T import from Cloudsmith works OK on rPi Buster.
> Ready for catalog update.
> ```
>
> For some reason my commit for Dashboard_Tactics v1.6.10.0 has totally disappeared from [my repository [...]

@rgleason 
You can retrieve the source code for past tags from 
https://github.com/rgleason/dashboard_tactics_pi/tags

there .zip and .tar.gz options.

If you make a github releases (instead of just tagging) hithub will also sore a copy of source code under 'releases'

Thanks

rgleason, 2021-12-13:
Mike thanls will look at that. Petri's upstream 2.x.x is further developed with greater features, but using some apps not compatible with this version if O until we move to wxwidgets 315.  You could try building it let us know how it goes.

rgleason, 2021-12-13:
`DashboardTactics settings Tactics Performance Parameters tab  in radio button group second Radio button (under Use Heel Sensor) is missing text.`
Yes, thank you will try to fix it.
This allows manual entry of the heel angle from a simple heel instrument on the boat. See the pop up message, also review the manual.

rgleason, 2021-12-13:
Mike, the answer about tactics  And dashboard_tactics is pretty clear.
1. Select the second radio button to set Manual Input of Heel.
2. Then adjust all the values below for each TWS/TWA.
So this problem isn't a problem.  Enter the TWS/TWA values below.

[image]

Full documents are here
https://dashboard-tactics-pi.readthedocs.io/en/v2.0.3/tweaks/tweaks.html#DashT/Tactics

rgleason, 2021-12-13:
Next is to figure out why Tactics Icons don't work on raspbian, but climatology and dashboard_tactics icons do work

rgleason, 2021-12-13:
Show me an exact screenshot please. This is not particularly easy to change.  Do you understand the point I made?
Now is this something else?

mgrouch, 2021-12-13:
Screensho
[image]
t

rgleason, 2021-12-13:
tactics_pi,cpp  line 3665
```
//--------------------
	m_ButtonFixedLeeway = new wxRadioButton(itemPanelNotebook03, wxID_ANY, _("fixed/max Leeway [\u00B0]:"), wxDefaultPosition, wxDefaultSize, 0);    <<<===============HERE
	itemFlexGridSizer06->Add(m_ButtonFixedLeeway, 0, wxALL, 5);
	m_ButtonFixedLeeway->SetValue(g_bUseFixedLeeway);
	m_ButtonFixedLeeway->SetToolTip(_("Dual purpose !\nIf Radiobutton is NOT set, then it's used to limit Leeway to a max value.\n If Radiobutton is set, then it fixes Leeway to this constant value."));

	m_ButtonFixedLeeway->Connect(wxEVT_COMMAND_RADIOBUTTON_SELECTED, wxCommandEventHandler(TacticsPreferencesDialog::OnManualHeelUpdate), NULL, this);

	m_fixedLeeway = new wxSpinCtrlDouble(itemPanelNotebook03, wxID_ANY, wxEmptyString, wxDefaultPosition, wxSize(-1, -1), wxSP_ARROW_KEYS, 0, 30, g_dfixedLeeway, 0.01);
	itemFlexGridSizer06->Add(m_fixedLeeway, 0, wxALIGN_LEFT, 0);
	m_fixedLeeway->SetValue(g_dfixedLeeway);
	//--------------------
```

rgleason, 2021-12-13:
` _("fixed/max Leeway [\u00B0]:")`

Does Raspian have a problem with  [\u00B0]  ?

rgleason, 2021-12-13:
Dashboard_tactics_pi    tactics_pi.cpp   LINE 3008

```
//--------------------
	m_ButtonFixedLeeway = new wxRadioButton(itemPanelNotebook03, wxID_ANY, _("fixed/max Leeway [\u00B0]:"), wxDefaultPosition, wxDefaultSize, 0);   <<<====HERE
	itemFlexGridSizer06->Add(m_ButtonFixedLeeway, 0, wxALL, 5);
	m_ButtonFixedLeeway->SetValue(g_bUseFixedLeeway);
	m_ButtonFixedLeeway->SetToolTip(_("Dual purpose !\nIf Radiobutton is NOT set, then it's used to limit Leeway to a max value.\n If Radiobutton is set, then it fixes Leeway to this constant value."));

	m_ButtonFixedLeeway->Connect(wxEVT_COMMAND_RADIOBUTTON_SELECTED, wxCommandEventHandler(TacticsPreferencesDialog::OnManualHeelUpdate), NULL, this);

	m_fixedLeeway = new wxSpinCtrlDouble(itemPanelNotebook03, wxID_ANY, wxEmptyString, wxDefaultPosition, wxSize(-1, -1), wxSP_ARROW_KEYS, 0, 30, g_dfixedLeeway, 0.01);
	itemFlexGridSizer06->Add(m_fixedLeeway, 0, wxALIGN_LEFT, 0);
	m_fixedLeeway->SetValue(g_dfixedLeeway);
```

rgleason, 2021-12-13:
The both look the same to me.

rgleason, 2021-12-14:
Looking towards the top of tactics_pi.cpp it appears that the code for the toolbar icons needs to be improved, which I will do in the next few days.

mgrouch, 2021-12-14:
windwane error on enabling:

ASSERT INFO:
../src/gtk/bitmap.cpp(924): assert "IsOk()" failed in GetWidth(): invalid bitmap

BACKTRACE:
[1] wxBitmap::GetWidth() const
[2] PlugInManager::LoadPlugInDirectory(wxString const&, bool, bool)
[3] PlugInManager::LoadAllPlugIns(bool, bool)
[4] pluginUtilHandler::OnPluginUtilAction(wxCommandEvent&)
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::ProcessEventLocally(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::ProcessPendingEvents()
[12] wxAppConsoleBase::ProcessPendingEvents()
[13] wxApp::DoIdle()

bdbcat, 2021-12-15:
Windvane:
More icon problems.

rgleason, 2021-12-17:
@mgrouch   Can you test a protype armhf Bullseye Plugin for Jon?  It is testplugin here  https://github.com/jongough/testplugin_pi/pull/235   He would like to know if it is working.

mgrouch, 2021-12-17:
> @mgrouch Can you test a protype armhf Bullseye Plugin for Jon? It is testplugin here [jongough/testplugin_pi#235](https://github.com/jongough/testplugin_pi/pull/235) He would like to know if it is working.

My environments are buster. I do not have working bullseye environment.

rgleason, 2021-12-17:
Is there anyone who has a bullseye environment who can test?

bdbcat, 2021-12-18:
Rick...
I'll give it smoke test sometime tonight.

bdbcat, 2021-12-18:
Jon/Rick...
Testplugin is NOGO on armhf Bullseye.
Probable cause:
   Looking at ocpn-plugins.xml, I do not see a build targeting raspbian-armhf, <target-version>11.

jongough, 2021-12-18:
The alpha cloudsmith repository contains testplugin 1.0.186.0 [https://cloudsmith.io/~opencpn/repos/testplugin-alpha/packages/detail/raw/testplugin_pi-1.0.186.0-raspbian-armhf-11-bullseye-armhf-tarball/1.0.186.0+8211.55ea34b/](https://cloudsmith.io/~opencpn/repos/testplugin-alpha/packages/detail/raw/testplugin_pi-1.0.186.0-raspbian-armhf-11-bullseye-armhf-tarball/1.0.186.0+8211.55ea34b/) which should work. I have not generated any of the other objects except the bullseye build. This should be importable into Plugin Manager if the file is downloaded from cloudsmith.

jongough, 2021-12-18:
I have just created a pull request in the alpha branch for this version of testplugin. Hopefully it should provide access without having to download the file manually.

[25 more replies]

closed 2022-03-25 by mgrouch