OpenCPN/OpenCPN #2497: Chart downloader: fails to download GSHHG catalogue

caesar, 2021-12-20:
Platform: macOS 12.1 (Arm64)
OCPN version: 5.6.0 (clean install)

When I add the "All GSHHG basemaps" catalogue and try to "update" it for the first time, I get the following error:

> Failed to Find New Catalog: https://raw.githubusercontent.com/chartcatalogs/catalogs/master/GSHHG_Catalog.xml

There is no such issue with other catalogues. The above URL appears to be valid.

(I'm not sure if this is an issue with the Chart Downloader built-in plugin or with the `chartcatalogs` data, but I note that that particular catalogue file hasn't been updated in the [chartcatalogs](https://github.com/chartcatalogs/catalogs/) repo for 4 years, so I assume it's the former.)

caesar, 2021-12-20:
I don't have this issue on Windows 11 on ARM (which runs OCPN in 32-bit mode using emulation).

nohal, 2021-12-20:
Seems OK on my Intel Mac as well. Is it only this particular catalog or is it reproducible with some others from `chartcatalogs`?

caesar, 2021-12-20:
Strange. I've only had an issue with that particular catalogue; every other one I've tried is fine.

bcn2, 2021-12-20:
We observe the same with our M1 macMini macOS 12.0.
[image]
1

caesar, 2021-12-20:
So far as I can see from the code, [that particular error message](https://github.com/OpenCPN/OpenCPN/blob/3bee7c232dcfc2c72059957b087a815010ea5779/plugins/chartdldr_pi/src/chartdldr_pi.cpp#L993-L1009) can only happen if there is a problem [copying the downloaded file to the destination location](https://github.com/OpenCPN/OpenCPN/blob/3bee7c232dcfc2c72059957b087a815010ea5779/plugins/chartdldr_pi/src/chartdldr_pi.cpp#L984) – or if the download failed altogether I guess.

Strange that it only happens with that file though. Others from the same repository are fine. I've also cloned the repo and modified the file, and it still doesn't work.

bcn2, 2021-12-20:
With this one I see the same error....
[image]

caesar, 2021-12-20:
Interesting, that one works fine for me.

nohal, 2022-05-04:
@caesar , @bcn2 is this still reproducible for you on any system with any catalog?

bcn2, 2022-05-05:
Still the same with that RNC Region 6 entry on macOS 12.3.1, all others do behave correctly. ENCs and RNCs. Had a failure with the GSHHG set, but deleting the entry in the chart list and pointing OCPN to the downloaded sets again, it does work now. Obviously in the case of GSHHG there are no updates, but the catalog gets retrieved correctly.

rgleason, 2022-05-05:
Win10
All GSHHS 
Custom Path
from: https://raw.githubusercontent.com/chartcatalogs/catalogs/master/GSHHG_Catalog.xml
to: C:\Data-Hesper\Charts\GSHHG

Works

bcn2, 2022-05-05:
Here we go:

Mac-mini:~ renate$ curl -v https://www.charts.noaa.gov/RNCs/06Region_RNCProdCat.xml > /dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 54.192.95.22:443...
* Connected to www.charts.noaa.gov (54.192.95.22) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/cert.pem
*  CApath: none
* (304) (OUT), TLS handshake, Client hello (1):
} [324 bytes data]
* (304) (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* (304) (IN), TLS handshake, Unknown (8):
{ [19 bytes data]
* (304) (IN), TLS handshake, Certificate (11):
{ [3922 bytes data]
* (304) (IN), TLS handshake, CERT verify (15):
{ [264 bytes data]
* (304) (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* (304) (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / AEAD-AES128-GCM-SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=Maryland; L=Silver Spring; O=NOAA; CN=charts.noaa.gov
*  start date: Sep 20 00:00:00 2021 GMT
*  expire date: Sep 20 23:59:59 2022 GMT
*  subjectAltName: host "www.charts.noaa.gov" matched cert's "www.charts.noaa.gov"
*  issuer: C=US; O=DigiCert Inc; CN=DigiCert TLS RSA SHA256 2020 CA1
*  SSL certificate verify ok.
* Using HTTP2, server supports multiplexing
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x14700a600)
> GET /RNCs/06Region_RNCProdCat.xml HTTP/2
> Host: www.charts.noaa.gov
> user-agent: curl/7.79.1
> accept: */*
> 
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 200 
< content-type: text/xml
< content-length: 227428
< date: Thu, 05 May 2022 12:05:11 GMT
< last-modified: Thu, 28 Apr 2022 13:59:42 GMT
< etag: "75b62e1e9938c49e26e1965dac6e33e4"
< x-amz-server-side-encryption: AES256
< x-amz-version-id: ioE8cx0o1IA7t19bKI91RnA_ntgY9bw9
< accept-ranges: bytes
< server: AmazonS3
< strict-transport-security: max-age=31536000; includeSubdomains; preload
< content-security-policy: default-src 'none'; img-src 'self' data: https://nauticalcharts.noaa.gov https://*.fastly.net; script-src 'self' 'unsafe-inline' 'unsafe-eval' data: https://dap.digitalgov.gov; script-src-elem 'self' 'unsafe-inline' data: https://dap.digitalgov.gov https://www.google-analytics.com; connect-src 'self' data: https://www.google-analytics.com https://secure.geonames.org; font-src 'self'; style-src 'self' 'unsafe-inline'; object-src 'none'
< x-content-type-options: nosniff
< x-frame-options: DENY
< x-xss-protection: 1; mode=block
< referrer-policy: same-origin
< x-cache: Miss from cloudfront
< via: 1.1 5a6a53b378e632acd04b15b8c3815b80.cloudfront.net (CloudFront)
< x-amz-cf-pop: MAD51-C3
< x-amz-cf-id: h71RoZZCmMLrAJ1kMBlIDvv-dDTLtXn2YxaJgSDtg8cSVSSSixPIkg==
< 
{ [1178 bytes data]
100  222k  100  222k    0     0   230k      0 --:--:-- --:--:-- --:--:--  231k
* Connection #0 to host www.charts.noaa.gov left intact

nohal, 2022-05-05:
OK, so no error and the only difference is that you get the file form a completely different AWS server than me, which is kind of normal on another continent...

bcn2, 2022-05-05:
Just tried to delete the RNC 06 catalog from the list and next adding it again, but that did not change anything..

bcn2, 2022-05-05:
From the OCPN log:
14:09:07.361   ERROR filefn.cpp:1045 File '/Users/renate/Documents/Charts/RNC/US_REGION06/06Region_RNCProdCat.xml' couldn't be removed (error 1: Operation not permitted)
14:09:07.361   ERROR filefn.cpp:932 Impossible to overwrite the file '/Users/renate/Documents/Charts/RNC/US_REGION06/06Region_RNCProdCat.xml' (error 0: Undefined error: 0)

nohal, 2022-05-05:
OK, try
```
ls -la /Users/renate/Documents/Charts/RNC/US_REGION06/06Region_RNCProdCat.xml
lsof|grep 06Region_RNCProdCat.xml
```
and post the output please

bcn2, 2022-05-05:
What I did: removed the XML catalog - the new catalog gets accepted and read. Select download all, start download.
All files rejected with:
14:18:36.808 MESSAGE chartdldr_pi.cpp:2037 chartdldr_pi: Going to extract '/Users/renate/Documents/Charts/RNC/US_REGION06/11547.zip'.
14:18:36.808   ERROR file.cpp:228 can't open file '/Users/renate/Documents/Charts/RNC/US_REGION06/11547/USERAGREEMENT.TXT' (error 1: Operation not permitted)
14:18:36.808 MESSAGE chartdldr_pi.cpp:2098 Can not create file '/Users/renate/Documents/Charts/RNC/US_REGION06/11547/USERAGREE
MENT.TXT'.

Permission issue?

nohal, 2022-05-05:
No idea, we need to have a look at those permissions

bcn2, 2022-05-05:
Mac-mini:~ renate$ ls -la /Users/renate/Documents/Charts/RNC/US_REGION06/06Region_RNCProdCat.xml
-rw-------@ 1 renate  staff  227428  5 may 14:17 /Users/renate/Documents/Charts/RNC/US_REGION06/06Region_RNCProdCat.xml
Mac-mini:~ renate$ lsof|grep 06Region_RNCProdCat.xml

nohal, 2022-05-05:
Yes, this is now OK, but you deleted and redownloaded it, so it is understandable, try `ls -la /Users/renate/Documents/Charts/RNC/US_REGION06/11547/`

bcn2, 2022-05-05:
which tells us:
Mac-mini:~ renate$ ls -la /Users/renate/Documents/Charts/RNC/US_REGION06/11547/
total 9200
drwxr-xr-x    5 renate  staff      160 11 sep  2018 .
drwxr-xr-x  128 renate  staff     4096  5 may 14:18 ..
-rw-r--r--    1 renate  staff      540 28 feb  2019 11547.BSB
-rw-r--r--    1 renate  staff  4696807 28 feb  2019 11547_1.KAP
-rw-r--r--    1 renate  staff     5897 28 feb  2019 USERAGREEMENT.TXT

nohal, 2022-05-05:
This is really weird, the permissions on the files and the directory containing them look completely sane to me

bcn2, 2022-05-05:
Deleted the directory, installed again, everything working....
Closed from my side.

nohal, 2022-05-05:
@caesar are you any smarter from what we found out here today? I'm not and tend to really close the issue if you won't give me some more juice as I have absolutely no idea what could be done to reproduce this...

caesar, 2022-05-05:
Hi Pavel, been super busy so I haven't looked at this again, sorry, but
I'll try it later when I get to my computer and see if I can still
reproduce. I'll let you know!

caesar, 2022-05-06:
Sooo… this definitely seems like a permissions issue of some sort, but it's hard to say what's causing it. I checked and just like @bcn2, I had the following in `opencpn.log`:
```log
23:55:46.587   ERROR filefn.cpp:1045 File '/Users/caesar/Documents/Charts/GSHHG/GSHHG_Catalog.xml' couldn't be removed (error 1: Operation not permitted)
23:55:46.587   ERROR filefn.cpp:932 Impossible to overwrite the file '/Users/caesar/Documents/Charts/GSHHG/GSHHG_Catalog.xml' (error 0: Undefined error: 0)
```
Deleting the offending `GSHHG_Catalog.xml` resolved the problem; when I then tried "updating" the catalogue it worked fine.

But the bizarre thing is, the newly-created file has identical permissions to the old one so far as I can see (I stupidly deleted the old one, so I can't actually see if they are identical files). 
`ls -la` gave this before:
```
-rw-------@ 1 caesar  staff   6.3K 11 Jun  2020 GSHHG_Catalog.xml
```
and this after:
```
-rw-------@ 1 caesar  staff   6.3K  7 May 00:02 GSHHG_Catalog.xml
```
so I'm fairly confused about what the actual problem could be.

The only thing that is clear is that the error message (`Failed to Find New Catalog: [url]`) is wrong! 🙂

caesar, 2022-05-06:
I said "just like @bcn2" but reading bcn's posts again, I see deleting the existing catalogue file didn't resolve the issue for him. For me it did. Strange... 😕

caesar, 2022-05-06:
I should add that it's a bit odd that the file is created with `-rw-------@` (600), I'd expect `-rw-r--r--@` (644). But since OCPN is running as my user that shouldn't cause this issue.

nohal, 2024-02-17:
In case anybody is still seeing this, please reopen or create a new issue and together with the relevant part of the log also post the output of
```sh
xattr /path/to/the/offending/file
xattr /path/to/the/directory/containing/it
```

closed 2024-02-17 by nohal