OpenCPN/OpenCPN #2498: Windows dark mode support

caesar, 2021-12-20:
OpenCPN does not appear to support Windows' dark mode at all – the app titlebar and controls still use light mode when Windows is set to dark mode (see screenshot below).
At least, this is the case on Windows 11… I haven't tested on Windows 10 lately but [last time I checked](https://github.com/OpenCPN/OpenCPN/issues/1412#issuecomment-498072301) it was the same (I did some work on this, mainly focused on Mac, a couple of years back in #1412 / #1420).

I haven't looked into this at all yet, so I don't know if proper dark mode support on Windows is feasible or maybe blocked by wxWidgets?
Either way I thought I'd open the issue to track this and I'll try and do some work on it when I can.

[image]

Hakansv, 2021-12-20:
I'm on Win 10 and it works as it use to. Toggle "half-dark - dark - half-dark - normal light" when using F5 or the menu.

Here half-dark:
[image]

nohal, 2021-12-20:
It probably won't be that easy, see https://forums.wxwidgets.org/viewtopic.php?t=45869

caesar, 2021-12-20:
> I'm on Win 10 and it works as it use to. Toggle "half-dark - dark - half-dark - normal light" when using F5 or the menu.

To clarify, OpenCPN's night mode is working "fine" (ie, as it always did), but the Windows theme is not reflected in the title bar or other native widgets. Perhaps this screenshot illustrates the issue better:

[image]

caesar, 2021-12-20:
> It probably won't be that easy, see https://forums.wxwidgets.org/viewtopic.php?t=45869

Ugh, I kind of suspected that. Anyway at least we can track the issue here even if we can't (fully?) solve it for now.

Hakansv, 2021-12-20:
That's true. We can't, what I learned, tweak Win's own territory. This is discussed in an old issue where @transmitterdan suggested a (partial) fix.

nohal, 2023-04-30:
Seems we will have it with wxWidgets 3.3, since https://github.com/wxWidgets/wxWidgets/pull/23028

nohal, 2024-02-13:
Postponing past 5.10 as we really need to wait for wx 3.3 being somewhat stable not to constantly trigger rebuilds of all the plugins.