OpenCPN/OpenCPN #2501: Issues with 4K resolution displays

mgrouch, 2021-12-29:
As described in this post

https://www.cruisersforum.com/forums/f134/feature-requests-30931-215.html#post3544277

rgleason, 2021-12-29:
I have a new Dell XPS 17 with 4k Display with Win10.

You have to set the OS System Display settings for text to 250%.
Opencpn does not have this function, I believe it is an OS function.
Doesn't kubuntu provide this important feature?
Here are my OS settings, OpenCPN settings, and right click result with 4k screen. 
[image]
[image]
[image]

rgleason, 2021-12-29:
Also see FAQ  here https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#small_menus_with_4k_displays