OpenCPN/OpenCPN #2503: appdata: wrong desktop file name

leamas, 2021-12-31:
The file data/opencpn.appdata.xml.in has wrong name for the desktop file. Practically, this means that GUI installers like Gnome Software cannot open OpenCPN after installation.

Problem discovered by automated Debian tests.

bdbcat, 2022-04-07:
Close?

leamas, 2022-04-07:
No, bug is still present in master. More specifically, the file opencpn.appdata.xml contains

     <launchable type="desktop-id">org.opencpn.OpenCPN.desktop</launchable>

However, the installed desktop filename is _opencpn.desktop_. The net result is that applications which tries to use the appdate info to start opencpn will fail.

Bug can be closed after merging #2505

PR #2505 (2022-04-28): Fix wrong desktop filename in appdata (#2503).
Warnings from debian checks -- the GUI install is not able  start the program.

EDIT: This fix is applied to downstream Debian packaging.

Closes: #2503
commit 56331988 (2022-04-28): Fix wrong desktop filename in appdata (#2503).

Warnings from debian checks -- the GUI install is not able
to start the program.

Closes: #2503
closed 2022-04-28 by nohal