OpenCPN/OpenCPN #2513: Flatpak grib_pi deosn't support xdg-email

xentac, 2022-01-08:
When attempting to use the Send feature of the grib plugin, the only error message that you see from the opencpn console log is "/usr/sbin/xdg-email: /usr/sbin/xdg-email: cannot execute binary file". I have xdg-utils installed, but I believe that flatpak doesn't have access directly to the xdg-email binary. I found https://github.com/flatpak/flatpak-xdg-utils and installed it, which seems to exist to bridge the flatpak->xdg-email divide, but I think that the grib_pi code needs to be updated to use it.

leamas, 2022-01-08:
>   I believe that flatpak doesn't have access directly to the xdg-email binary

Correct, this indeed a fact.

> but I think that the grib_pi code needs to be updated to use it.

There are some options here:
- Send the message directly using SMTP, avoiding the command line  dependency
- Make sure that flatpak-xdg-utils is installed.
- Break out of the sandbox, see for example https://unix.stackexchange.com/questions/412869

All of these options has their problems. we need to thinnk about it. There might be other options as well. Let's for now state that installing flatgpak-xdg-utils is a usable work-around.

Thanks for a good bug report!

leamas, 2022-01-11:
Procrastinating.

Breaking out of the sandbox just seems wrong, let's avoid that.

Depending on flatpak-xdg-utils is tricky since this is a "host" package. For example on Fedora, this is Fedora package. However, when we install the Flatpak opencpn, this is platform-independent and cannot have any dependencies on the Fedora package.

Hence we cannot automatically install the flatpak-xdg-utils package when installing the flatpak OpenCPN. 

It is possible to send email using libcurl, a library already used. This will depend on user configuration of SMTP host:port, and normally also user and password. Storing the password raises security concerns. This is probably more work and more to configure for user. OTOH, it's an option without external dependencies.

This could (should?) be configured. If the email option by default is disabled, no problems. When enabling it, we could check that prerequisites (xdg-utils or SMTP host:port) is available.