OpenCPN/OpenCPN #2516: pluginmanager: broken jail mechanism.

leamas, 2022-01-09:
The commits 72cbf98bb4 and  a2c6903c45a tries to handle repeated error messages when loading incompatible plugins by renaming the plugin.so library file to plugin.so.jail.

This is sort of broken by design since the plugin libraries lives in directories managed by the installer and  thus should be treated as read-only.  Nothing here should be changed outside of the installer, and it will fail in all cases this is enforced by the os, for example in legacy Linux plugins and some cases  also Flatpak.

We need to change this mechanism to some  kind of blacklist which lives in the configuration directory

bdbcat, 2022-01-09:
Not following.
If the "directories managed by the installer " are writable by OCPN installer in user space, what will the OS enforce?  I see no difference between the OCPN installer, and the rest of the plugin management facility. Maybe this is a flatpak thing?  And btw, legacy linux plugins are deprecated, and they "all" work without jailing today.

bdbcat, 2022-01-09:
Not to say that a blacklist might be cleaner, but is it necessary?

leamas, 2022-01-10:
> Not to say that a blacklist might not be cleaner, but is it necessary?

If we accept the idea that anything could change any part in the code, it's certainly  not necessary.

However, opencpn is not  a trivial 5 or 10 kloc project. Summing up src/ and include I see about 260 kloc.  As I see it, the only way to maintain such a big project is try to define smaller parts and define boundaries between them.

One such part is the installer. Besides the issue discussed here it owns the data it installs. It knows how to enumerate it since all installation paths are logged. It also knows it can wipe out any data in the directories it owns at it's own discretion. The documentation is in  https://opencpn-manuals.github.io/main/plugin-installer/0.0.1/Terminology.html. This is written for plugin developers, so it's a bit besides here. But still. 

The code discussed here breaks the installer boundaries, breaks the list of installed files, etc.

A cleaner solution is  strictly not "necessary" but   IMHO definitely worth it's price.  FWIW, I think the current fix will bite us sooner or later.

If it helps, you can put this issue  on my TODO list by assigning it to me.

bdbcat, 2022-01-10:
OK, you sold me.
No need to have a formal TODO list.  It is now an "issue", and so will not be lost/ignored.
We have a BLACKLIST already, hard-coded in the source.  Should expand on that code flow.
As an additional feature, we could extend this, by adding clauses to ocpn-plugins.xml which control/manage the BLACKLIST dynamically.  Food for thought.
Thanks for trying to keep us organized and forward thinking.
Dave

leamas, 2022-01-12:
One thought is that when we have a list of broken, unloadable  or blacklisted plugins this can be made visible in the GUI, perhaps as a new icon in the usual plugin list. After all, that a plugin just doesn't work is pretty important user feedback...

leamas, 2022-03-17:
@bdbcat : After filing #2516 I see two open issues:

  - Is the idea behind _lower_versions_ necessary? That is, is there any usecase where we want to blacklist a specific version, allowing both lower and higher ones? The normal use is to just say that any version  <= x.y is blacklisted and thus should be updated. FWIW, my gut feeling is that this should be simplified, dropping the _lower_versions_ boolean.
  - How do we handle totally obsolete plugins like upcoming oeSENC etc? To blacklist them is trivial, but we need a specific message in this case like "this plugin is obsoleted by o-charts_pi" in this case, right?

Thoughts?

leamas, 2022-11-30:
For some reason, probably related to the big pluginmanager - plugin_loader refacoring, the code which moves offending plugins to a .jail version is resurrected in current master 022feb2ef. Here, the evil moves is done in plugin_loader.cpp which obviously uses the code before #2582. 

Sigh

bdbcat, 2022-11-30:
Q: What is it that we want to do with plugins that appear to be compatible, but fail on load?  If we do nothing, this situation occurs on every  OCPN startups.  User will be annoyed.  Blacklist?

leamas, 2022-11-30:
Yes, we will blacklist, the code for the "new" blacklist is in already in place since #2582.

The logic for now is to show the message once in each session i. e. once when opencpn is started. I think this is the right thing to do, user could either live with the message or uninstall the plugin. What we should not do is to repeat the message after displaying it once in the startup.

PR #2582 (2022-04-28): Re-factor plugin blacklist (#2516)  
Create a separate blacklist object, avoiding the plugin renaming in #2516. 

The blacklist contains both hard-coded and dynamically added entries. It provides a basic framework which could be expanded with user GUI feedback and/or configuration files blacklisting.

Closes: #2516
PR #2991 (2023-02-06): plugin_loader: Drop resurrected jailing code  (#2516)
Re-factor mark_unloadable in new blacklist to return status, review and re-factor GUI messages.

Basically, restore things as they were before the failed rebase related to the plugin_manager -> plugin_loader refactoring + polish after more testing.

Closes: #2516